nextcloud/nextcloudpi #311: WebUI access : 403 returned after accepting cert.

AlanHasty, 2017-11-21:
I'm getting the following 403 when trying to use the WebUI i.e. 
https://github.com/nextcloud/nextcloudpi/wiki/How-to-configure-NextCloudPi ( WebUI section)

After accepting the security exception: I get the following error. 

"Forbidden
You don't have permission to access / on this server. "

I've tried rebooting after the installation (x2) and no change. 
The /var/log/apache2/error.log seems to concur
[Tue Nov 21 12:31:36.169917 2017] [authz_core:error] [pid 840:tid 1985111088] [client 172.27.17.14:39474] AH01630: client denied by server configuration: /var/www/ncp-web/
[Tue Nov 21 12:32:12.673589 2017] [authz_core:error] [pid 841:tid 1783784496] [client 172.27.17.14:39552] AH01630: client denied by server configuration: /var/www/ncp-web/
[Tue Nov 21 12:32:12.886170 2017] [authz_core:error] [pid 841:tid 1985111088] [client 172.27.17.14:39552] AH01630: client denied by server configuration: /var/www/ncp-web/favicon.ico

Paste the output of `sudo ncp-report` - I'd love to but it doesn't appear to be present.
root@nextcloudpi:/home/pi# find / -name "ncp*" 
/etc/apache2/sites-available/ncp.conf
/etc/apache2/sites-enabled/ncp.conf
/etc/cron.daily/ncp-check-version
/var/lib/apache2/site/enabled_by_admin/ncp
/var/www/ncp-web
/var/www/ncp-web/img/ncp-logo.svg
/var/www/ncp-web/wizard/img/ncp-logo.svg
/var/www/ncp-web/ncp-launcher.php
/var/www/ncp-web/ncp.js
/var/www/ncp-web/ncp-output.php
/var/www/ncp-web/ncp.css
/usr/include/linux/ncp_fs.h
/usr/include/linux/ncp_no.h
/usr/include/linux/ncp_mount.h
/usr/include/linux/ncp.h
/usr/local/etc/ncp-ascii.txt
/usr/local/etc/ncp-version
/usr/local/etc/ncp-baseimage
/usr/local/bin/ncp-notify-unattended-upgrade
/usr/local/bin/ncp-test-updates
/usr/local/bin/ncp-check-version
/usr/local/bin/ncp-notify-update
/usr/local/bin/ncp-scan
/usr/local/bin/ncp-update
/usr/local/bin/ncp-check-updates
/usr/share/doc/netcat-openbsd/examples/ncp
/usr/share/doc/netcat-traditional/examples/scripts/ncp
/home/www/ncp-launcher.sh

This is the image that I used which I downloaded.
$md5sum NextCloudPi_11-12-17.tar.bz2 
536734788ab5e7732e981346916ab104  NextCloudPi_11-12-17.tar.bz2
That I got from this url
https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/

nachoparker, 2017-11-21:
update before trying to run `ncp-report` from `sudo nextcloudpi-config`, or `sudo ncp-update`

nachoparker, 2017-11-21:
what password did you use to try to access ncp-web?

AlanHasty, 2017-11-21:
I was able to determine that the following files were preventing my access. 
In the /etc/apache2 directory: 
./sites-available/ncp.conf ./sites-enabled/ncp.conf
After adding my ip range to each of these, (in my case 172.27)
I was able to connect to the WebUI.

AlanHasty, 2017-11-21:
I wasn't even presented with an opportunity to give a password.

nachoparker, 2017-11-21:
ah I see, so you have a 172.xx password... I had to do that once in order to access on the docker bridge. 

Yes, I will include that range in the future

thanks for reporting

AlanHasty, 2017-11-21:
No that is not my password, my IP range is in the 172...

nachoparker, 2017-11-21:
> No that is not my password, my IP range is in the 172...

sorry, that's what I meant. Let'st leave it open until we include this in the config

nachoparker, 2017-11-24:
fixed, thank you

closed 2017-11-24 by nachoparker