nextcloud/nextcloudpi #312: [Suggestion] - Default Overclock settings

Windyo, 2017-11-21:
Hi ! So after testing Raid Arrays, I decided to test Nextcloudpi performance over different overclock settings.

What I realized is that even when under load from Apache or from the sync client, the Pi with the default image does not scale the proc rate. Running a monitoring showed that it continuously ran at 600Mhz, as if there was no actual load. This resulted in a VERY slow loading time via the web.

Activating force_turbo solved most of the issues, and the pi chugged along merrily at 120000.

The max performance I got with no stability issues on a RPi3 is the following (though YMMV as all boards are not minted equal)
arm_freq=1420
core_freq=500
over_voltage=6
disable_splash=1
boot_delay=1
gpu_mem=16
sdram_freq=500
force_turbo=1

That said, I think that by default all NCPi images should have at least force_turbo set to 1 to avoid aforementionned issues.
Putting other overclock settings as commented lines in the config.text could allow others to enable them without as much trial and error.

nachoparker, 2017-11-21:
Thank you, that is very interesting. The raspberry pi image only builds on top of Raspbian, so we are running with default settings.

What would be the consecuences of doing this? would it be harmful for the longevity of the board?

Windyo, 2017-11-21:
Overclock itself will lower the board's longevity.
Force_turbo forces the cpu to run at max speed all the time - probably will harm the board in the long run also, though to a much lesser extent.

I think you can offer force_turbo (coupled with boot_delay = 1) as an option in ncp-config with little to no issue other than a warning that it makes the CPU run at max speed constantly.

Enabling the other overclock options should only be done as an option with a warning attached to it regarding system instability, and care should be taken to sanitize user input to avoid people trying something stupid like a clock of 250000 or an over voltage of 20. Note that the raspberry pi 2 factory supports overclocking (though the 3 doesn't).

All in all I would offer to do an option in ncp-config for forceturbo/bootdelay, but for the other options, just document somewhere that oveclocking can be good if the system feels sluggish. Someone undertaking overclocking should know exactly what they're up against.

Force turbo and boot delay should be totally fine though. Jst make sure you do set boot delay when you set force turbo.

Pant, 2017-11-21:
> Running a monitoring showed that it continuously ran at 600Mhz

Please also share with us, what monitoring tool you used?

Windyo, 2017-11-21:
while sleep 1; do top -n1 >> cpu.log; done

I cross-checked it with a cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq >> cpucurfreq.log

Pant, 2017-11-21:
I did a successful experiment. First of all you where right and I verify that cpu freq was mostly at 600Mhz when browsing the files from the browser.

The experiment:
I run `while sleep 1; do cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq; done`  for tracking the cpu frequency every 1 second. On a second terminal I run htop (could be `top`) and I saw that Apache processes had `nice` value of `0`. So I run `renice -n -20 -u www-data`, so that all processes from user `www-data` run with `nice` value `-20`, which means higher priority (in the context of cpu time).

What happen is what we want, when browsing Nextcloud files, the frequency raised to 900Mhz. When the page finished loading, the frequency dropped back to 600Mhz.

So in my opinion we just should find a permanent way to `nice` all processes from user `www-data`  to `-20`. And I "vote" against `force_turbo` because this will lead the pi to work always on 900Mhz.

@Windyo Thank you for mentioning this issue, the result of this discussion will help a lot of people :-)

Windyo, 2017-11-21:
This is much more elegant than using turbo. If you find a way to ``nice`` the process every time, it's better on every single level.

I'll keep my settings for the moment (sooo fast right now 👍 ), but once it's in master i'll just remove force_turbo :)

I'll untag myself from the discussion tomorrow as my knowledge proabbly isn't good enough to help you further on this. Good luck gents, and thanks for NexctCloudPi :)

nachoparker, 2017-11-21:
good stuff gentlemen. well, the modern equivalent of nice is cgroups and that can probably be set in the systemd unit file for apache/php.

Still, it surprises me a bit that the CPU stays low when there is load without the nice setting. Might have to investigate the freq governor settings and such. 

In any case, Raspbian defaults are probably there for a reason, but I don't see why we cannot study the situation and offer the option (or different options) for overclocking to the user.

Thoughts still welcome

Windyo, 2017-11-23:
Funnily enough, checking load gives me results of 0 to 0.3 when not in use on NCP using Raid1. Before I always got loads of 0.8. Wondering if it's linked to the CPU freq or to the fact that I used a new image. Posting here to remember to investigate

GHoeher, 2017-12-07:
Hi, 
if you are using a Pi3 and are only lookinig at CPU0, this might be the reason, why you don't see frequency scaling. 
If I use the `while sleep 1; do cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq; done ` "script and extend it too Show also cpu1,cpu1 and cpu3, I can see that some CPU-cores are scaling up the frequency to 120000 whereas some stay at 60000 ...
So for me it Looks like the CPU scaling works as designed.

Windyo, 2017-12-07:
That's interesting but doesn't explain why the systeam reactivity on web browsing has become much more responsive just by _nice_ ing the processes....

More testing required I guess

GHoeher, 2017-12-07:
I agree that some more testing is required.
I just tried nicing my processes like Pant descibed - I can't see any significant changes in responsiveness on my system at all.
I know, that a single try might not be representativ, but I stopped following times (first value with default settings, 2nd value with nice -20):
open Webpage ~2sec   ~2sec
logon ~5 sec   ~7sec
switch from Files-view to dashboard ~3sec   ~3sec
switch from Dashboard to Calendar view (with two active calendars) ~5sec   ~5sec

Also subjective, I can't feel any difference - navigating the webinfterface is a little bit sluggish but acceptable no matter how niced the process is.

If priorizing the process speeds up browsing in your case, might it be, that another process you are running is the root cause and if you priorize www-data, then it gets more CPU time than the concurrenting process?

nachoparker, 2018-01-02:
I am thinking that this doesn't make sense as a setting, because now we don't only target the Raspberry Pi. Closing

closed 2018-01-02 by nachoparker