nextcloud/nextcloudpi #313: password field for  DDNS no-ip provider doesn't accept special characters.

AlanHasty, 2017-11-23:
In setting up the DDNS service for no-ip, I'm not able to enter my password.
The page gives an invalid input message after trying to run. 
Removing the special characters in my pw - causes the error message to go away, but of course I'm unable to update the IP. 
It appears that the password field does not accept the complete range of characters.

AlanHasty, 2017-11-23:
The special chars in my pw were * and &

nachoparker, 2017-11-23:
does it work from `sudo nextcloudpi-config`?

AlanHasty, 2017-11-25:
Hey,

The answer to the question "Does it work from nextcloudpi-config ? is no. Further the & is causing the following : "PASS_" to be added to the password field when I return to the no-ip config screen in nextcloudpi-config

nachoparker, 2017-11-25:
yes, this has been reported before now that I remember it. There's strong sanitizing for security purposes in ncp-web.

I'll try to have a look see if I can improve that in nextcloudpi-config. From there maybe we can lift up the input checks in ncp-web.

The whole thing is to try to prevent code injection attacks, so special characters for bash are not accepted

AlanHasty, 2017-11-27:
Hey, 
So if I wanted to take a look at fixing this i.e sanitizing the web input for the pw field, where would I start looking? 
I cloned the repo , as a start.   :-) 
Also do I need to build/debug on a pi or can it be done on a "regular" linux box?

Pant, 2017-11-27:
Hi @AlanHasty,

All stuff about NOIP is run in file ` nextcloudpi/etc/nextcloudpi-config.d/no-ip.sh`, which after the build is located in `/usr/local/etc/nextcloudpi-config.d/no-ip.sh`.

Anything you want to test should be done in this file.

A simple hardfix could be editing `no-ip.sh` and putting your password after the `PASS_=` variable.

P.s. @nachoparker correct me if I am wrong!

nachoparker, 2017-11-27:
you can test things in the docker [development environment](https://github.com/nextcloud/nextcloudpi/wiki/Development-environment). It is meant for web development, but you can also alter the files in `/usr/local/etc/nextcloud-config.d/` 

Another option I use regularly is to run an image with QEMU and develop there.

https://ownyourbits.com/2017/02/06/raspbian-on-qemu-with-network-access/

Maybe it would be easier to achieve this with virtualbox

If you want to allow characters that are special for bash, you have to scape them somehow

Daniel-Krebs, 2017-12-09:
We should add a hint for the user, that special characters are not supported ATM.

nachoparker, 2017-12-19:
added input validation to `v0.43.1`

closed 2017-12-19 by nachoparker