nextcloud/nextcloudpi #314: USBdrive not recognized after nc-format-USB

Ricardosgeral, 2017-11-23:
Hi

I'm trying to do a fresh install and i'm facing a problem that i never had.

I run nc-format-USB, giving the label myCloudDrive. The drive has 3Tb and the process is sucessful after about 15. 
however, after doing `cd /media` and `ls -lha` nothing is shown. Even after enabling nc-automount.

The strange thing is that if i put the drive in my laptop (with Ubunto) the drive is recognized.

I'm not understanding

nachoparker, 2017-11-24:
can you post the output of 

```
ls -l /media
```

and

```
systemctl status nc-automount
```

and

```
systemctl status nc-automount-links?
```

lsblk -f

and

dmesg | tail -20

AliasAndCo, 2017-11-24:
hi,

i have the same problem with hard drive 1TO; the installation wizard is ok but the cloud storage is on the sd card and not on the hard drive.
this afternoon, i will post the result of
ls -l /media & systemctl status nc-automount & systemctl status nc-automount-links?

nachoparker, 2017-11-24:
thanks, I am receiving reports of this lately, so later today I will also test this again myself with the latest image to be sure.

I am also dealing with this here

https://help.nextcloud.com/t/nextcloudpi-only-647-6-mb-available-despite-using-nc-automount/21377/55

https://help.nextcloud.com/t/nextcloudpi-issues-with-new-install/23744/10

If it happens to someone that can help investigate why udiskie is not working for some people, it would be nice because it is not happening for me

AliasAndCo, 2017-11-24:
i have only 647.6mb of memory on sd card also. it s just for info.

nachoparker, 2017-11-24:
I think we [found it](https://help.nextcloud.com/t/nextcloudpi-only-647-6-mb-available-despite-using-nc-automount/21377/60).

Try

```
sudo rm /etc/udev/rules.d/90-qemu.rules
sudo reboot
```

and after reboot automount should work again

```
ls -l /media
```

nachoparker, 2017-11-24:
Or just update to the latest version and try again

Please, reopen if the problem persists

closed 2017-11-24 by nachoparker

AliasAndCo, 2017-11-24:
I install with the new update but i have again the same problem.
So i applied your link : 

`sudo rm /etc/udev/rules.d/90-qemu.rules`

`sudo reboot`

nc-datadir in nextcloudpi panel and

`/media/myCloudDrive/ncdata`

And i am very happy ;) it's ok, i have the real stockage on my nextcloud (sorry for my english)

AliasAndCo, 2017-11-24:
re,

after severals test and reboot hard with power cut, MyCloudDrive go out when i do 

`ls -l /media`.

USBdrive is ok.

nachoparker, 2017-11-24:
@AliasAndCo great! thanks for the feedback

Ricardosgeral, 2017-11-25:
MyCloudDrive shows up now
thanks

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

nachoparker, 2017-11-25:
thanks for reporting

Ricardosgeral, 2017-11-26:
i'm having the problem again.

I've formated the USBdrive with nc-format-usb and then nc-automont
next i've remoded the drive and put it on my laptop to copy (used rsync)
the backup tar file to the drive.

Then when i put again the drive in the raspberry and the drive is not
showing up

root@nextcloudpi:/media# ls -l /media
total 0

root@nextcloudpi:/media# systemctl status nc-automount
● nc-automount.service - Automount USB drives
   Loaded: loaded (/usr/lib/systemd/system/nc-automount.service; enabled;
vendor
   Active: active (running) since Sun 2017-11-26 00:48:44 UTC; 50s ago
 Main PID: 2991 (udiskie)
   CGroup: /system.slice/nc-automount.service
           └─2991 /usr/bin/python3 /usr/bin/udiskie -NTF

Nov 26 00:48:44 nextcloudpi systemd[1]: Started Automount USB drives.
Nov 26 00:48:50 nextcloudpi udiskie[2991]: failed to mount
/org/freedesktop/UDis

[1]+  Stopped                 systemctl status nc-automount

root@nextcloudpi:/media# lsblk -f
NAME        FSTYPE LABEL        UUID
MOUNTPOINT
sda
└─sda1      ext4   myCloudDrive 4c15ea3c-5238-4cc5-b572-8a46d97e0d88
mmcblk0
├─mmcblk0p1 vfat   boot         B60A-B262                            /boot
└─mmcblk0p2 ext4                9a7608bd-5bff-4dfc-ac1d-63a956744162 /

root@nextcloudpi:/media# dmesg | tail -20
[  368.582613] systemd[1]: apt-daily-upgrade.timer: Adding 22min 6.558311s
random time.
[  381.494596] systemd[1]: apt-daily.timer: Adding 8h 2min 15.220475s
random time.
[  381.495085] systemd[1]: apt-daily-upgrade.timer: Adding 37min 59.151487s
random time.
[  381.801511] systemd[1]: apt-daily.timer: Adding 2h 47min 39.671630s
random time.
[  381.802000] systemd[1]: apt-daily-upgrade.timer: Adding 35min 34.684737s
random time.
[  386.151736] JBD2: Invalid checksum recovering block 277 in log
[  386.246994] JBD2: Invalid checksum recovering block 10 in log
[  386.251931] JBD2: Invalid checksum recovering block 11 in log
[  386.318020] JBD2: Invalid checksum recovering block 20 in log
[  386.339721] JBD2: Invalid checksum recovering block 23 in log
[  386.383231] JBD2: Invalid checksum recovering block 30 in log
[  386.428944] JBD2: Invalid checksum recovering block 37 in log
[  386.430425] JBD2: Invalid checksum recovering block 37 in log
[  386.437839] JBD2: Invalid checksum recovering block 38 in log
[  386.451250] JBD2: Invalid checksum recovering block 40 in log
[  386.488409] JBD2: Invalid checksum recovering block 46 in log
[  386.489984] JBD2: Invalid checksum recovering block 46 in log
[  386.540949] JBD2: Invalid checksum recovering block 54 in log
[  388.164897] JBD2: recovery failed
[  388.164914] EXT4-fs (sda1): error loading journal

this is frustating

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Ricardosgeral, 2017-11-26:
if i place the nextcloud directory in the usbdrive first and then copy the
tar file in my other linux machine it is recognized

this means that i haveto do nc-changedir first before copying the tar

nachoparker, 2017-11-27:
From the log it seems that the filesystem got corrupted. Did you unplug the USB drive without umounting it first? 

> this means that i haveto do nc-changedir first before copying the tar

no, that's not the case. It seems like you extracted the drive without umounting because I see filesystem trying to recover the journal (and failing)

Ricardosgeral, 2017-11-27:
ok. that may have happened yes