nextcloud/nextcloudpi #315: "Distro not supported"

tarbucndge, 2017-11-24:
Paste the output of `sudo ncp-report`

tarbucndge, 2017-11-24:
I have tried to run this command: "curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash" on my raspberry pi3 (the current image on the sdcard is dietpi) and I got the "Distro not supported" message. 
Please anyone let me know why? And what should I do?
Thanks very much.

nachoparker, 2017-11-24:
You need to be on Raspbian 9 stretch to use the 'curl installer'. You are probably on jessie 8

```
cat /etc/issue
```

closed 2017-11-24 by nachoparker

tarbucndge, 2017-11-24:
Thank you very much.

nachoparker, 2017-11-24:
if you don't want to upgrade your distro, you can use the docker image

tarbucndge, 2017-11-24:
how would I do that? I am a newbie
Thank you

nachoparker, 2017-11-24:
https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/

tarbucndge, 2017-11-24:
I will give it a try. Thank you!