nextcloud/nextcloudpi #318: Upgrade to 12.0.4 breaks my custom theme

sunjam, 2017-12-05:
After upgrading, my custom login page and logo have been replaced by the stock Nextcloud logo and backdrop.    
    
My custom Name, Web Address, and Slogan are still correct.

nachoparker, 2017-12-05:
how did you change the theme? through admin settings?

sunjam, 2017-12-05:
Yes, I changed the theme through admin settings a couple weeks ago.  The image and logo were replaced with Nextcloud's default during the upgrade to 12.0.4, but my custom theme color is still correct.  
[image]    
Interestingly, the NextcloudPi logo and my yellow theme do show up once I'm logged in...
[image]

nachoparker, 2017-12-06:
I tested upgrading from 12.0.3 and the NCP theme was unaffected. I suggest you report this to nextcloud/server

closed 2017-12-06 by nachoparker