nextcloud/nextcloudpi #319: LetsEncrypt failing :(

deljones, 2017-12-07:
Hi all and good morning

Done the upgrade to 0.41.6 and 12.0.4.
Everything ok but I just ran LetsEncrypt which failed. Rerun port-forwarding but still no joy.

Attached is the screen grab from running LetsEncrypt

Thank

Del

[image]

ovpc, 2017-12-07:
Hi,
Thanks for reporting!
Looks like some connection failed.
Can you reproduce the error? 
What actions did you take precisely? Be specific about steps you took, before and after, if you can, please. 

I have upgraded to 0.41.6 and 12.0.4.also, have not tried to use letsencrypt tho....
I ran certbot manually without any issue and obtained my certificate.

I know Nacho is out of town for cpl days, so bear with us if no one else jumps in. Thnx

deljones, 2017-12-07:
Hi and thanks for responding so quick...

Well I didn't do much really... 

- Update to 0.41.6 which went ok. 

- Logged into my NC instance as admin

- Used the updater there to update to NC12.0.4 which went ok

- Logged into the ncp web panel and checked through all of my setting there to make sure that all was well. Checked locations of ncdata, database etc. Ran port forwarding to make sure that was ok, then LetsEncrypt to make sure that was ok but ended up with the attached in my opening attachment.

My system is running fine, no issues at all, its just that I'm not sure now that the certificate is installed correctly... Looking at the screenshot I'm guessing not :(

Del

ovpc, 2017-12-07:
What does it look like in webbrowser? Green or red lock, safe or unsafe?
Mine looks like this
[image]
You can right-click on the lock to get more info on certificat.

deljones, 2017-12-07:
Yes.. The same .. Locked. Its the renewal that might be tricky!

Del

nachoparker, 2017-12-12:
seems like an update issue with certbot, does this work?

```
/etc/letsencrypt/letsencrypt-auto --help
```

deljones, 2017-12-13:
Hi.. Sadly that didn't work

a problem while installing Python packages.

Here are the results

pip prints the following errors:
=====================================================
Collecting argparse==1.4.0 (from -r /tmp/tmp.NpSwCquzWi/letsencrypt-auto-requirements.txt (line 11))
Exception:
Traceback (most recent call last):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py", line 795, in get_page
    resp.raise_for_status()
  File "/opt/eff.org/certbot/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/models.py", line 893, in ra
se_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 404 Client Error: Not Found

nachoparker, 2017-12-13:
how about

```
sudo /etc/letsencrypt/letsencrypt-auto --help --no-bootstrap
```

nachoparker, 2017-12-13:
also, what's the output of

```
sudo ncp-report
```

and

```
cd /etc/letsencrypt
git describe
```

deljones, 2017-12-17:
Hi and sorry for the late reply...

I have been trying to solve this by my self but failed miserably!

git describe gives me the following:

v0.18.0-8-g68283940

many thanks

Del

On 13-Dec-17 6:40 AM, nachoparker wrote:
> |cd /etc/letsencrypt git describe|

-- 
* *
*Derek Jones Stage and Television Lighting Design
<http://www.derekjones.net/>*
*
*
w: www.derekjones.net <http://www.derekjones.net/>
t: +44 7957 140 512
 
Your Data. Your Way. Safe... <https://digitalcarnage.webhop.me>
 
The information in this E-Mail is confidential and may be legally
privileged. 
It may not represent the views of Derek Jones. 
It is intended solely for the addressees. 
Access to this E-Mail by anyone else is unauthorised. 
If you are not the intended recipient, any disclosure, copying,
distribution or any action taken or omitted to be taken in reliance on
it, is prohibited and may be unlawful. 
Any unauthorised recipient should advise the sender immediately of the
error in transmission.

nachoparker, 2017-12-17:
I am at `v0.19.0-33-ge2ab940a`. Try this first

```
cd /etc/letsencrypt
git pull
```

also, did you try this?

```
sudo /etc/letsencrypt/letsencrypt-auto --help --no-bootstrap
```

deljones, 2017-12-17:
Hello again...

|sudo ncp-report gives the following output: <details>
<summary>NextCloudPi diagnostics</summary> ``` NextCloudPi version
v0.41.13 NextCloudPi image NextCloudPi_09-12-17 distribution Raspbian
GNU/Linux 9 \n \l automount yes USB devices sda sdb datadir
/media/USBdrive/ncdata data in SD no data filesystem ext2/ext3 data disk
usage 14G/916G rootfs usage 1.9G/2.7G Nextcloud check ok Nextcloud
version 12.0.4.3 HTTPD service up PHP service up MariaDB service up
Redis service up Postfix service up internet check ok port check 80 open
port check 443 open IP 192.168.0.11 gateway 192.168.0.1 interface eth0
certificates none certs due none NAT loopback no uptime 1:00 ```
</details> <details> <summary>Nextcloud configuration</summary> ``` {
"system": { "passwordsalt": "***REMOVED SENSITIVE VALUE***", "secret":
"***REMOVED SENSITIVE VALUE***", "trusted_domains": [ "localhost",
"192.168.0.11", "nextdesk.co.uk", "digitalcarnage.co.uk",
"digitalcarnage.webhop.me" ], "datadirectory":
"\/media\/USBdrive\/ncdata", "overwrite.cli.url":
"https:\/\/nextdesk.co.uk", "dbtype": "mysql", "version": "12.0.4.3",
"dbname": "nextcloud", "dbhost": "localhost", "dbport": "",
"dbtableprefix": "oc_", "mysql.utf8mb4": true, "dbuser": "***REMOVED
SENSITIVE VALUE***", "dbpassword": "***REMOVED SENSITIVE VALUE***",
"installed": true, "instanceid": "ocah3xtdazof", "mail_smtpmode":
"smtp", "mail_smtpauthtype": "LOGIN", "mail_smtpsecure": "tls",
"mail_from_address": "xxxxx", "mail_domain": "nextdesk.co.uk",
"mail_smtphost": "auth.smtp.xxxxx.co.uk", "mail_smtpport": "587",
"mail_smtpauth": 1, "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
"mail_smtppassword": "***REMOVED SENSITIVE VALUE***", "maintenance":
false, "theme": "", "loglevel": "2", "log_type": "file",
"memcache.local": "\\OC\\Memcache\\Redis", "memcache.locking":
"\\OC\\Memcache\\Redis", "redis": { "host":
"\/var\/run\/redis\/redis.sock", "port": 0, "timeout": 0, "password":
"***REMOVED SENSITIVE VALUE***" } } } ``` </details> <details>
<summary>HTTPd logs</summary> ``` [Sun Dec 17 15:02:13.778793 2017]
[ssl:warn] [pid 675:tid 1992757248] AH01909: localhost:4443:0 server
certificate does NOT include an ID which matches the server name [Sun
Dec 17 15:02:14.045737 2017] [ssl:warn] [pid 898:tid 1992757248]
AH01909: localhost:4443:0 server certificate does NOT include an ID
which matches the server name [Sun Dec 17 15:02:15.003556 2017]
[mpm_event:notice] [pid 898:tid 1992757248] AH00489: Apache/2.4.25
(Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations [Sun
Dec 17 15:02:15.003677 2017] [core:notice] [pid 898:tid 1992757248]
AH00094: Command line: '/usr/sbin/apache2' ``` </details> <details>
<summary>Database logs</summary> ``` tail: cannot open
'/var/log/mysql/*.log' for reading: No such file or directory ```
</details> <details> <summary>Nextcloud logs</summary> ```
{"reqId":"WjahRH8AAQEAAAPCNmUAAAAO","level":"error","time":"2017-12-17T16:54:28+00:00","remoteAddr":"192.168.0.1","user":"deljones","app":"no
app in
context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v1\/notifications?format=json","message":"","userAgent":"Mozilla\/5.0
(Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.4.3"}
{"reqId":"WjaiIH8AAQEAAAPDkygAAABV","level":"error","time":"2017-12-17T16:58:08+00:00","remoteAddr":"192.168.0.1","user":"deljones","app":"no
app in
context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v1\/notifications?format=json","message":"","userAgent":"Mozilla\/5.0
(Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.4.3"}
{"reqId":"WjaicH8AAQEAAAPCNm4AAAAY","level":"error","time":"2017-12-17T16:59:28+00:00","remoteAddr":"192.168.0.1","user":"deljones","app":"no
app in
context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v1\/notifications?format=json","message":"","userAgent":"Mozilla\/5.0
(Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.4.3"}
{"reqId":"Xx7PvbTCPuJR06AVTn5c","level":"error","time":"2017-12-17T17:00:01+00:00","remoteAddr":"","user":"--","app":"no
app in
context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}
{"reqId":"Wjaip38AAQEAAAPDkz0AAABC","level":"error","time":"2017-12-17T17:00:23+00:00","remoteAddr":"192.168.0.1","user":"--","app":"no
app in
context","method":"GET","url":"\/index.php\/apps\/activity\/rss.php?token=7AXOps3o6lwX7qzcOzDRaVt93zv6z0","message":"","userAgent":"Mozilla\/5.0
(X11; Linux x86_64) AppleWebKit\/537.36(KHTML, like Gecko)
Chrome\/50.0.2661.75 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WjajTH8AAQEAAAPDk1YAAABL","level":"error","time":"2017-12-17T17:03:09+00:00","remoteAddr":"192.168.0.1","user":"deljones","app":"no
app in
context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v1\/notifications?format=json","message":"","userAgent":"Mozilla\/5.0
(Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.4.3"}
[... 50 lines cut ...]

On 13-Dec-17 6:40 AM, nachoparker wrote:
> |sudo ncp-report|

-- 
* *
*Derek Jones Stage and Television Lighting Design
<http://www.derekjones.net/>*
*
*
w: www.derekjones.net <http://www.derekjones.net/>
t: +44 7957 140 512
 
Your Data. Your Way. Safe... <https://digitalcarnage.webhop.me>
 
The information in this E-Mail is confidential and may be legally
privileged. 
It may not represent the views of Derek Jones. 
It is intended solely for the addressees. 
Access to this E-Mail by anyone else is unauthorised. 
If you are not the intended recipient, any disclosure, copying,
distribution or any action taken or omitted to be taken in reliance on
it, is prohibited and may be unlawful. 
Any unauthorised recipient should advise the sender immediately of the
error in transmission.

deljones, 2017-12-17:
Finally....

|sudo /etc/letsencrypt/letsencrypt-auto --help --no-bootstrap|

give the following output:

Creating virtual environment...

Installing Python packages...
Had a problem while installing Python packages.

pip prints the following errors:
=====================================================
Collecting argparse==1.4.0 (from -r
/tmp/tmp.9SaoyjWIax/letsencrypt-auto-require                         
ments.txt (line 11))
Exception:
Traceback (most recent call last):
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/basecomm                         
and.py", line 215, in main
    status = self.run(options, args)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/commands                         
/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_                         
set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_                         
set.py", line 554, in _prepare_file
    require_hashes
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_                         
install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py                         
", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py                         
", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py                         
", line 568, in _get_pages
    page = self._get_page(location)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py                         
", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/index.py                         
", line 795, in get_page
    resp.raise_for_status()
  File
"/opt/eff.org/certbot/venv/share/python-wheels/requests-2.12.4-py2.py3-no                         
ne-any.whl/requests/models.py", line 893, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 404 Client Error: Not Found for url:
https://www.piwheels.hostedpi.co                         
m/simple/zope.components/argparse/
=====================================================

Certbot has problem setting up the virtual environment.

We were not be able to guess the right solution from your pip
output.

Consult
https://certbot.eff.org/docs/install.html#problems-with-python-virtual-e                         
nvironment
for possible solutions.
You may also find some support resources at
https://certbot.eff.org/support/ .

-- 
* *
*Derek Jones Stage and Television Lighting Design
<http://www.derekjones.net/>*
*
*
w: www.derekjones.net <http://www.derekjones.net/>
t: +44 7957 140 512
 
Your Data. Your Way. Safe... <https://digitalcarnage.webhop.me>
 
The information in this E-Mail is confidential and may be legally
privileged. 
It may not represent the views of Derek Jones. 
It is intended solely for the addressees. 
Access to this E-Mail by anyone else is unauthorised. 
If you are not the intended recipient, any disclosure, copying,
distribution or any action taken or omitted to be taken in reliance on
it, is prohibited and may be unlawful. 
Any unauthorised recipient should advise the sender immediately of the
error in transmission.

nachoparker, 2017-12-17:
maybe if `git pull` doesn't work, just nuke everything and start over, that **has** to work

```
sudo rm -rf /etc/letsencrypt
sudo apt-get purge -y \
    make \
    git \
    augeas-lenses \
    binutils \
    cpp \
    cpp-6 \
    gcc \
    gcc-6 \
    libasan3 \
    libaugeas0 \
    libc-dev-bin \
    libc6-dev \
    libcc1-0 \                                                                                                                                                                                                                                 
    libexpat1-dev \
    libffi-dev \
    libgcc-6-dev \
    libgomp1 \
    libisl15 \
    libmpc3 \
    libpython-dev \
    libpython2.7-dev \
    libssl-dev \
    libubsan0 \
    linux-libc-dev \
    python-dev \
    python-virtualenv \
    python2.7-dev \
    python-pkg-resources \
    python3-pkg-resources \
    python3-virtualenv \
    virtualenv \
    python-pip-whl
sudo rm /usr/local/etc/nextcloudpi-config.d/letsencrypt.sh
sudo ncp-update
```

deljones, 2017-12-17:
Thanks for you quick reply

Followed the instructions but nothing.. The same error

pi@raspberrypi:~ $ sudo rm
/usr/local/etc/nextcloudpi-config.d/letsencrypt.sh
pi@raspberrypi:~ $ sudo ncp-update
Downloading updates
Performing updates
Installing letsencrypt
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
python2.7-minimal is already the newest version (2.7.13-2+deb9u2).
python2.7-minimal set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.
Cloning into 'letsencrypt'...
remote: Counting objects: 51588, done.
remote: Compressing objects: 100% (130/130), done.
remote: Total 51588 (delta 93), reused 83 (delta 40), pack-reused 51416
Receiving objects: 100% (51588/51588), 16.12 MiB | 2.89 MiB/s, done.
Resolving deltas: 100% (36975/36975), done.
Bootstrapping dependencies for Debian-based OSes... (you can skip this
with --no-bootstrap)
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
augeas-lenses is already the newest version (1.8.0-1+deb9u1).
ca-certificates is already the newest version (20161130+nmu1).
gcc is already the newest version (4:6.3.0-4).
libaugeas0 is already the newest version (1.8.0-1+deb9u1).
libffi-dev is already the newest version (3.2.1-6).
libssl-dev is already the newest version (1.1.0f-3+deb9u1).
openssl is already the newest version (1.1.0f-3+deb9u1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
virtualenv is already the newest version (15.1.0+ds-1).
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.
Creating virtual environment...
Installing Python packages...
Had a problem while installing Python packages.

pip prints the following errors:
=====================================================
Collecting argparse==1.4.0 (from -r
/tmp/tmp.PS7rqElHjx/letsencrypt-auto-requirements.txt (line 11))
Exception:
Traceback (most recent call last):
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/basecommand.py",
line 215, in main
    status = self.run(options, args)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/commands/install.py",
line 342, in run
    requirement_set.prepare_files(finder)
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py",
line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File
"/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/req/req_set.py",
line 554, in _prepare_file
[... 62 lines cut ...]

On 17-Dec-17 5:29 PM, nachoparker wrote:
> |sudo ncp-update|

-- 
* *
*Derek Jones Stage and Television Lighting Design
<http://www.derekjones.net/>*
*
*
w: www.derekjones.net <http://www.derekjones.net/>
t: +44 7957 140 512
 
Your Data. Your Way. Safe... <https://digitalcarnage.webhop.me>
 
The information in this E-Mail is confidential and may be legally
privileged. 
It may not represent the views of Derek Jones. 
It is intended solely for the addressees. 
Access to this E-Mail by anyone else is unauthorised. 
If you are not the intended recipient, any disclosure, copying,
distribution or any action taken or omitted to be taken in reliance on
it, is prohibited and may be unlawful. 
Any unauthorised recipient should advise the sender immediately of the
error in transmission.

nachoparker, 2017-12-17:
why doesn't it fail for other people?

starting to think we should report it to cerbot, like last time something like this happened

https://github.com/nextcloud/nextcloudpi/issues/230

Halloran, 2017-12-21:
I ran into this issue last night and ended up resolving it by commenting all lines in /etc/pip.conf

Would guess that it would also work to remove pip.conf altogether, though I did not actually try that.

nachoparker, 2017-12-21:
@deljones, could you try that? also, let's first check the contents of `/etc/pip.conf` to see if we discover what's going on.

```
sudo cat /etc/pip.conf
```

@Halloran thanks

deljones, 2017-12-21:
Hi all

I have just commented out the lines suggested by @Halloran and ran Letsencrypt again from ncp_config. Perfect.... After some installing etc I now get a message saying the following:

Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Cert not yet due for renewal
Keeping the existing certificate

-------------------------------------------------------------------------------
Certificate not yet due for renewal; no action taken.
-------------------------------------------------------------------------------
System config value trusted_domains => 4 set to string digitalcarnage.webhop.me
System config value overwrite.cli.url set to string https://digitalcarnage.webhop.me
Done. Press any key...

At last!... Seems to work.. Many thanks to @Halloran for spotting it. I'll keep my eye on it and see it if there is an auto renewal....

many thanks

Del

nachoparker, 2017-12-21:
that's good news... can you show us the content of that file?

closed 2017-12-21 by nachoparker

Halloran, 2017-12-21:
FWIW, in my case, the certificate renewal was needed, and it succeeded.

File contents below:
```
$ more /etc/pip.conf
[global]
# extra-index-url=https://www.piwheels.hostedpi.com/simple
# extra-index-url=https://www.piwheels.hostedpi.com/simple/zope.components
# extra-index-url=https://www.piwheels.hostedpi.com/simple/zope.component/
$
```

nachoparker, 2017-12-22:
I am suspecting that is a leftover for the workaround when #230 happened