nextcloud/nextcloudpi #320: no update - no internet connectivity

colognaut, 2017-12-07:
wanted to update from nextcloudpi 0.39 to newest version.
it always answers "no internet connectivity". 
i can ping ownyourbits.com. further updates had no problem.

Tercus, 2017-12-08:
I am currently having the same error. Also mentioned it in [this forum](https://help.nextcloud.com/t/nextcloudpi-update-fails-no-internet-connectivity/24469) thread over at the nextcloud forum.

svankan, 2017-12-08:
I have the same problem because I am not able to ping github.com. The script verify internet connection in ncp-update with this.

`ping  -W 2 -w 1 -q github.com &>/dev/null || { echo "No internet connectivity"; exit 1; }`

A friend with the same ISP have the same problem and I it works with other ISP vendors. 
I have opened a support case for that.
Is it possible change to curl instead of ping? 
As a workaround I change github.com to something else where ICMP is allowed and is working.
After the upgrade I have to change manually again.

colognaut, 2017-12-08:
@svankan where can i find the ncp-update script?

svankan, 2017-12-08:
@colognaut Look in /usr/local/bin/ directory.

colognaut, 2017-12-08:
@svankan helped. thx!

Daniel-Krebs, 2017-12-08:
I had the same problem yesterday, for me changing the DNS Server in my router did it. My ISP somehow gives wrong information for github.com (and only for github.com)

xedoc64, 2017-12-08:
Got the same problem. I also get no ICMP response from different providers, maybe github changed some network stuff at their site.

It would be nice if the host which is used for the internet connectivity check can be changed atone place (config menu or a variable in library.sh?).

ChriKn, 2017-12-08:
May I believe some of you are on untiymedia in germany ? Same problem here... 
And I can't change my DNS inside my router sadly...

Daniel-Krebs, 2017-12-08:
Germany but Vodafone

xedoc64, 2017-12-08:
Germany and Telecolumbus.

matzel82, 2017-12-09:
I have the same Problem.
I dont have any solution.

Daniel-Krebs, 2017-12-09:
@matzel82 can you change your DNS server in your router?

matzel82, 2017-12-09:
@Daniel...

Yes...its possible. How ?

Es wäre möglich. Ich nutze eine Fritzbox. In was sollte ich es ändern?

matzel82, 2017-12-09:
Its working very fine 

8.8.8.8
8.8.4.4

Daniel-Krebs, 2017-12-09:
Glad that worked, if you don't want to use google (8.8.8.8 belongs to google) you can also use 9.9.9.9. See [heise](https://www.heise.de/newsticker/meldung/Quad9-Datenschutzfreundliche-Alternative-zum-Google-DNS-3890741.html)

ChriKn, 2017-12-09:
Is this just needed for updates or constantly ?
I can't install letsencrypt right now because of this and would just go to a neighbour with a FritzBox in this case to be able to install it quickly if it is not needed for day-to-day use...

Daniel-Krebs, 2017-12-09:
Don't know, to mee it is not clear if this is some github dns misconfiguration or vodafone/unitymedia error

matzel82, 2017-12-09:
I have Vodafone too. This can it explain

Tercus, 2017-12-09:
working temporary solution for me was this:

ssh into your raspberrypi. Check if this only affects github by manually running `ping example.com` and then `ping github.com`. If example.com returns your ping, but github doesn't run `sudo nano /usr/local/bin/ncp-update` and change in the 7th line the `ping github.com` to `ping example.com` and save the changes. The simply run `sudo ncp-update` to run the update.

ChriKn, 2017-12-09:
This works for ncp-update (this was explained in the 5th post, but you wrote it clearer).
However, letsencrypt doesn't work with this method.
I get this as a result : ./letsencrypt.sh: line 68: /etc/letsencrypt/letsencrypt-auto: No such file or directory

Daniel-Krebs, 2017-12-09:
This doesn't work because letsencrypt.sh git clones https://github.com/letsencrypt/letsencrypt which obviously is on github.com...

beyercenter, 2017-12-10:
yes, there is something wrong with DNS on github.com:

`nslookup github.com 8.8.8.8
Server:         8.8.8.8
Address:        8.8.8.8#53

Non-authoritative answer:
Name:   github.com
Address: 192.30.253.113
Name:   github.com
Address: 192.30.253.112

XXX@nextcloudpi:~ $ nslookup github.com 84.200.69.80
Server:         84.200.69.80
Address:        84.200.69.80#53

Non-authoritative answer:
Name:   github.com
Address: 18.195.85.27
Name:   github.com
Address: 35.159.8.160
Name:   github.com
Address: 18.194.104.89
`
Last IPs are Amazon EC2. This is what ping github.com on nextcloudpi is looking for (at my connection). Second DNS is https://dns.watch and first one is from quad9.net

pixeldrama, 2017-12-11:
You cannot rely on ping. github.com switched off ping in Germany and in a few other regions anymore, since they switched ping off this weekend.

nachoparker, 2017-12-11:
> You cannot rely on ping. github.com switched off ping in Germany and in a few other regions anymore, since they switched ping off the last weekend.

Have they deliberatedly and permantently switched off ping?

pixeldrama, 2017-12-11:
> Have they deliberatedly and permantently switched off ping?

Yes. Just on Saturday. Temporarily it will work for the moment. But never rely on ping.

nachoparker, 2017-12-12:
In that case we'll move away from pinging github. Changed to google.com in `v0.41.7`

closed 2017-12-12 by nachoparker