nextcloud/nextcloudpi #321: occ: command not found

sunjam, 2017-12-07:
`ls` shows that occ is in the assumed directory, but here is output when I attempt to run nc-info from the terminal.  [Also, someone else is posting about the occ command not found in this help thread.](https://help.nextcloud.com/t/nextcloudpi-error-sudo-var-www-nextcloud-occ-command-not-found/24462/2)
```
Launching nc-info
cat: /usr/local/etc/ncp-baseimage: No such file or directory
sudo: /var/www/nextcloud/occ: command not found

```

nachoparker, 2017-12-12:
can you verify that `occ` has no executable right? how could this happen?

nachoparker, 2017-12-14:
I suspect that this could have happened after upgrading to NC 12.0.4... is this possible?

nachoparker, 2017-12-14:
I am pretty sure that's what happened (look at other thread)

will fix soon

nachoparker, 2017-12-14:
fixed in `v0.41.9`

closed 2017-12-14 by nachoparker

rahul606, 2018-08-06:
Hi, while running bash talk.sh command I having this error:

> sudo: /var/www/nextcloud/occ: command not found
> /dev/fd/63: line 631: [: : integer expression expected
> /dev/fd/63: line 667: [: : integer expression expected
> Could not open input file: /var/www/nextcloud/occ
> Sorry but something went wrong. Please report this issue to https://github.com/nextcloud/vm/issues and include the output of the error message. Thank you!
> sudo -u www-data php /var/www/nextcloud/occ -V failed

Just to mention I have installed the NextCloud in a different location on my server as: /var/www/abc/html/ 

So, command is looking for occ file in a location which does not exist in my server. Please let me know the solution.