nextcloud/nextcloudpi #324: fail2ban not working if ncdata on USBdrive

peter-sc, 2017-12-13:
Hello, after the NC data is moved to USB drive, then the config `/etc/fail2ban/jail.conf`  should also change the location of `nextcloud.log` from `/var/www/nextcloud/data/nextcloud.log` to `/media/USBdrive/ncdata/nextcloud.log`.

From:
```
/etc/fail2ban/jail.conf 
[nextcloud]
enabled  = true
port     = http,https
filter   = nextcloud
logpath  = /var/www/nextcloud/data/nextcloud.log
maxretry = 3
```

To:
```
/etc/fail2ban/jail.conf 
[nextcloud]
enabled  = true
port     = http,https
filter   = nextcloud
logpath  = /media/USBdrive/ncdata/nextcloud.log
maxretry = 3
```

nachoparker, 2017-12-13:
There's a  PR that fixes this

https://github.com/nextcloud/nextcloudpi/pull/323

thanks for reporting

nachoparker, 2017-12-14:
fixed in `v0.48.1`

closed 2017-12-14 by nachoparker