nextcloud/nextcloudpi #326: Dnsmasq isn't working after restart

escoand, 2017-12-14:
After some investigating I realized it is just listening on lo interface.
Adding the real interface to the config solved the problem.
https://daffy.za.net/2015/06/dnsmasq-suddenly-stopped-working/

nachoparker, 2017-12-15:
mmm I have not had that issue...

I have seen that but it still works for me. I guess it would be better to include the line anyway

```
interface=eth0
```

escoand, 2017-12-15:
Don't know why this is happening now but it's replicable. Some weeks ago it worked without that problem, I think.

When I reconfigure dnsmasq in the web ui it is working without the option but not after a reboot. First I thought it's because the interface is maybe up too late. But also with restarting the daemon it's not working.

By the may, on my RPI the interface is called ▄▄▄▄▄▄▄▄▄▄▄▄▄.

nachoparker, 2017-12-16:
implemented in `v0.41.10`

closed 2017-12-16 by nachoparker