nextcloud/nextcloudpi #327: Feature Exploration - Lynis system audit

Iolaum, 2017-12-15:
Exploring [lynis](https://github.com/CISOfy/Lynis), a security auditing for UNIX derivatives.

Paste the output of `sudo ncp-report`

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.41.8
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      1.3G/917G
rootfs usage         1.6G/3.6G
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.20
gateway              192.168.1.1
interface            eth0
certificates         my.custom.domain
certs due            88 days
NAT loopback         no
uptime               1:42

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "my.custom.domain:443",
            "1": "192.168.1.20",
            "4": "my.custom.domain",
            "2": "my.custom.domain"
        },
        "datadirectory": "\/media\/USBdrive\/ncdata",
        "overwrite.cli.url": "https:\/\/my.custom.domain",
        "dbtype": "mysql",
        "version": "12.0.4.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "ocs86rkb5syt",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
[... 710 lines cut ...]

================================================================================

  Lynis 2.5.7

  Auditing, system hardening, and compliance for UNIX-based systems
  (Linux, macOS, BSD, and others)

  2007-2017, CISOfy - https://cisofy.com/lynis/
  Enterprise support available (compliance, plugins, interface and tools)

================================================================================

  [TIP]: Enhance Lynis audits by adding your settings to custom.prf (see /etc/lynis/default.prf for all settings)

```

Since this is automated some warnings/suggestions might be off.

The ones I think are worthy of conideration (feel free to point out why I 'm wrong) are:

- [Can't find any security repository in /etc/apt/sources.list or sources.list.d directory [PKGS-7388]](https://cisofy.com/controls/PKGS-7388/)
- [No MySQL root password set [DBS-1816]](https://cisofy.com/controls/DBS-1816/)

Some others might be interested for our users to implement by themselves based on their specific setup.

nachoparker, 2017-12-15:
great stuff!

I'll have a close look later. Right now I can say that mysql does have root password ( in `/root/.my.cnf` ) that is randomly generated.

About security repository, there is no such thing in Raspbian, look [here](https://www.raspberrypi.org/forums/viewtopic.php?t=98006&p=680175)

nachoparker, 2017-12-16:
I had the redis password thing on my list already but it is good to know that it gets checked by lynis.

I suggest we start with the biggest offenders

 - [ ] SSH (only for new images)
 - [ ] redis
 - [ ] postfix
 - [ ] kernel hardening

after this, we should harden with some iptables rules and finally we could have a look at some of the tools proposed, such as the audit daemon.

Thoughts?

nachoparker, 2017-12-16:
first phase is done

> after this, we could harden further with some iptables rules and finally we could have a look at some of the tools proposed, such as the audit daemon.

this would be next

closed 2017-12-16 by nachoparker

nachoparker, 2017-12-17:
After a second round we went from 62 to 81!

```
Hardening index : 81 [################    ]
```

I added a internal root password to the database, even though `mysql_secure_installation` says that this is not required because of `/etc/.my.cnf`.

Most of the remaining warnings stem from the fact that a RPi is not a powerful server so we're not changing things like the partitioning scheme.

Also stuff such as samhain and tripwire use up a lot of resources so they won't be used by default.

Once we have the iptables rules the score should go up a little bit more ;)

nachoparker, 2017-12-17:
https://github.com/CISOfy/lynis/issues/288