nextcloud/nextcloudpi #330: Updating to v0.41.13 problems

NuclearFuzion, 2017-12-18:
Performed this update today, now it seems no webserver is running, netstaty show no service open on port 443:

Downloading updates
Performing updates
grep: /etc/redis/redis.conf: No such file or directory
sed: can't read /etc/redis/redis.conf: No such file or directory
sed: can't read /etc/redis/redis.conf: No such file or directory
sed: can't read /etc/postfix/main.cf: No such file or directory
sed: can't read /etc/postfix/main.cf: No such file or directory
fs.protected_hardlinks = 1
fs.protected_symlinks = 1
kernel.core_uses_pid = 1
kernel.dmesg_restrict = 1
kernel.kptr_restrict = 2
kernel.sysrq = 0
net.ipv4.conf.all.accept_redirects = 0
net.ipv4.conf.all.log_martians = 1
net.ipv4.conf.all.rp_filter = 1
net.ipv4.conf.all.send_redirects = 0
net.ipv4.conf.default.accept_redirects = 0
net.ipv4.conf.default.accept_source_route = 0
net.ipv4.conf.default.log_martians = 1
net.ipv4.tcp_timestamps = 0
net.ipv6.conf.all.accept_redirects = 0
net.ipv6.conf.default.accept_redirects = 0
grep: /root/.my.cnf: No such file or directory
^CNextCloudPi updated to version v0.41.13

Paste the output of `sudo ncp-report`

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-baseimage: No such file or directory
Cannot load Zend OPcache - it was already loaded

nachoparker, 2017-12-18:
You must have a quite old image that you haven't updated in a loong time. This is not recommended at all. Please try to run the update at least once a month, to keep things smooth and to get all the security enhancements and bugfixes.

Also the output from `ncp-report` seems truncated, are you sure you copied everything?

NuclearFuzion, 2017-12-19:
Sorry you're right, I didn't wait until the command was finished.
I try to use a reminder for updating, but it's true having to SSH into the Pi makes it less frequent, plus I usually update Debian base and forget about NextcloudPi.

Maybe I can can on manually until last release by installing something myself?

> <--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-baseimage: No such file or directory
Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.41.13
NextCloudPi image
distribution         Raspbian GNU/Linux 8 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      6.7G/29G
rootfs usage         6.7G/29G
Nextcloud check      ok
Nextcloud version    11.0.5.1
HTTPD service        down
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      down
internet check       ok
port check 80        open
port check 443       closed
IP                   10.10.10.123
gateway              10.10.10.10
interface            eth0
certificates         emika.fr 
certs due            32 days 
NAT loopback         no
uptime               12:35

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": [
            "localhost",
            "10.10.10.123",
            "emika.fr"
        ],
        "datadirectory": "\/var\/www\/nextcloud\/data",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "11.0.5.1",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
[... 34 lines cut ...]
```

{"reqId":"WjUOiX8AAQEAAFxEXkIAAAAB","remoteAddr":"164.52.7.131","app":"core","message":"Trusted domain error. \"164.52.7.131\" tried to access using \"83.203.28.86:443\" as host.","level":2,"time":"2017-12-16T12:16:11+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjUm5X8AAQEAAHuhuVcAAABS","remoteAddr":"216.218.206.67","app":"core","message":"Trusted domain error. \"216.218.206.67\" tried to access using \"83.203.28.86\" as host.","level":2,"time":"2017-12-16T14:00:06+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjWuv38AAQEAAAPjmocAAAAP","remoteAddr":"60.191.38.77","app":"core","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"83.203.244.122:443\" as host.","level":2,"time":"2017-12-16T23:39:43+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjZ@JX8AAQEAABg2GTMAAAAI","remoteAddr":"216.218.206.68","app":"core","message":"Trusted domain error. \"216.218.206.68\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T14:24:38+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjaJ2X8AAQEAABg5khcAAABT","remoteAddr":"60.191.38.77","app":"core","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"83.203.244.122:443\" as host.","level":2,"time":"2017-12-17T15:14:34+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"Wjacvn8AAQEAABg2GaUAAAAX","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:35:10+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjadOH8AAQEAABg5kpcAAABR","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:37:13+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"Wjadsn8AAQEAABg5kpgAAABF","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:39:14+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjaeLH8AAQEAABg5kpkAAABJ","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:41:16+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"Wjaepn8AAQEAABg5kpoAAABG","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:43:19+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjafIH8AAQEAABg5kpsAAABN","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:45:20+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjafmX8AAQEAABg2GbAAAAAH","remoteAddr":"72.189.7.25","app":"core","message":"Trusted domain error. \"72.189.7.25\" tried to access using \"83.203.244.122\" as host.","level":2,"time":"2017-12-17T16:47:22+00:00","method":"POST","url":"\/","user":"--","version":"11.0.5.1"}
[... cut ...]
{"reqId":"WjcwuH8AAQEAABg5lIkAAABS","remoteAddr":"60.191.38.77","app":"core","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"90.31.243.204:443\" as host.","level":2,"time":"2017-12-18T03:06:33+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}
{"reqId":"WjfDDH8AAQEAAD6T7h8AAABI","remoteAddr":"216.218.206.69","app":"core","message":"Trusted domain error. \"216.218.206.69\" tried to access using \"90.31.243.204\" as host.","level":2,"time":"2017-12-18T13:30:53+00:00","method":"GET","url":"\/","user":"--","version":"11.0.5.1"}

```
</details>

nachoparker, 2017-12-19:
> Maybe I can catch on manually until last release by installing something myself?

You are on Raspbian 8. Only Raspbian 9 is supported. Don't have Redis... nothing like that. You really need to update :D 

Also, NC11... totally outdated.

I would try to resurrect this, export everything (contacts, calendars, data...), flash the latest image and reimport.

We are missing the apache logs, try to find them and let's see why it is not up. Maybe also try

```
systemctl status apache2
```

In the future, note that you can manually update through the web interface (on port 4443), or just setup automatic daily updates and forget about it ( `nc-update-ncp` option ).

NuclearFuzion, 2017-12-19:
I'd be glad to help more but I'm not skilled enough to find Apache logs you would need, I have a ton of .gz and some .log files in /var/log/apache2...

I have no problem going throught the whole process of starting from scratch (Raspbian9, ncp and let's encrypt) as long as I can revive the web interface to export the data. My family members use this cloud with +8h timeshift and they're not skilled enough to export them for me from their PC/Android. If i recreate the exact same accounts, would you think their clients would connect back to the updated version?

Thanks for your help, I promise I'd have actived autoupdates if I knew about this feature beforehand ;)

pi@OwnYourBits:/var/log/apache2 $ systemctl status apache2
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: failed (Result: exit-code) since Mon 2017-12-18 15:17:10 AST; 1 day 3h ago
  Process: 610 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

nachoparker, 2017-12-20:
> If i recreate the exact same accounts, would you think their clients would connect back to the updated version?

I have made this work, should be fine

Let's try playing with apache to see why it's not starting. Probably a certificate not found issue or similar.

Try to restart it

```
systemctl restart apache2
```

and check status and logs

```
cat /var/log/apache2/error.log
```

```
systemctl status apache2
```

We should be able to see an error message this way

deljones, 2017-12-20:
Hi…

I have two SD cards which I rotate every time there is an update. my 0.41.13 was broken, hence this report so I went back to my previous version SD card, updated that to the current version 0.43.1. All my ncdata, database etc is on the SSD/USB. Everything is working except the Letsencrypt, the site is secure and passed the Nextcloud security test as well as SSLLABS. My concern is what will happen when the certificate expires. Will it auto renew etc?

With a fully working system using 43.1, I exported out the ncpi configuration, swapped back to the 41.13 CD card and imported the working  configuration back into that version to see if it would work seen as it was from a working SD card (43.1). 

Sadly that didn’t work I still have no access via the web. I ran the commands you posted and the replies to those commands are below. Looks like the apache server is fried.

My option is to reformat the broken SD card and do a copy of the current working card to that…

I’ll wait though just in case you find what the problem is

Best wishes

Del

> If i recreate the exact same accounts, would you think their clients would connect back to the updated version?
>
> I have made this work, should be fine
>
> Let's try playing with apache to see why it's [...]
Gives:
Failed to restart apachGe2.service: The name org.freedesktop.PolicyKit1 was not provided by any .service files
See system logs and 'systemctl status apache2.service' for details
> and check status and logs
> 
> cat /var/log/apache2/error.log
Gives:
> cat: /var/log/apache2/error.log: No such file or directory
> systemctl status apache2
Gives: apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-12-20 13:07:00 UTC; 3min 18s ago
  Process: 730 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)
  Process: 716 ExecStartPre=/bin/mkdir -p /var/log/apache2 (code=exited, status=0/SUCCESS)

Dec 20 13:06:58 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Dec 20 13:07:00 raspberrypi apachectl[730]: AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.con
Dec 20 13:07:00 raspberrypi apachectl[730]: SSLCertificateFile: file '/etc/letsencrypt/live/digitalcarnage.webhop
Dec 20 13:07:00 raspberrypi apachectl[730]: Action 'start' failed.
Dec 20 13:07:00 raspberrypi apachectl[730]: The Apache error log may have more information.
Dec 20 13:07:00 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Dec 20 13:07:00 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Dec 20 13:07:00 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Dec 20 13:07:00 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.

nachoparker, 2017-12-20:
@deljones  I think you meant to write in the other thread, I'll paste your comment there

nachoparker, 2017-12-22:
I'll close because it is not really an 'issue' but we can keep talking

closed 2017-12-22 by nachoparker

deljones, 2017-12-22:
@nachoparker excellent. Many thanks.

NuclearFuzion, 2018-01-03:
Yeah I finally retreived the files directly from SD from root. Had some problems with files rights and it involved a lot of "chmod", but the clients accepted new accounts on the new NextCloudPi with the exact sames login/passwords.

I strongly suggest that in the future the auto-updating is activated by default, or that we can update only gradually one step at a time (not sure it's possible to implement?).

Many thanks again for the great work :)

nachoparker, 2018-01-03:
I am glad you figured it out.

Did you re-upload the data? you can run `nc-fix-permissions` to be sure the permissions are fine.

Well, you can activate autoupdates. It doesn't come activated by default, but it does get activated if one does the first run wizard.

I thought that was a good compromise, but opinions are welcome ;D