nextcloud/nextcloudpi #331: No Web access to Nextcloudpi since 0.41.13

deljones, 2017-12-19:
I do the Nextcloudpi updates very regularly and have had no problems. I did the 0.41.13 update and all seemed well but some apps started to behave badly. I did a restart of the server but now I cant get any web access at all.  Client dont work either now. I can SSH into the server and I can do a sudo nextcloudpi-config and that all works...

Can anyone help? I'm lost and in a bit of a panic!

Many thanks

Del

nachoparker, 2017-12-19:
can you show the output of `sudo ncp-report`?

nachoparker, 2017-12-20:
> We should be able to see an error message this way

(pasted message from @deljones at #330)
---
Hi…

I have two SD cards which I rotate every time there is an update. my 0.41.13 was broken, hence this report so I went back to my previous version SD card, updated that to the current version 0.43.1. All my ncdata, database etc is on the SSD/USB. Everything is working except the Letsencrypt, the site is secure and passed the Nextcloud security test as well as SSLLABS. My concern is what will happen when the certificate expires. Will it auto renew etc?

With a fully working system using 43.1, I exported out the ncpi configuration, swapped back to the 41.13 CD card and imported the working  configuration back into that version to see if it would work seen as it was from a working SD card (43.1). 

Sadly that didn’t work I still have no access via the web. I ran the commands you posted and the replies to those commands are below. Looks like the apache server is fried.

My option is to reformat the broken SD card and do a copy of the current working card to that…

I’ll wait though just in case you find what the problem is

Best wishes

Del

> 
> If i recreate the exact same accounts, would you think their clients would connect back to the updated version?
> 
> I have made this work, should be fine
> 
> Let's try playing with apache to see why it's not starting. Probably a certificate not found issue or similar.
> 
> Try to restart it
> 
> systemctl restart apache2
Gives:
Failed to restart apachGe2.service: The name org.freedesktop.PolicyKit1 was not provided by any .service files
See system logs and 'systemctl status apache2.service' for details
> and check status and logs
> 
> cat /var/log/apache2/error.log
Gives:
> cat: /var/log/apache2/error.log: No such file or directory
> systemctl status apache2
Gives: apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-12-20 13:07:00 UTC; 3min 18s ago
  Process: 730 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)
  Process: 716 ExecStartPre=/bin/mkdir -p /var/log/apache2 (code=exited, status=0/SUCCESS)

Dec 20 13:06:58 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Dec 20 13:07:00 raspberrypi apachectl[730]: AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.con
Dec 20 13:07:00 raspberrypi apachectl[730]: SSLCertificateFile: file '/etc/letsencrypt/live/digitalcarnage.webhop
Dec 20 13:07:00 raspberrypi apachectl[730]: Action 'start' failed.
Dec 20 13:07:00 raspberrypi apachectl[730]: The Apache error log may have more information.
Dec 20 13:07:00 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Dec 20 13:07:00 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Dec 20 13:07:00 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Dec 20 13:07:00 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.

nachoparker, 2017-12-20:
We can see that your problem is that the letsencrypt certificate is not there, so it's not found at `/etc/letsencrypt/live/digitalcarnage.webhop...`

A quick fix is to delete those lines from `/etc/apache2/sites-available/nextcloud.conf` and `/etc/apache2/sites-available.ncp.conf`, but in the long run letsencrypt should be fixed so it can be summoned and autorenewed. Did you try to get help from the certbot people?

It is a bit hard to help because you ask one thing, I ask for the output of `ncp-report`, you give me something else, and also do many things in between, so by the third message the situation is completely different.

I would suggest that you export and backup, then import and restore in a fresh image. I think it's worth a try. If that doesn't work, write here and I'll help you step by step.

deljones, 2017-12-20:
So very sorry for any confusion…

I do have two issues and the confusion started when I jumped in with someone else who had the same problem with 0.41.13 (#331)

Issue 1. 
LetsEncrypt does not work for me. It used to, I got at least two renewal notices but then they stopped and when I ran the script I got errors. Sadly I didn’t get anywhere with the folk at certbot. :(
I thought I did send through the ncp-report but the confusion was that maybe I sent it under another enquiry. Sorry.
As of now: I am running a good system using version of 0.43.1 without Letsencrypt working as expected.

Below is the ncp-report

Issue 2.
When I did the upgrade to 0.41.13 I was not able to get any web access at all. I could SSH in but that was all. As we have moved on to another version I’ll leave that enquiry.

Regarding issue 2. it has kind of gone away now as that was on another SD card. As I explained I rotate the SD cards with each update, just in case something goes wrong I can always go back a step and have a working NC quickly.

I did take your advice and create a fresh image and imported a backup of the current working instance and the NCpi config onto my second SD card but when I tried to run it I had no web access. :( Any help you can give here would be appreciated. At the moment I’m just copying the SD cards but of course thats just kicking the letsencypt issue down the road each time .

Again sorry for the confusion and thanks so much for your help.

——
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": [
            "localhost",
            "192.168.0.11",
            "nextdesk.co.uk",
            "digitalcarnage.co.uk",
            "digitalcarnage.webhop.me"
        ],
        "datadirectory": "\/media\/USBdrive\/ncdata",
        "overwrite.cli.url": "https:\/\/nextdesk.co.uk",
        "dbtype": "mysql",
        "version": "12.0.4.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "ocah3xtdazof",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_smtpsecure": "tls",
        "mail_from_address": "admin",
        "mail_domain": "nextdesk.co.uk",
        "mail_smtphost": "auth.smtp.1and1.co.uk",
        "mail_smtpport": "587",
        "mail_smtpauth": 1,
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "maintenance": false,
        "theme": "",
        "loglevel": "2",
        "log_type": "file",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "\/var\/run\/redis\/redis.sock",
[... 38 lines cut ...]
{"reqId":"VGLYp49B9Vy0J7YovBiB","level":"error","time":"2017-12-20T16:06:38+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}
{"reqId":"W56557Z47O3vxww1d28b","level":"error","time":"2017-12-20T16:07:30+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}
{"reqId":"PmJ6gKTT5RK0wQRlXZTc","level":"error","time":"2017-12-20T16:09:22+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}
{"reqId":"nHgUL4OidLXDO4gY0Gwf","level":4,"time":"2017-12-20T16:15:38+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"An exception occurred while executing 'SELECT * FROM `oc_appconfig`':\n\nSQLSTATE[42S02]: Base table or view not found: 1146 Table 'nextcloud.oc_appconfig' doesn't exist","userAgent":"--","version":"12.0.4.3"}
{"reqId":"3oZ3cKmYhfLULmjmhxqN","level":"error","time":"2017-12-20T16:10:07+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WjqZG38AAQEAAAQLV58AAAMB","level":"error","time":"2017-12-20T17:08:43+00:00","remoteAddr":"192.168.0.1","user":"--","app":"no app in context","method":"GET","url":"\/","message":"","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_13_2) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.84 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WjqZG38AAQEAAAQLV6AAAAMC","level":"error","time":"2017-12-20T17:08:43+00:00","remoteAddr":"192.168.0.1","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/login","message":"","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_13_2) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.84 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WjqZHH8AAQEAAAQLV64AAAMK","level":"error","time":"2017-12-20T17:08:45+00:00","remoteAddr":"192.168.0.1","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/js\/core\/merged-share-backend.js?v=ba36aaa50f690b9f54d442fc75bfa9dc-41","message":"","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_13_2) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.84 Safari\/537.36","version":"12.0.4.3"}
[... cut ...]
{"reqId":"JZjn4Soj4Yu5peMCfvLd","level":"error","time":"2017-12-20T17:17:02+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}
{"reqId":"qTKzrBuTl9dw3BJlDIzM","level":"error","time":"2017-12-20T17:17:41+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"","userAgent":"--","version":"12.0.4.3"}

```
</details>

I did do and export and backup of this working SD card, created a fresh image on a new one the restored but I was not able to get web access.

nachoparker, 2017-12-21:
Ok, thanks for the clarifications, I was getting confused :D 

Now, let's see why you don't have web access... what is exactly the error? in your output of `sudo ncp-report` I am missing the first section, where it checks if services are up and all that.

nachoparker, 2017-12-21:
also, do you have access to ncp-web?

nachoparker, 2017-12-21:
seems like your problems were solved in the other thread ;)

closed 2017-12-21 by nachoparker

deljones, 2017-12-22:
@nachoparker yes thank you very much... Very happy

pixelriot, 2018-11-06:
Hi guys,

since I have a similar problem after upgrading to 0.66.1 I'll post here instead of opening a new thread:

Since the last update I cannot access nextcloud or the web-interface anymore although ssh-ing is still possible and everything else seems to run fine.

Here is my ncp-report:

```
pi@nextcloudpi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

NextCloudPi version  v0.66.1
NextCloudPi image    NextCloudPi_10-23-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.4G/118G
rootfs usage         2.4G/118G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.3.0
HTTPD service        down
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               9:43

</details>

<details>
<summary>Nextcloud configuration</summary>

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "192.168.1.100",
            "3": "rbey.ddns.net",
            "2": "rbey.ddns.net",
            "4": "rbey.ddns.net"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.3.0",
        "overwrite.cli.url": "https:\/\/rbey.ddns.net\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 40 lines cut ...]
{"reqId":"W9iUDd@meSiWWjfp@HNZUQAAShY","level":3,"time":"2018-10-30T17:25:34+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iUwd@meSiWWjfp@HNZnwAAVxY","level":3,"time":"2018-10-30T17:28:39+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iVQN@meSiWWjfp@HNZ2QAARg8","level":3,"time":"2018-10-30T17:30:41+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iVdN@meSiWWjfp@HNZ8AAAURk","level":3,"time":"2018-10-30T17:31:33+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iV6N@meSiWWjfp@HNaJAAAVBk","level":3,"time":"2018-10-30T17:33:31+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iWGt@meSiWWjfp@HNaNQAARRk","level":3,"time":"2018-10-30T17:34:18+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iXNt@meSiWWjfp@HNasAAAQxY","level":3,"time":"2018-10-30T17:39:04+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var\/www\/nextcloud\/apps\/music\/3rdparty\/getID3\/getid3\/module.tag.apetag.php#25","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}
{"reqId":"W9iXUN@meSiWWjfp@HNauwAAQxY","level":3,"time":"2018-10-30T17:39:29+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"POST","url":"\/index.php\/apps\/music\/api\/scan","message":"Undefined index: filesize at \/var [... cut ...]
{"reqId":"W9ij@N@meSiWWjfp@HNeSgAASw0","level":3,"time":"2018-10-30T18:33:30+00:00","remoteAddr":"92.194.31.150","user":"richi","app":"PHP","method":"GET","url":"\/index.php\/apps\/music\/api\/file\/25143\/details","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/music\/utility\/detailshelper.php#46","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"14.0.3.0"}

</details>
```

The problem probably comes from `HTTPD service down` but I cannot find a way to get that up running again.

Please help. I'm lost here :(

ovpc, 2018-11-06:
@pixelriot  This is not the place to ask questions or get help. Use [help.nextcloud.com](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)
Your question about starting HTTPD is answered [here](https://help.nextcloud.com/t/ssl-error-connection-refused-on-nextcloudpi/40579)

nachoparker, 2018-11-07:
@pixelriot this could also help

https://help.nextcloud.com/t/cannot-access-nextcloudpi-connection-refused-httpd-service-down/40637/6