nextcloud/nextcloudpi #332: Nextcloud inaccessible after restart

magdeoz, 2017-12-20:
I did a reboot of my raspberry (sudo reboot) and now it does not start the apache server

<details>
<summary>systemctl status apache2.service</summary>

● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-12-18 17:23:42 UTC; 1 day 23h ago

Dec 18 17:23:41 nextcloudpi systemd[1]: Starting The Apache HTTP Server...
Dec 18 17:23:42 nextcloudpi apachectl[631]: AH00526: Syntax error on line 1 of /etc/apache2/conf-enabled/javascript-common.conf:
Dec 18 17:23:42 nextcloudpi apachectl[631]: Invalid command 'Alias', perhaps misspelled or defined by a module not included in the server configuration
Dec 18 17:23:42 nextcloudpi apachectl[631]: Action 'start' failed.
Dec 18 17:23:42 nextcloudpi apachectl[631]: The Apache error log may have more information.
Dec 18 17:23:42 nextcloudpi systemd[1]: apache2.service: Control process exited, code=exited status=1
Dec 18 17:23:42 nextcloudpi systemd[1]: Failed to start The Apache HTTP Server.
Dec 18 17:23:42 nextcloudpi systemd[1]: apache2.service: Unit entered failed state.
Dec 18 17:23:42 nextcloudpi systemd[1]: apache2.service: Failed with result 'exit-code'.

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.41.13
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/Next-Drive
data in SD           no
data filesystem      btrfs
data disk usage      7.1G/466G
rootfs usage         1.9G/118G
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        down
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   192.168.0.20
gateway              192.168.0.1
interface            eth0
certificates         owdroid.com 
certs due            87 days 
NAT loopback         no
uptime               5min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.0.20",
[... 84 lines cut ...]
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Using Linux native AIO
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Using generic crc32 instructions
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Completed initialization of buffer pool
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Highest supported file format is Barracuda.
2017-12-18 17:23:54 1989148672 [Note] InnoDB: 128 rollback segment(s) are active.
2017-12-18 17:23:54 1989148672 [Note] InnoDB: Waiting for purge to start
2017-12-18 17:23:54 1989148672 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 9361970
2017-12-18 17:23:54 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-12-18 17:23:54 1989148672 [Note] Plugin 'FEEDBACK' is disabled.
2017-12-18 17:23:54 1989148672 [Note] Server socket created on IP: '127.0.0.1'.
2017-12-18 17:23:55 1989148672 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```
{"reqId":"WjbwT38AAQEAABj1wpsAAABW","level":4,"time":"2017-12-17T22:31:45+00:00","remoteAddr":"192.168.0.1","user":"admin","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/admin\/Im%C3%A1genes\/1494556477%25E3%2580%258BBy%2520Joker%2520king.jpg","message":"Exception: {\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\BadRequest\",\"Message\":\"expected filesize 1025355 got 327680\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php(151): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->put(Resource id #22)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('1494556477%E3%8...', Resource id #22)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('files\\\/admin\\\/Im\\\\xC3...', Resource id #22, NULL)\\n#3 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Server.php(258): Sabre\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#9 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#10 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/File.php\",\"Line\":151}","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.3 (Nextcloud)","version":"12.0.4.3"}
{"reqId":"Wjbwtn8AAQEAABj0edAAAAAJ","level":4,"time":"2017-12-17T22:33:27+00:00","remoteAddr":"192.168.0.1","user":"admin","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/uploads\/admin\/1177541579\/00000000","message":"Exception: {\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\BadRequest\",\"Message\":\"expected filesize 10000000 got 294912\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php(151): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->put(Resource id #22)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Upload\\\/UploadFolder.php(39): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('00000000', Resource id #22)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Upload\\\\UploadFolder->createFile('00000000', Resource id #22)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('uploads\\\/admin\\\/1...', Resource id #22, NULL)\\n#4 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Server.php(258): Sabre\\\\DAV\\\\Server->exec()\\n#9 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#10 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#11 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/File.php\",\"Line\":151}","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.3 (Nextcloud)","version":"12.0.4.3"}
{"reqId":"Wjb4s38AAQEAABj1xIsAAABS","level":3,"time":"2017-12-17T23:07:31+00:00","remoteAddr":"45.79.65.120","user":"--","app":"no app in context","method":"GET","url":"\/robot.txt","message":"Exception: {\"Exception\":\"Symfony\\\\Component\\\\Routing\\\\Exception\\\\RouteNotFoundException\",\"Message\":\"Unable to generate a URL for the named route \\\"theming.Theming.getLogo\\\" as such route does not exist.\",\"C [... cut ...] ate->fetchPage()\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/Base.php(132): OC_Template->fetchPage()\\n#11 \\\/var\\\/www\\\/nextcloud\\\/core\\\/templates\\\/404.php(10): OC\\\\Template\\\\Base->printPage()\\n#12 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/routing\\\/Generator\\\/UrlGenerator.php\",\"Line\":130}","userAgent":"Mozilla\/5.0 (Windows; U; Windows NT 5.1; en-US; rv:1.8.1.13) Gecko\/20080311 Firefox\/2.0.0.13","version":"12.0.4.3"}

nachoparker, 2017-12-21:
Why do you have that file `/etc/apache2/conf-enabled/javascript-common.conf`?

Did you install something else besides Nextcloud in NCP?

In any case, if you need mod_alias, you can try

```
sudo a2enmod alias
```

, but I would like to know how that file got there...

magdeoz, 2017-12-21:
The only package that I have installed is fish, I guess it's because of its web interface

nachoparker, 2017-12-21:
did enabling the module solve it?

magdeoz, 2017-12-21:
I have uninstalled fish and it works perfectly again, I tried to enable the module but still gave me the same error

nachoparker, 2017-12-21:
I think that you have to restart apache for the changes to take effect.

closed 2017-12-21 by nachoparker