nextcloud/nextcloudpi #334: symlink /media/usbdrive lost after reboot

BETEP-ok, 2017-12-25:
as a followup to #181
----

sorry to say, but I also have issues with the fresh install of NextcloudPi

after fresh install nd configuration- every time I reboot- the symlink is lost.
I still can see my usb drive as /media/myCloudDrive
but /media/USBDrive is just not there.

good enough the trick with /usr/local/etc/nc-automount-links worked and symlink popped up, but it doesn't seem like a bulletproof solution.

I don't want to type in ssh commands after every powerloss )))

nextcloudpi ver 0.43.1 (the latest as of now)

nachoparker, 2017-12-26:
> good enough the trick with /usr/local/etc/nc-automount-links worked and symlink popped up, but it doesn't seem like a bulletproof solution.

can you explain this "trick"? does it survive a reboot? if it doesn't, what's the output of

```
systemctl status nc-automount-links
```

Also, please include the output of `sudo ncp-report`

BETEP-ok, 2017-12-26:
Hi ))  sure thing.

steps 2 reproduce:

1 fresh instal the ncp, set it up, change default passwords
2 nc-database
until first reboot everything works.
but after reboot- the USBdrive symlink is lost:

**ls -l /media/**

total 16
drwxr-xr-x 1 root root 20 Dec 15 19:44 myCloudDrive

**systemctl status nc-automount-links**
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-12-26 09:56:31 UTC; 39s ago
 Main PID: 353 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─353 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─372 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─373 grep --line-buffered ISDIR
           └─374 /bin/bash /usr/local/etc/nc-automount-links-mon

Dec 26 09:56:31 nextcloudpi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Dec 26 09:56:31 nextcloudpi nc-automount-links-mon[353]: Setting up watches.
Dec 26 09:56:31 nextcloudpi nc-automount-links-mon[353]: Watches established.
Dec 26 09:56:38 nextcloudpi nc-automount-links-mon[353]: myCloudDrive DELETE,ISDIR
Dec 26 09:56:43 nextcloudpi nc-automount-links-mon[353]: myCloudDrive CREATE,ISDIR

3 then I do the trick described in #181 :
"Try this command and let's see if the links appear

/usr/local/etc/nc-automount-links
ls -l /media"

**pi@nextcloudpi:~ $ sudo /usr/local/etc/nc-automount-links
pi@nextcloudpi:~ $ ls -l /media**
total 16
drwxr-xr-x 1 root root 20 Dec 15 19:44 myCloudDrive
lrwxrwxrwx 1 root root 19 Dec 26 10:02 USBdrive -> /media/myCloudDrive

the USBdrive is back.
but this symlink only stays until the next reboot. After the reboot- have to trigger the "sudo /usr/local/etc/nc-automount-links" again.

nachoparker, 2017-12-28:
that is weird... nc-automount-links is enabled and it detects the drive mount

```
Dec 26 09:56:43 nextcloudpi nc-automount-links-mon[353]: myCloudDrive CREATE,ISDIR
```

please, include the output of `sudo ncp-report` to see if I get some ideas.

Anyhow, it is recommended that you use `myCloudDrive` where possible, because also if you plug in another USB drive, you have no guarantees as of which one will be `USBdrive` and which one `USBdrive1`. It is better to use labels always if you know them.

I would change to `myCloudDrive` in the mysqld configuration file, and really everywhere (config.php if you used nc-datadir, and so on)

BETEP-ok, 2018-01-01:
hi again. well... on one side - "you are god damn right"
on the other - I specifically purchased this raps in order to have it as a dedicated file server.
So 99 percent of the time it is supposed to stay ON and that usb drive connected to it.
The setup wizard don't offer an alternative - so as a stupid user I rely on the default setting as a correct one.
and now I am simply scared t change anything in the existing set up because I don`t want to upload 200 gigs of family photos again )
so IF changing the USBdrive symlink => to /media/myCloudDrive goes beyond just using the **nc-database** script - I don`t dare to play with the settings deeper. ))) 

I think I will rarely connect additional USB drives to that machine. so most probably this will not be an issue for me, but I can easily imagine other users building more complicated set ups, utilising all the USB ports of the PI. so I guess the wizard routing needs to be adjusted.

as requested - sudo ncp-report:

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.1
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      109G/932G
rootfs usage         1.9G/15G
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   10.0.1.200
gateway              10.0.1.1
interface            eth0
certificates         betep.strangled.net
certs due            72 days
NAT loopback         no
uptime               6:51

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "10.0.1.200",
            "3": "betep.strangled.net",
            "2": "betep.strangled.net",
            "4": "betep.strangled.net"
        },
        "datadirectory": "\/media\/USBdrive\/ncdata",
        "overwrite.cli.url": "https:\/\/betep.strangled.net",
        "dbtype": "mysql",
        "version": "12.0.4.3",
[... 93 lines cut ...]
```

{"reqId":"WkO0138AAQEAACSlpK4AAABO","level":2,"time":"2017-12-27T14:57:27+00:00","remoteAddr":"148.72.168.39","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"148.72.168.39\" tried to access using \"188.194.162.43\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WkQqXX8AAQEAACSlqJ8AAABT","level":2,"time":"2017-12-27T23:18:53+00:00","remoteAddr":"148.72.168.39","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"148.72.168.39\" tried to access using \"188.194.162.43\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WkRYp38AAQEAACSlqbwAAABC","level":2,"time":"2017-12-28T02:36:23+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"188.194.162.43\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WkSXdH8AAQEAAEAOP5gAAABH","level":2,"time":"2017-12-28T07:04:20+00:00","remoteAddr":"139.162.113.204","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.113.204\" tried to access using \"188.194.162.43\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/50.0.2661.102 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WkTg7n8AAQEAAEAOQhAAAABX","level":2,"time":"2017-12-28T12:17:50+00:00","remoteAddr":"151.217.178.37","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"151.217.178.37\" tried to access using \"188.194.162.43\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/62.0.3202.62 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"WkT1a38AAQEAAEAOQrAAAABL","level":2,"time":"2017-12-28T13:45:15+00:00","remoteAddr":"74.82.47.5","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"74.82.47.5\" tried to access using \"188.194.162.43\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WkXM1H8AAQEAAEANtEIAAAAA","level":2,"time":"2017-12-29T05:04:20+00:00","remoteAddr":"184.154.189.90","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.154.189.90\" tried to access using \"188.194.162.43\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"12.0.4.3"}
{"reqId":"WkYvKX8AAQEAAFrmvAsAAABS","level":2,"time":"2017-12-29T12:03:53+00:00","remoteAddr":"185.35.63.68","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"185.35.63.68\" tried to access using \"188.194.162.43\" as host.","userAgent":"fa","version":"12.0.4.3"}
{"reqId":"Wkb4BH8AAQEAAAPsJT0AAAAO","level":2,"time":"2017-12-30T02:20:52+00:00","remoteAddr":"164.52.6.150","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"164.52.6.150\" tried [... cut ...]
{"reqId":"WklX6X8AAQEAAC3uHTAAAACH","level":3,"time":"2017-12-31T21:34:33+00:00","remoteAddr":"181.143.8.34","user":"--","app":"PHP","method":"GET","url":"\/admin\/config.php","message":"Exception: The requested uri(\/admin\/config.php) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"curl\/7.15.5 (x86_64-redhat-linux-gnu) libcurl\/7.15.5 OpenSSL\/0.9.8b zlib\/1.2.3 libidn\/0.6.5","version":"12.0.4.3"}

```
</details>

nachoparker, 2018-01-02:
Well, I see no reason why the link dissappears, because as shown in the output of `systemctl status nc-automount-links`, the drive dir creation is detected...

In any case, if you are scared to play around and want a quick solution, you can copy and paste these command and you should be good to go, as long as you don't touch the setting again

```
sudo systemctl disable nc-automount-links
sudo ln -s /media/myCloudDrive /media/USBdrive
```

If someone else is seeing the same, please reopen and see if we can detect what's going on. It is strange that noone else has reported this.

closed 2018-01-02 by nachoparker

BETEP-ok, 2018-01-02:
hi again, @nachoparker 

I generated the report AFTER the symlink was re-generated (manually)

stupid me.

will make another one tonight right after the reboot - maybe you can find a hint why its not being generated.

nachoparker, 2018-01-02:
sure, we'll do that

BETEP-ok, 2018-01-03:
hi, @nachoparker 

as promised- sudo ncp-report right after reboot

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.2
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata (doesn't exist)
rootfs usage         1.9G/15G
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   10.0.1.200
gateway              10.0.1.1
interface            eth0
certificates         betep.strangled.net
certs due            70 days
NAT loopback         no
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:145
Stack trace:
#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Jan 03 19:07:59.364966 2018] [ssl:warn] [pid 618:tid 1991979008] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Jan 03 19:08:00.058627 2018] [ssl:warn] [pid 762:tid 1991979008] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Jan 03 19:08:01.003367 2018] [mpm_event:notice] [pid 762:tid 1991979008] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Jan 03 19:08:01.003518 2018] [core:notice] [pid 762:tid 1991979008] AH00094: Command line: '/usr/sbin/apache2'
[Wed Jan 03 19:08:05.637041 2018] [authz_host:error] [pid 948:tid 1976718384] [client 10.0.1.3:57314] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Wed Jan 03 19:08:05.992333 2018] [authz_host:error] [pid 948:tid 1968329776] [client 10.0.1.3:57314] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Wed Jan 03 19:08:06.193244 2018] [authz_host:error] [pid 947:tid 1976718384] [client 10.0.1.3:57316] AH01753: access check of 'localhost' to /apple-touch-icon-precomposed.png failed, reason: unable to get the remote host name
[Wed Jan 03 19:08:06.273410 2018] [authz_host:error] [pid 947:tid 1968329776] [client 10.0.1.3:57316] AH01753: access check of 'localhost' to /apple-touch-icon.png failed, reason: unable to get the remote host name
[Wed Jan 03 19:08:06.278608 2018] [authz_host:error] [pid 947:tid 1959941168] [client 10.0.1.3:57316] AH01753: access check of 'localhost' to /favicon.ico failed, reason: unable to get the remote host name
[... 28 lines cut ...]
2018-01-03 19:08:11 1988460544 [Note] InnoDB: Using generic crc32 instructions
2018-01-03 19:08:11 1988460544 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2018-01-03 19:08:11 1988460544 [Note] InnoDB: Completed initialization of buffer pool
2018-01-03 19:08:12 1988460544 [Note] InnoDB: Highest supported file format is Barracuda.
2018-01-03 19:08:12 1988460544 [Note] InnoDB: 128 rollback segment(s) are active.
2018-01-03 19:08:12 1988460544 [Note] InnoDB: Waiting for purge to start
2018-01-03 19:08:12 1988460544 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 280614183
2018-01-03 19:08:13 1438643008 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-01-03 19:08:13 1988460544 [Note] Plugin 'FEEDBACK' is disabled.
2018-01-03 19:08:14 1988460544 [Note] Server socket created on IP: '127.0.0.1'.
2018-01-03 19:08:14 1988460544 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/USBdrive/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

nachoparker, 2018-01-04:
thanks, what I am interested in is the output right after reboot of

```
systemctl status nc-automount-links
```
Also, you can try if the above solution works for you

BETEP-ok, 2018-01-04:
hi, again, @nachoparker 

soo... I did a reboot, and the output of systemctl status nc-automount-links:

**systemctl status nc-automount-links**
● nc-automount-links.service - Monitor /media for mountpoints and create USBdriv
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled;
   Active: active (running) since Thu 2018-01-04 19:40:25 UTC; 47s ago
 Main PID: 352 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─352 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─388 inotifywait --monitor --event create --event delete --format %f
           ├─389 grep --line-buffered ISDIR
           └─390 /bin/bash /usr/local/etc/nc-automount-links-mon

Jan 04 19:40:25 nextcloudpi systemd[1]: Started Monitor /media for mountpoints a
Jan 04 19:40:26 nextcloudpi nc-automount-links-mon[352]: Setting up watches.
Jan 04 19:40:26 nextcloudpi nc-automount-links-mon[352]: Watches established.
Jan 04 19:40:29 nextcloudpi nc-automount-links-mon[352]: myCloudDrive DELETE,ISD
Jan 04 19:40:38 nextcloudpi nc-automount-links-mon[352]: myCloudDrive CREATE,ISD
lines 1-15/15 (END)

now I do the proposed magic: 
`sudo /usr/local/etc/nc-automount-links
sudo systemctl disable nc-automount-links
sudo ln -s /media/myCloudDrive /media/USBdrive `

and trigger again the **systemctl status nc-automount-links**
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-01-04 19:40:25 UTC; 2min 27s ago
 Main PID: 352 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─352 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─388 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─389 grep --line-buffered ISDIR
           └─390 /bin/bash /usr/local/etc/nc-automount-links-mon

Jan 04 19:40:25 nextcloudpi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Jan 04 19:40:26 nextcloudpi nc-automount-links-mon[352]: Setting up watches.
Jan 04 19:40:26 nextcloudpi nc-automount-links-mon[352]: Watches established.
Jan 04 19:40:29 nextcloudpi nc-automount-links-mon[352]: myCloudDrive DELETE,ISDIR
Jan 04 19:40:38 nextcloudpi nc-automount-links-mon[352]: myCloudDrive CREATE,ISDIR

now the symlink in /Media is back again.

media $ ls
myCloudDrive  USBdrive

now again - reboot.....

aaand... symlink stays in /media:

/media $ ls
myCloudDrive  USBdrive

Sweet! 
your kung-fu is better then mine )
but still don§t get it why this problem happens at all with a fresh installation. )

nachoparker, 2018-01-05:
> but still don`t get it why this problem happens at all with a fresh installation

Me neither, because we can se the detection working 

> Jan 04 19:40:38 nextcloudpi nc-automount-links-mon[352]: myCloudDrive CREATE,ISDIR

At least the workaround works for you...

cp2732, 2020-06-21:
Hello, I am getting a similar issue when trying to use an already formatted USB drive using the btrfs filesystem. I originally formatted this USB drive using a different NextCloud instance on a separate 4GB SD card. After I upgraded to a new 64GB SD card using NextCloud, I decided to compare my old NextCloud configuration with the new one and keep the configurations I desired to keep. I did this by plugging in a SD to USB thumb drive, since the Raspberry Pi does not have another SD slot. For this, I manually mounted this thumb drive using
`sudo mkdir -p /media/old
sudo mount -t ext4 -o defaults /dev/sda1 /media/old`
and unmounted with
`sudo umount /media/old`

Since I also had a bunch of data on my USB drive, I tried to plug that into the Raspberry Pi with the new SD card, but cannot get the drive to automount, even automount was enabled though the web-ui.

Here's what I got using `sudo lsblk --fs`:
```
NAME        FSTYPE LABEL        UUID                                 FSAVAIL FSUSE% MOUNTPOINT
sda                                                                                 
└─sda1      btrfs  myCloudDrive 4b5da605-a5b6-4d06-a62f-e2cf9e543662                
mmcblk0                                                                             
├─mmcblk0p1 vfat   boot         4BBD-D3E7                               199M    21% /boot
└─mmcblk0p2 ext4   rootfs       45e99191-771b-4e12-a526-0779148892cb   53.8G     4% /
```

Before I ran the three commands suggested above
```
sudo /usr/local/etc/nc-automount-links
sudo systemctl disable nc-automount-links
sudo ln -s /media/myCloudDrive /media/USBdrive
```

I got:
```systemctl status nc-automount-links
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-06-21 11:03:29 EDT; 1min 35s ago
 Main PID: 340 (nc-automount-li)
    Tasks: 4 (limit: 4915)
   Memory: 2.1M
   CGroup: /system.slice/nc-automount-links.service
           ├─340 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─353 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─354 grep --line-buffered ISDIR
           └─355 /bin/bash /usr/local/etc/nc-automount-links-mon

Jun 21 11:03:29 top-notch-ncp systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Jun 21 11:03:29 top-notch-ncp nc-automount-links-mon[340]: Setting up watches.
Jun 21 11:03:29 top-notch-ncp nc-automount-links-mon[340]: Watches established.
Jun 21 11:03:42 top-notch-ncp nc-automount-links-mon[340]: myCloudDrive CREATE,ISDIR
Jun 21 11:03:42 top-notch-ncp nc-automount-links-mon[340]: myCloudDrive DELETE,ISDIR
```

After running those three commands and rebooting, I now get
```
systemctl status nc-automount-links
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
```

I am using NextCloud version 18.0.6 and NextCloudPi version 1.26.2. I have not changed the data directory yet from the default, but when I try to run nc-datadir in the UI wizard, I get
`/media/USBdrive does not exist`

I would be grateful if there is a way that exists, where I do not need to reformat the USB drive.

weiwang11, 2020-06-25:
This looks similar to https://github.com/nextcloud/nextcloudpi/issues/1084

Try following this [post](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931) and see if that helps. 

> The most common symptoms of a misbehaving UAS device are
> - Extremely slow performance - in the kilobytes per second range
> - **Frequent disconnects-reconnects of the device with the desktop repeatedly displaying the "removable medium inserted" dialogue box**
> - The kernel message log (dmesg) reports errors relating to a UAS device that look like this: 
```

[ 501.594683] usbcore: registered new interface driver usb-storage
[ 501.599729] scsi host6: uas
[ 501.599800] usbcore: registered new interface driver uas
...
[ 573.203294] sd 6:0:0:0: [sda] tag#29 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD OUT
[ 573.203302] sd 6:0:0:0: [sda] tag#29 CDB: Write(10) 2a 00 00 4f a0 00 00 04 00 00
[ 573.205063] sd 6:0:0:0: [sda] tag#28 uas_eh_abort_handler 0 uas-tag 10 inflight: CMD OUT
[ 573.205070] sd 6:0:0:0: [sda] tag#28 CDB: Write(10) 2a 00 00 4f a4 00 00 04 00 00
[ 573.208537] sd 6:0:0:0: [sda] tag#27 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD OUT
...
[ 573.269992] scsi host6: uas_eh_device_reset_handler start
[ 573.393710] usb 2-4: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[ 573.414256] scsi host6: uas_eh_device_reset_handler success

```