nextcloud/nextcloudpi #335: No Access Again!

deljones, 2017-12-27:
Hi all..

I dint tinker with my NC instance, if its running I leave it and use it. However there has been an auto update, again, and now I have no access...

Here is the output of:  systemctl status nc-automount-links

NextCloudPi v0.43.1 is up to date
pi@raspberrypi:~ $ systemctl status nc-automount-links
● nc-automount-links.service - Monitor /media for mountpoints and create USBdriv
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled;
   Active: active (running) since Tue 2017-12-26 23:59:49 UTC; 11h ago
 Main PID: 436 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─436 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─464 inotifywait --monitor --event create --event delete --format %f
           ├─465 grep --line-buffered ISDIR
           └─466 /bin/bash /usr/local/etc/nc-automount-links-mon

Dec 26 23:59:49 raspberrypi systemd[1]: Started Monitor /media for mountpoints a
Dec 26 23:59:49 raspberrypi nc-automount-links-mon[436]: Setting up watches.
Dec 26 23:59:49 raspberrypi nc-automount-links-mon[436]: Watches established.
Dec 26 23:59:54 raspberrypi nc-automount-links-mon[436]: ncbackup1 DELETE,ISDIR
Dec 26 23:59:54 raspberrypi nc-automount-links-mon[436]: myCloudDrive1 DELETE,IS
Dec 26 23:59:55 raspberrypi nc-automount-links-mon[436]: myCloudDrive1 CREATE,IS
Dec 26 23:59:55 raspberrypi nc-automount-links-mon[436]: ncbackup1 CREATE,ISDIR

I can see all the diretories on my HDD on USB
sudo ncp-report give the following:

NextCloudPi v0.43.1 is up to date
pi@raspberrypi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.43.1
NextCloudPi image    NextCloudPi_09-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda sdb
datadir              /media/USBdrive/ncdata (doesn't exist)
rootfs usage         1.9G/2.7G
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      down
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.0.11
gateway              192.168.0.1
interface            eth0
certificates         digitalcarnage.webhop.me
certs due            69 days
NAT loopback         no
uptime               1:06

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

[... 90 lines cut ...]
<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/USBdrive/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

It reports that it can't see ncdata but I have no idea how this has happend. I can SSH into the pi, run ncp on SSH as well as ncp-web.

Can anyone help?

Del

ovpc, 2017-12-27:
Thanks for reporting.
Sorry to hear you're having issues.
My NCP is also up to date v0.43.1, but not having issues at all.
Nacho usually fixes this kind of issue really fast.

nachoparker, 2017-12-27:
Hi, 

There hasn't been an update for over a week now. 

What I see is that you have two USB drives, and you use the link name `/media/USBdrive` for ncdatadir and probably also for nc-database.

If you have two drives, it is _random_ which drive will take `/media/USBdrive` and which one `/media/USBdrive1`. The idea of using `/media/USBdrive` is for a basic setup with one HDD for people that don't know their drives' labels. My guess is that this is what's happening. Did it stop working after reboot?

So that is one thing I would improve

- turn the pi off
- unplug the backup drive
- turn on again, it should be working again
- in ncp-web, re-run nc-datadir and nc-database (and anything else where it shows USBdrive), and replace `USBdrive` for `myCloudDrive1`
- if everything is working now, you can plug in the second USB drive.

We can also see (before doing all that) what happened to the database

```
systemctl status mysqld
```

Some people have reported that the database can go down if they move it to the drive. For most people it works fine, but if this happens again, you can consider moving it back to the SD

Hope it helps.

deljones, 2017-12-29:
Hi

The update(s) is not the nextcloudpi update. If I log in as admin I get a notice that there has been unattended updates which, I have now turned off. These updates are .libc files and such like. Everything was ok until one day I was notified of an unattended update had taken place. I rebooted and that was it... Nothing!

I am back up again all is well... I just made another SD image, imported my ncp-config. I have unplugged the second drive that was acting as my backup which I will get back on again soon.

what I have noticed is that the NextCloud login takes an age to get logged in... Much much much longer than I remember. Once in its fine. Is there anywhere I can look to see why it takes so long to log in?

Otherwise all good. Again

Many thanks... Happy New Year

Del

nachoparker, 2018-01-02:
Well, thankfully it is getting easier to migrate to a new instance with nc-import-ncp and nc-backup ;)

It sounds weird to me that an unattended upgrade messed up automount in any way. Still, i recommend that you use label names instead of `/media/USBdrive`

Always it takes a long time to login the first time, and then after that the PHP code is cached and it is much faster.

Well, hopefully it won't happen again! If it happens, tell us so we can discover what happened. If it happens to anyone else please reopen.

Happy New year

closed 2018-01-02 by nachoparker