nextcloud/nextcloudpi #336: MariaDB issue at installation

leolivier, 2017-12-27:
Hi, 
I'm trying to install ncp upon an existing nextcloud installation (using MariaDB)
When I run the install script 
`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash`

The first part runs fine but when it comes to MariaDB, I get these errors and the ioctl error runs forever...

```
NOTE: RUNNING ALL PARTS OF THIS SCRIPT IS RECOMMENDED FOR ALL MariaDB
      SERVERS IN PRODUCTION USE!  PLEASE READ EACH STEP CAREFULLY!

In order to log into MariaDB to secure it, we'll need the current
password for the root user.  If you've just installed MariaDB, and
you haven't set the root password yet, the password will be blank,
so you should just press enter here.

stty: 'entrée standard': Ioctl() inapproprié pour un périphérique
Enter current password for root (enter for none): 
stty: 'entrée standard': Ioctl() inapproprié pour un périphérique
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)
stty: 'entrée standard': Ioctl() inapproprié pour un périphérique
Enter current password for root (enter for none): 
stty: 'entrée standard': Ioctl() inapproprié pour un périphérique
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)
```

So, is it possible to do this? Should I change the root password somewhere?
I looked at the lamp.sh file where I think the problem lies and it seems that the password for MariaDB is created randomly whether MariaDB is already installed or not...

I tried to run the installer.sh script in interactive mode by myself with 
`./installer.sh lamp.sh 192.168.1.10`

but it simply says "installation failed" 
I added 'set -ex' at the beginning of the script and I get (after a bunch of lines):
```
+ type sshpass
+ [[ '' == '' ]]
+ ssh -q -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -o ServerAliveInterval=20 -o ConnectTimeout=20 -o LogLevel=quiet pi@192.168.1.10
+ echo 'Installation to 192.168.1.10 failed'
Installation to 192.168.1.10 failed
+ return 1

```

nachoparker, 2017-12-28:
Hi,

The script is thought to be installed on a clean system, so no LAMP. You probably could see this warning at the top of the script execution

```
>>> WARNING: existing mysqld configuration will be changed <<<
```

 If you want to install it on an already running system, you are "on your own", but if you want to help make this happen, we can discuss it here and if you want can send a PR ;)

So the issue is that the database already exists and the script doesn't have the password. The script should store the password in `/etc/.my.cnf`. I would reset your current database password to that one, and then try again. 

Keep in mind that your current Nextcloud instance will be completely wiped out

leolivier, 2017-12-28:
Thanks @nachoparker, I think I will give up for now.
I tried this because I losed a disk and my nextcloud wasn't working anymore. Now, I managed to get it back so I don't need that anymore.
I used ncp on another rasp for a gift to my daughter and I found it very convenient for the admin part so I wanted to try it on my existing install.
I think it's worth thinking to this use case but I don't know if many people are in the same case...

closed 2017-12-28 by leolivier