nextcloud/nextcloudpi #337: Seems like a racecondition on mounting usb drive vs. generation of swap file

sebastianmonzel, 2017-12-28:
Hi there,

seems for me like a racecondition between mounting usb drive and generation of swap file. actual log - failing one:

Dec 28 18:31:28 nextcloudpi systemd[1]: Starting LSB: Autogenerate and use a swap file...
Dec 28 18:31:28 nextcloudpi systemd-udevd[200]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Dec 28 18:31:28 nextcloudpi systemd[1]: Started Regular background program processing daemon.
Dec 28 18:31:28 nextcloudpi systemd[1]: Started Automount USB drives.
Dec 28 18:31:28 nextcloudpi thd[319]: Found socket passed from systemd

if the usb drive wins everthing is fine - swapfile is placed on usbdrive. But if the generation of the swapfile wins the usbdrive is not placed in /media/pimaster - it's placed in /media/pimaster1 - and from now on the webserver does not work because the nextcloud datafolder is placed on the wrong location.

otherwise there exists a second location /media/pimaster/ on the sdcard with the swapfile.

sebastianmonzel, 2017-12-28:
My workaround is actually to mount swapfile directly in root folder: /swapfile

nachoparker, 2017-12-29:
hi

thanks for reporting.

well... this has always been a bit... dirty. Originally, a `sleep 15` wait was introduced that seemed to be enough in my system. I guess it could be increased... or maybe we should do it right and actively check for the existence of the mountpoint

https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-swapfile.sh#L41

nachoparker, 2018-03-03:
I reviewed the dependencies for automount. I think it should be fixed in `v0.46.37`, but I cannot verify because this issue doesn't happen to me.

Please, run

```
sudo ncp-update devel
```

, and verify. 

We will reopen if the issue persists.

Thanks for the feedback!

closed 2018-03-03 by nachoparker