nextcloud/nextcloudpi #338: Issue getting fail2ban to work

apow98, 2017-12-29:
I'm having difficulty getting fail2ban to work properly with nextcloud. The counters remain at zero no matter how many failed attempts, however the ssh jail does work properly. I stumbled upon this thread https://github.com/nextcloud/nextcloudpi/issues/116 but none of the potential solutions arised there helped.

nachoparker, 2018-01-02:
hi, there was a fix recently in `v0.41.8` ( [link](https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md) )

try running fail2ban again (even if it says it's active), and if it still fails paste the output of 

```
sudo ncp-report
```
and

```
sudo fail2ban-client status nextcloud
```

apow98, 2018-01-02:
ya still nothing so here's the two outputs. 

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.2
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      108G/932G
rootfs usage         1.8G/30G
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.194
gateway              192.168.1.1
interface            wlan0
certificates         apow.ddns.net
certs due            83 days
NAT loopback         no
uptime               1:16

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.194",
            "2": "192.168.1.193",
            "3": "apow.ddns.net",
            "4": "apow.ddns.net"
        },
        "datadirectory": "\/media\/USBdrive\/ncdata",
        "overwrite.cli.url": "https:\/\/apow.ddns.net",
        "dbtype": "mysql",
        "version": "12.0.4.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "ocs86rkb5syt",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
[... 73 lines cut ...]
{"reqId":"Wkt6Jn8AAQEAACTI4QsAAAAI","level":2,"time":"2018-01-02T12:25:10+00:00"                                                                                          ,"remoteAddr":"184.105.139.70","user":"--","app":"core","method":"GET","url":"\/                                                                                          ","message":"Trusted domain error. \"184.105.139.70\" tried to access using \"71                                                                                          .115.211.102\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WkuFK38AAQEAACTI4QwAAAAL","level":2,"time":"2018-01-02T13:12:11+00:00"                                                                                          ,"remoteAddr":"164.132.91.13","user":"--","app":"core","method":"GET","url":"\/"                                                                                          ,"message":"Trusted domain error. \"164.132.91.13\" tried to access using \"71.1                                                                                          15.211.102\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_1                                                                                          0_1) AppleWebKit\/537.36","version":"12.0.4.3"}
{"reqId":"WkuNZX8AAQEAACTJ4ooAAABF","level":2,"time":"2018-01-02T13:47:17+00:00"                                                                                          ,"remoteAddr":"54.153.36.222","user":"--","app":"core","method":"GET","url":"\/"                                                                                          ,"message":"Trusted domain error. \"54.153.36.222\" tried to access using \"71.1                                                                                          15.211.102:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X                                                                                           10_11_5) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/50.0.2661.102 Safari\/5                                                                                          37.36","version":"12.0.4.3"}
{"reqId":"WkvPM38AAQEAACTJ4vYAAE0F","level":2,"time":"2018-01-02T18:28:04+00:00"                                                                                          ,"remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/in                                                                                          dex.php\/login","message":"Login failed: 'admin' (Remote IP: '192.168.1.1')","us                                                                                          erAgent":"Mozilla\/5.0 (Linux; Android 8.1.0; Pixel XL Build\/OPM1.171019.011) A                                                                                          ppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.111 Mobile Safari\/537.                                                                                          36","version":"12.0.4.3"}
{"reqId":"WkvPS38AAQEAACTI4R8AABAQ","level":2,"time":"2018-01-02T18:28:27+00:00"                                                                                          ,"remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/in                                                                                          dex.php\/login?user=admin","message":"Login failed: 'admin' (Remote IP: '192.168                                                                             [... cut ...]                                                 ppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.111 Mobile Safari\/537.                                                                                          36","version":"12.0.4.3"}

```
</details>

Status for the jail: nextcloud
|- Filter
|  |- Currently failed: 0
|  |- Total failed:     0
|  `- File list:        /var/www/nextcloud/data/nextcloud.log
`- Actions
   |- Currently banned: 0
   |- Total banned:     0
   `- Banned IP list:

nachoparker, 2018-01-03:
your problem is that the fail2ban jail looks into this log file

```
/var/www/nextcloud/data/nextcloud.log
```
, whereas your log is most likely

```
/media/USBdrive/ncdata/nextcloud.log
```

, because you moved with nc-datadir. 

It should be updated when running fail2ban again from ncp-web or `nextcloudpi-config`

Like I said, this has already been fixed, so I'll close, but we can still talk.

closed 2018-01-03 by nachoparker