nextcloud/nextcloudpi #340: fail2ban blocking me out

deljones, 2018-01-02:
My Nextcloudpi setup has the UFW active, I tried to activate the fail2ban which activated ok but within 24 hours or less I found that my router 192.168.0.1 was blocked by fail2ban and I had a number of emails telling me. Have I set this up right? Everything has to go through the router so how do successive unwanted IPs get through. Also do I need both UFW and faile2ban?

D

ovpc, 2018-01-02:
It is possible to [white-list](https://www.fail2ban.org/wiki/index.php/Whitelist) 192.168.0.1 and any other IP/range you want, so they wont be banned.
If NCP developers set up UFW and fail2ban, you can assume its advisable to run both if you are running a system with access from www. I havent set up fail2ban yet, but was running UFW before it was added to NCP. It depends on how sensitive your data is and how many users will be using the system.

nachoparker, 2018-01-02:
your gateway was blocked by fail2ban? because of SSH access, or because of Nextcloud access?

why would your router try to log in through SSH? that smells bad!

maybe the email has more information, or you can do the 

```
sudo fail2ban-client status nextcloud
```

and [similar commands](https://ownyourbits.com/2017/02/24/nextcloudpi-fail2ban-installer/) to find out.

deljones, 2018-01-05:
Hello and sorry for the delay...

I reactivated fail2ban yesterday and everything was fine again. However today I added a new laptop with a local ip of 192.168.0.16 and within minutes I got a fail2ban alert saying that the address of my router was blocked, 192.168.0.1

I deactivated fail2ban again.

I do need to see it happen again to see if it's just the router, the first thing I did was panic and get rid of fail2ban. I'll do a more methodical approach to the error next time... 

I have attached the email to see if there is any kind of story there. I do see a 192.168.0.0/16 which is the address of the laptop that started the second block 

--------

Hi,

The IP 192.168.0.1 has just been banned by NextCloudPi after
6 attempts against nextcloud.

Here is more information about 192.168.0.1 :

#
# ARIN WHOIS data and services are subject to the Terms of Use
# available at: https://www.arin.net/whois_tou.html
#
# If you see inaccuracies in the results, please report at
# https://www.arin.net/public/whoisinaccuracy/index.xhtml
#

#
# The following results may also be obtained via:
# https://whois.arin.net/rest/nets;q=192.168.0.1?showDetails=true&showARIN=false&showNonArinTopLevelNet=false&ext=netref2
#

NetRange:       192.168.0.0 - 192.168.255.255
CIDR:           192.168.0.0/16
NetName:        PRIVATE-ADDRESS-CBLK-RFC1918-IANA-RESERVED
NetHandle:      NET-192-168-0-0-1
Parent:         NET192 (NET-192-0-0-0-0)
NetType:        IANA Special Use
OriginAS:       
Organization:   Internet Assigned Numbers Authority (IANA)
RegDate:        1994-03-15
Updated:        2013-08-30
Comment:        These addresses are in use by many millions of independently operated networks, which might be as small as a single computer connected to a home gateway, and are automatically configured in hundreds of millions of devices.  They are only intended for use within a private context  and traffic that needs to cross the Internet will need to use a different, unique address.
Comment:        
Comment:        These addresses can be used by anyone without any need to coordinate with IANA or an Internet registry.  The traffic from these addresses does not come from ICANN or IANA.  We are not the source of activity you may see on logs or in e-mail records.  Please refer to http://www.iana.org/abuse/answers
Comment:        
Comment:        These addresses were assigned by the IETF, the organization that develops Internet protocols, in the Best Current Practice document, RFC 1918 which can be found at:
Comment:        http://datatracker.ietf.org/doc/rfc1918
Ref:            https://whois.arin.net/rest/net/NET-192-168-0-0-1

OrgName:        Internet Assigned Numbers Authority
OrgId:          IANA
Address:        12025 Waterfront Drive
Address:        Suite 300
City:           Los Angeles
StateProv:      CA
PostalCode:     90292
Country:        US
RegDate:        
Updated:        2012-08-31
Ref:            https://whois.arin.net/rest/org/IANA

OrgAbuseHandle: IANA-IP-ARIN
OrgAbuseName:   ICANN
OrgAbusePhone:  +1-310-301-5820 
OrgAbuseEmail:  abuse@iana.org
OrgAbuseRef:    https://whois.arin.net/rest/poc/IANA-IP-ARIN

OrgTechHandle: IANA-IP-ARIN
OrgTechName:   ICANN
OrgTechPhone:  +1-310-301-5820 
OrgTechEmail:  abuse@iana.org
OrgTechRef:    https://whois.arin.net/rest/poc/IANA-IP-ARIN

#
# ARIN WHOIS data and services are subject to the Terms of Use
# available at: https://www.arin.net/whois_tou.html
#
# If you see inaccuracies in the results, please report at
# https://www.arin.net/public/whoisinaccuracy/index.xhtml
#

Lines containing IP:192.168.0.1 in /dev/null

Regards,

NextCloudPi

deljones, 2018-01-05:
I'll give that a go... I assume logging using ssh? Or both NC and SSH?

deljones, 2018-01-05:
OK.. Just used a machine and did multipule SSH wroing passwords and sure enough I was banned.

pi@nextcloudpi:~ $ sudo fail2ban-client status nextcloud
Status for the jail: nextcloud
|- Filter
|  |- Currently failed: 0
|  |- Total failed:     2
|  `- File list:        /media/USBdrive/ncdata/nextcloud.log
`- Actions
   |- Currently banned: 0
   |- Total banned:     0
   `- Banned IP list:

I used the command to unban the offending machine but got...

pi@nextcloudpi:~ $ sudo fail2ban-client set nextcloud unbanip 192.168.0.16
ERROR  NOK: ('IP 192.168.0.16 is not banned',)
IP 192.168.0.16 is not banned

I'm going ot deactivate faile2ban... Not sure I fully understand it. Requires a bit more readinf on my part

Thanks again

DJ

nachoparker, 2018-01-05:
I meant on nextcloud. if you want to check what IP was banned on SSH you can use

```
sudo fail2ban-client status ssh
```

Do this while you have an SSH session open already, because once banned you won't be able to issue commands

deljones, 2018-01-05:
Ok just done 6 failed attempts on a machine trying to log into NC and was banned but the banning is showing the router address and not the offending machine

Hi,

The IP 192.168.0.1 has just been banned by NextCloudPi after
6 attempts against nextcloud.

Here is more information about 192.168.0.1 :

#
# ARIN WHOIS data and services are subject to the Terms of Use
# available at: https://www.arin.net/whois_tou.html
#
# If you see inaccuracies in the results, please report at
# https://www.arin.net/public/whoisinaccuracy/index.xhtml
#

#
# The following results may also be obtained via:
# https://whois.arin.net/rest/nets;q=192.168.0.1?showDetails=true&showARIN=false&showNonArinTopLevelNet=false&ext=netref2
#

NetRange:       192.168.0.0 - 192.168.255.255
CIDR:           192.168.0.0/16
NetName:        PRIVATE-ADDRESS-CBLK-RFC1918-IANA-RESERVED
NetHandle:      NET-192-168-0-0-1
Parent:         NET192 (NET-192-0-0-0-0)
NetType:        IANA Special Use
OriginAS:       
Organization:   Internet Assigned Numbers Authority (IANA)
RegDate:        1994-03-15
Updated:        2013-08-30
Comment:        These addresses are in use by many millions of independently operated networks, which might be as small as a single computer connected to a home gateway, and are automatically configured in hundreds of millions of devices.  They are only intended for use within a private context  and traffic that needs to cross the Internet will need to use a different, unique address.
Comment:        
Comment:        These addresses can be used by anyone without any need to coordinate with IANA or an Internet registry.  The traffic from these addresses does not come from ICANN or IANA.  We are not the source of activity you may see on logs or in e-mail records.  Please refer to http://www.iana.org/abuse/answers
Comment:        
Comment:        These addresses were assigned by the IETF, the organization that develops Internet protocols, in the Best Current Practice document, RFC 1918 which can be found at:
Comment:        http://datatracker.ietf.org/doc/rfc1918
Ref:            https://whois.arin.net/rest/net/NET-192-168-0-0-1

OrgName:        Internet Assigned Numbers Authority
OrgId:          IANA
Address:        12025 Waterfront Drive
Address:        Suite 300
City:           Los Angeles
StateProv:      CA
PostalCode:     90292
Country:        US
RegDate:        
Updated:        2012-08-31
Ref:            https://whois.arin.net/rest/org/IANA

OrgAbuseHandle: IANA-IP-ARIN
OrgAbuseName:   ICANN
OrgAbusePhone:  +1-310-301-5820 
OrgAbuseEmail:  abuse@iana.org
OrgAbuseRef:    https://whois.arin.net/rest/poc/IANA-IP-ARIN

OrgTechHandle: IANA-IP-ARIN
OrgTechName:   ICANN
OrgTechPhone:  +1-310-301-5820 
OrgTechEmail:  abuse@iana.org
OrgTechRef:    https://whois.arin.net/rest/poc/IANA-IP-ARIN

#
# ARIN WHOIS data and services are subject to the Terms of Use
# available at: https://www.arin.net/whois_tou.html
#
# If you see inaccuracies in the results, please report at
# https://www.arin.net/public/whoisinaccuracy/index.xhtml
#

Lines containing IP:192.168.0.1 in /dev/null

Regards,

NextCloudPi

DJ

deljones, 2018-01-05:
UPDATE:

So the ban is showing that the router address is banned but I can still access to my NC instance outside of my network with my usual credentials no problem.. After a while though I was able to log back in on the offending machine without unbanning it...
The banned IP list is empty?

pi@nextcloudpi:~ $ sudo fail2ban-client status nextcloud
Status for the jail: nextcloud
|- Filter
|  |- Currently failed:	2
|  |- Total failed:	10
|  `- File list:	/media/USBdrive/ncdata/nextcloud.log
`- Actions
   |- Currently banned:	0
   |- Total banned:	1
   `- Banned IP list:	

DJ

nachoparker, 2018-01-06:
ok, so it seems it just always uses your gateway for the email? that's weird, but ok

it's normal that after a while you are able to log in again, that's the way it works

you should first log in through SSH, then stay logged in and do the wrong password a few times until it blocks you. Then, from the SSH session you can see the status of the nextcloud or ssh jail and you should see the banned ip.

deljones, 2018-01-06:
Fantastic … Thanks very much.. Again

D

> ok, so it seems it just always uses your gateway for the email? that's weird, but ok
>
> it's normal that after a while you are able to log in again, that's the way it works
>
> you should first log in [...]

Derek Jones Stage and Television Lighting Design <http://derekjones.net/>

w: www.derekjones.net <http://derekjones.net/>
t: +44 7957 140 512
 
Your Data. Your Way. Safe... <https://digitalcarnage.webhop.me/>
 
The information in this E-Mail is confidential and may be legally privileged. 
It may not represent the views of Derek Jones. 
It is intended solely for the addressees. 
Access to this E-Mail by anyone else is unauthorised. 
If you are not the intended recipient, any disclosure, copying, distribution or any action taken or omitted to be taken in reliance on it, is prohibited and may be unlawful. 
Any unauthorised recipient should advise the sender immediately of the error in transmission.

closed 2018-01-13 by nachoparker