nextcloud/nextcloudpi #342: web user and password and port 80 trouble

rubensx, 2018-01-03:
great work! it's a only cloud I've can use in raspberry, great!!!

some problems here:

1- port 80 non SSL goes to blank page, and my dynamic IP and ISP block ports are blocking me to access from internet outside my cloud in my lan (yes, I've redirect ports but no lucky, please can you help to use without SSL?)

2- web user is not admin - ownyourbits or other else, please help!

ovpc, 2018-01-04:
Not very clear what you are asking or stating .....
- port 80 goes to blank page (not set up correctly, clearly)
- my dynamic IP and ISP block ports. There are ways to work around dynamic IP, Tried this [wiki](https://github.com/nextcloud/nextcloudpi/wiki/How-to-access-from-outside-your-network
) yet? Might solve the next one too....
- ISP reserved ports and port forwarding (can be tricky, depending software on ISP/router, what is yours? Some models you need to create rules first, then add them to NAT/port-forward)
- using NC without SSL is not wise/recommended unless its in LAN only.
- web user is not admin (if you mean [ncp-config web user](https://github.com/nextcloud/nextcloudpi/wiki/How-to-configure-NextCloudPi) is 'ncp')

If possible; make screenshots or pictures of errors and output and include them when reporting, it makes it easier for others to help? Thank you for reporting.

rubensx, 2018-01-04:
Oh thank you for your help:

About the ISP and my router:
Yes, not a problem at now, it’s a mikrotik cloud router and with a simple NAT rule, I’m redirec the ports:

No-IP goes from port 80 to 8080 and in my mikrotik 8080 translated to 80 again.

But, trying to convert the same operation in 443 to 8081 and in mikrotik 8081 to 443 again, the error message from apache is:

“You are trying to view a https page with a http document”

Means like a NO-IP don’t translate a ssl page, and trying use https:// before the link but still same result.

Another rule in my NAT, for a test are a DMZ and still no lucky...

First problems is from a “trusted domains” - the nceans my isp dynamic ip are not a trusted domain (and to a test I put the IP in config.php, solve the problem at the moment but, when It’s change, I need change too... another problem).

About secure, yes, port 80 is not a better way, but with the routes and this issues, I’m only testing to see if work, and still can’t see the nc working from internet at all...

When I not receive a apache message https/http problem, receive a untrusted domain log message... it’s frustrating and useless, either a iOS app cannot use too...

Maybe only for test has a button to disable trusted domains or simple work in 80 port (in LAN to test only) but not work!!

rubensx, 2018-01-04:
About “ncp” user, in documment or wiki it’s confusing and not clear! I’ll trying this user and password, maybe it’s make my life more easy.

Thank you again

nachoparker, 2018-01-13:
I'm simplifying the user names, everything will be ncp / ownyourbits by default

thanks for your feedback

closed 2018-01-13 by nachoparker