nextcloud/nextcloudpi #343: Collabora online in nextcloud pi

Ricardosgeral, 2018-01-04:
Did anyone install successfully collabora online in nextcloud using the RP 3 as server? 

I can't do it

Paste the output of `sudo ncp-report`

ovpc, 2018-01-04:
Noop
Collabora server runs on x86
not on ARM
Collabora app in NCP works fine tho
Why would you want to run Collabora server on RPI?

Ricardosgeral, 2018-01-04:
I want collabora to open and edit MS word files in the server that is a
raspberry pi 3. Is this stupid?

nachoparker, 2018-01-04:
Duplicate of #51

closed 2018-01-04 by nachoparker

ovpc, 2018-01-04:
The only stupid question is the one not asked ;-)
but Imho its just not worth the trouble, it's quit an accomplishment to make a RPI run a usable cloud as it is. I've Collabora app working in NextCloudPi, with a docker Collabora/code server running in VM on PC, but do not use it, it just too slow in loading and interface not nice and limited functions. Read Collabora limits number of open-documents/active-users to 10/20 in free version, which is another reason for me not to like it, even if I am the only user.