nextcloud/nextcloudpi #344: PHP Startup: Unable to load dynamic library '/usr/lib/php/20151012/gd.so' - /usr/lib/arm-linux-gnueabihf/libgd.so.3

MatthiasPaul, 2018-01-04:
Hi,
the following is mainly describing my issue

Paste the output of `sudo ncp-report`:
PHP Warning:  PHP Startup: Unable to load dynamic library '/usr/lib/php/20151012/gd.so' - /usr/lib/arm-linux-gnueabihf/libgd.so.3: undefined symbol: � in Unknown on line 0
Nextcloud is in maintenance mode - no app have been loaded

I am running NExtCloudPi for several months now. Since January, 3rd 2018 I can not log into my NextCloud instance any more. I get a very similar notice when I access my NextCloud instance via a web browser. I am running NextCloudPi on my Pi3, version 0.44.2.

In the meantime I have purged and reinstalled php7.0-gd, started phpenmod gd and restarted the whole system, apache2, php, .... 
I did: php -i | grep ini and can see that the 

I am stuck right now. Has anyone any idea or suggestion? Thanks a lot!
If you need more information from me, please let me know.

MatthiasPaul, 2018-01-04:
In addition to the info mentioned above the NextCloudPi Info System gave the following result:

PHP Warning:  PHP Startup: Unable to load dynamic library '/usr/lib/php/20151012/gd.so' - /usr/lib/arm-linux-gnueabihf/libgd.so.3: undefined symbol: � in Unknown on line 0
Nextcloud is in maintenance mode - no app have been loaded

Gathering information...
NextCloudPi version  v0.44.2
NextCloudPi image    NextCloudPi_10-29-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/datadir
data in SD           no
data filesystem      ext2/ext3
data disk usage      12G/916G
rootfs usage         1.7G/15G
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.xxxxx
gateway              192.xxxxx
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               7:48

You should enable dnsmasq to use your domain inside home

You should run Lets Encrypt for trusted encrypted access

nachoparker, 2018-01-05:
mmm suspect a corrupt SD card? did you have a power cut? look in the wiki FAQ how to check your SD card

MatthiasPaul, 2018-01-05:
Thanks for the reply. I will check the SD card. I have not thought about this possibility but its worth trying.

MatthiasPaul, 2018-01-05:
I followed the wiki FAQ but could not find an issue with the SD card. 

I restarted the NextCloudPi instance from scratch and ran through the backup-restore process. All worked very well and my NextCloud instance is running again. Thanks for this great product!!

closed 2018-01-05 by MatthiasPaul