nextcloud/nextcloudpi #345: nc-audit fail

deljones, 2018-01-05:
Hi..

Tried to run nc-audit and received the following error...

General security audit
./nc-audit.sh: line 71: lynis: command not found
Known vulnerabilities in this system
./nc-audit.sh: line 74: debsecan: command not found

I read the blog regarding this software.  Is it ok to go ahead and reinstall or is there some trickery required to link it all together?

Del

nachoparker, 2018-01-05:
mmm... man, you must be my most unlucky user!! :S 

why didn't it install it?

try reinstalling it manually

```
sudo rm /usr/local/etc/nextcloudpi-config.d/nc-audit.sh
sudo ncp-update
```

deljones, 2018-01-05:
Morning...
I ran 
sudo rm /usr/local/etc/nextcloudpi-config.d/nc-audit.sh
and got...
/usr/local/etc/nextcloudpi-config.d/nc-audit.sh': No such file or directory

:( :(

Running ncp-update gave:

Installing nc-audit
Get:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [11.7 MB]               
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [127 kB]                      
Fetched 11.8 MB in 10s (1,085 kB/s)                                                                                                                  
Reading package lists... Done
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
cp: cannot stat '/etc/lynis/default.prf': No such file or directory
./etc/nextcloudpi-config.d/nc-audit.sh: line 25: /etc/lynis/ncp.prf: No such file or directory

DJ

nachoparker, 2018-01-05:
> dpkg was interrupted

so there was some poweroff during updating, or maybe a ctrl+c, is it possible?

try to fix it 

```
sudo dpkg --configure -a
```
and try again

deljones, 2018-01-05:
Hello... Thanks for helping out!

sudo dpkg gives:

dpkg: dependency problems prevent configuration of python2.7:
 python2.7 depends on python2.7-minimal (= 2.7.13-2+deb9u2); however:
  Version of python2.7-minimal on system is 2.7.13-2.
 python2.7 depends on libpython2.7-stdlib (= 2.7.13-2+deb9u2); however:
  Version of libpython2.7-stdlib:armhf on system is 2.7.13-2.

dpkg: error processing package python2.7 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python2.7-dev:
 python2.7-dev depends on python2.7 (= 2.7.13-2+deb9u2); however:
  Package python2.7 is not configured yet.
 python2.7-dev depends on libpython2.7-dev (= 2.7.13-2+deb9u2); however:
  Package libpython2.7-dev:armhf is not installed.
 python2.7-dev depends on libpython2.7 (= 2.7.13-2+deb9u2); however:
  Version of libpython2.7:armhf on system is 2.7.13-2.

dpkg: error processing package python2.7-dev (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 python2.7
 python2.7-dev

DJ

nachoparker, 2018-01-08:
you will probably have more luck googling, because these problems can be painful to solve. You will have to try to force reinstall just one package with dpkg. Try [this](https://unix.stackexchange.com/questions/109698/dpkg-dependency-problems-prevent-configuration-of-initramfs-tools#109703) first

sometimes aptitude is able to figure out hard situations automatically

do you know what could have happened? did you interrupt an installation, or reboot, or have a power cut?

closed 2018-02-04 by nachoparker