nextcloud/nextcloudpi #347: nc-automount attributes ownership of newly mounted drive to user pi

ovpc, 2018-01-07:
[image]

In process of preparing screenshots for wiki, noticed that my pi4, which I am using as an example to export-backup and import-restore into a new ncp instance and new external drive, will mount the a drive as expected as /media/btrfs (LABEL=btrfs), but it will attribute ownership to user pi.
For security reasons I always disable the default pi user, after creating my own account and adding it to the sudo group.

For myself this is fine for backup/restore I am going to use sudo anyway, and a chown or chgrp is easy enough. 

But as NCP is all about making stuff easy, an extra field in nc-automount enabling a choice of users or even better, let script check /etc/passwd for any user above 1000(pi) and 1001(ncp) and attribute ownership to 1002. 
 

Paste the output of `sudo ncp-report`

nachoparker, 2018-01-08:
this was automounted, right? not in fstab. In my case I get

```
$ server ls -l /media/
drwxrwxrwx 1 root root 416 Jan  8 07:30 myCloudDrive
```

which sounds about right. This is set up by automount automatically.

ovpc, 2018-01-08:
Yes done by automount
But owner is pi ? not root and permisions 700? Not as the other drives, which are als auto-mounted.
drwx------ 1 pi  pi      6 jan   16:05    btrfs

nachoparker, 2018-01-08:
yes, that's weird... I wonder if it has anything to do with the fact that you are using berryboot?

ovpc, 2018-01-13:
Yeah, we'll see, if it comes up again can open this issue again.

closed 2018-01-13 by ovpc