nextcloud/nextcloudpi #348: Certificate problem preventing calendar sync from DAVdroid and Thunderbird

cohenjm98, 2018-01-07:
Hi - New user of nextcloudpi here, and apologize if this is the wrong place to post this issue.  I've had a working raspi 3 nextcloud for about a week, synchronizing calendar through DAVdroid and Thunderbird successfully.  Yesterday started receiving DAVdroid sync errors, and Thunderbird asks for security exceptions. 

I suspect a cert issue, since Thunderbird says:

_"Wrong Site
The certificate belongs to a different site, which means that someone is trying to impersonate this site.
Unknown Identity
The certificate is not trusted... "_

I ran nc-report and there are several of these warnings:
[Sun Jan 07 06:25:02.687654 2018] [ssl:warn] [pid 720:tid 1992560640] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name

So -
How can I remove/renew/reinstall/whatever my certificate(s) to get a fresh start and undo whatever happened?

Thanks

cohenjm98, 2018-01-07:
and... the NextCloud Android app reports "Local file does not exist"  when I try to manage my account.

nachoparker, 2018-01-08:
hi,

You can run letsencrypt again and it will renew the certificates. It is weird what happened... I would investigate more.

the best place for asking these things is in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm). Please, continue there if still in trouble.

closed 2018-01-08 by nachoparker