nextcloud/nextcloudpi #351: Can't add new files to nextcloud instance - locked!

Ricardosgeral, 2018-01-09:
Hi you all. 

Does anyone facing this problem after the latest update of nextcloudpi?

It seems that it's not possible to add new files to the nc instance.

in Windows client it says  " server replied: Locked"
in firefox gives the error "No Reason Phrase"

I have loots of free space in the drive so its not that.

Paste the output of `sudo ncp-report`

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.11
NextCloudPi image    NextCloudPi_11-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sdb
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      828G/2.7T
rootfs usage         2.5G/2.7G
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.4
gateway              192.168.1.1
interface            wlan0
certificates         ricarcloud.duckdns.org
certs due            46 days
NAT loopback         no
uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.4",
            "3": "ricarcloud.ddns.net",
            "4": "ricarcloud.duckdns.org"
        },
        "datadirectory": "\/media\/myCloudDrive\/ncdata",
        "overwrite.cli.url": "https:\/\/ricarcloud.duckdns.org",
        "dbtype": "mysql",
        "version": "12.0.4.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
[... 59 lines cut ...]
```

{"reqId":"WlSXY38AAQEAAAW4hhgAAAAV","level":4,"time":"2018-01-09T10:20:19+00:00","remoteAddr":"193.136.108.166","user":"ricardo","app":"webdav","method":"DELETE","url":"\/remote.php\/dav\/files\/ricardo\/PC_LNEC\/Ricardos\/Journal%20paper%20Reviews\/Geotechnique_Pore%20water%20pressure%20and%20total%20hor%20stress%20EPBM.docx","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\FileLocked\",\"Message\":\"\\\"PC_LNEC\\\/Ricardos\\\/Journal paper Reviews\\\/Geotechnique_Pore water pressure and total hor stress EPBM.docx\\\" is locked\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Tree.php(179): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->delete()\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(287): Sabre\\\\DAV\\\\Tree->delete('files\\\/ricardo\\\/P...')\\n#2 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpDelete(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:DELETE', Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Server.php(258): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/File.php\",\"Line\":353}","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"12.0.4.3"}
{"reqId":"WlSXY38AAQEAAAW4hhgAAAAI","level":4,"time":"2018-01-09T10:20:20+00:00","remoteAddr":"193.136.108.166","user":"ricardo","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/ricardo\/PC_LNEC\/Ricardos\/Journal%20paper%20Reviews\/Geotechnique_Pore%20water%20pressure%20EPBM.docx","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\FileLocked\",\"Message\":\"\\\"PC_LNEC\\\/Ricardos\\\/Journal paper Reviews\\\/Geotechnique_Pore water pressure EPBM.docx\\\" is locked\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('Geotechnique_Po...', Resource id #10)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('files\\\/ricardo\\\/P...', Resource id #10, NULL)\\n#2 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Server.php(258): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php\",\"Line\":159}","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"12.0.4.3"}
{"reqId":"WlSXfH8AAQEAAAW6EGMAAABC","level":4,"time":"2018-01-09T10:20:45+00:00","remoteAddr":"193.136.108.166","user":"ricardo","app":"webdav","method":"MKCOL","url":"\/remote.php\/dav\/files\/ricardo\/PC_LNEC\/Ricardos\/Journal%20paper%20Reviews\/Fritzing","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\FileLocked\",\"Message\":\"\\\"PC_LNEC\\\/Ricardos\\\/Journal paper Reviews\\\/Fritzing\\\" is locked\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1210): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createDirectory('Fritzing')\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(590): Sabre\\\\DAV\\\\Server->createCollection('files\\\/ricardo\\\/P...', Object(Sabre\\\\DAV\\\\MkCol))\\n#2 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpMkcol(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:MKCOL', Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Server.php(258): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php\",\"Line\":190}","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"12.0.4.3"}
{"reqId":"WlSXfH8AAQEAAAW4hhsAAAAO","level":4,"time":"2018-01-09T10:20:45+00:00","remoteAddr":"193.136.108.166","user":"ricardo","app":"webdav","method":"DELETE","url":"\/remote.php\/dav\/files\/ricardo\/PC_LNEC\/Ricardos\/Journal%20paper%20Reviews\/Geotechnique_Pore%20water%20pressure%20and%20total%20hor%20stress%20EPBM.docx","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\FileLocked\",\"Message\":\"\\\"PC_LNEC\\\/Ricardos\\\/Jo [... cut ...] /Server.php(258): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(33): OCA\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php\",\"Line\":159}","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"12.0.4.3"}

```
</details>

nachoparker, 2018-01-10:
mmm weird, supposedly with redis this should be fixed... was there a power cut or similar?

if you go to the admin panel, does it say that cron ran successfully?

Ricardosgeral, 2018-01-10:
well i have made a nc-restore (still running) so i can go to admin panel.

nachoparker, 2018-01-13:
ok, tell us if you find more problems

closed 2018-01-13 by nachoparker