nextcloud/nextcloudpi #353: NextCloudPi "hangs" every few weeks

RalfJahns, 2018-01-11:
pi@nextcloudpi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

![grafik](https://user-images.githubusercontent.com/15796653/34842312-6be6c9f2-f70b-11e7-8144-53db5298ad0d.png)

ovpc, 2018-01-12:
I have had RPI's stalling by:
- insufficient power supply (or perhaps USB socket from PC), or a dodgy micro-usb cable from dump shop.
- corrupt sdcards (bad quality or end of life/number of read/write exceeded)

Does command dmesg show any problem. Various other log files are stored in directory /var/log directory.

nachoparker, 2018-01-12:
well, if you want to check your suspiction, you can check the unattended logs

```
sudo cat /var/log/unattended-upgrades/unattended-upgrades.log
```

RalfJahns, 2018-01-12:
Hello and thank you very much for your replies!

So, easiest things first:

Regarding HW quality: I bought the Pi and the SD card explicitly for having a NextCloudPi server approximately 3 months ago. The SD Card is a SanDisk Extreme 32GB. The power supply is a 2A power supply that was also recommended for Cubietruck board (which needs more power than a Pi).

The unattended-upgrades.log shows nothing that would explain the hang. But is seems some components will not get automatically updated, because manual input is needed!? (Don't understand what conffile prompt is.)

<prev>
2018-01-12 06:04:11,865 WARNING Package 'samba' has conffile prompt and needs to be upgraded manually
2018-01-12 06:04:14,726 WARNING Package 'php7.0-fpm' has conffile prompt and needs to be upgraded manually
2018-01-12 06:04:26,839 INFO package 'libsmbclient' not upgraded
2018-01-12 06:04:38,202 INFO package 'libwbclient0' not upgraded
2018-01-12 06:04:54,211 INFO package 'php7.0-cli' not upgraded
2018-01-12 06:05:04,558 INFO package 'php7.0-common' not upgraded
2018-01-12 06:05:14,904 INFO package 'php7.0-curl' not upgraded
2018-01-12 06:05:25,256 INFO package 'php7.0-fpm' not upgraded
2018-01-12 06:05:35,605 INFO package 'php7.0-gd' not upgraded
2018-01-12 06:05:45,950 INFO package 'php7.0-json' not upgraded
2018-01-12 06:05:56,298 INFO package 'php7.0-mbstring' not upgraded
2018-01-12 06:06:06,650 INFO package 'php7.0-mcrypt' not upgraded
2018-01-12 06:06:16,996 INFO package 'php7.0-mysql' not upgraded
2018-01-12 06:06:27,348 INFO package 'php7.0-opcache' not upgraded
2018-01-12 06:06:37,700 INFO package 'php7.0-readline' not upgraded
2018-01-12 06:06:48,047 INFO package 'php7.0-xml' not upgraded
2018-01-12 06:06:58,396 INFO package 'php7.0-zip' not upgraded
2018-01-12 06:07:09,750 INFO package 'python-samba' not upgraded
2018-01-12 06:07:21,075 INFO package 'samba' not upgraded
2018-01-12 06:07:32,431 INFO package 'samba-common' not upgraded
2018-01-12 06:07:43,756 INFO package 'samba-common-bin' not upgraded
2018-01-12 06:07:55,111 INFO package 'samba-libs' not upgraded
2018-01-12 06:08:06,349 INFO Packages that will be upgraded : php7.0
2018-01-12 06:08:06,350 INFO Writing dpkg log to '/var/log/unattended-upgrades/unattended-upgrades-dpkg.log'
2018-01-12 06:08:18,076 INFO All upgrades installed
</prev>

Next thing is /var/log/kern.log. I have compared the messages from the boot on 9th of january (after that the the Pi stopped working) with some other boot logs and can see no important differences. After the device is booted, the kernel is quiet (no more log entries). There are some "EXT4-fs (mmcblk0p2): INFO: recovery required on readonly filesystem" messages in some boot logs, but after that the Pi worked without problems. I had to unpower the Pi to reboot it, because login did not work.

In /var/log/messages, there are nearly the same log entries (mostly from the bootup) like in kern.log

There are different log entries in syslog and I will attach it here. Perhaps anyone can be so kind to have a look.
[syslog.2.txt](https://github.com/nextcloud/nextcloudpi/files/1627253/syslog.2.txt)

Best Regards
Ralf

nachoparker, 2018-01-13:
```
INFO: recovery required on readonly filesystem" 
```
that does make me suspect SD card problems...

Also, I just noticed the unattended upgrades warnings for the first time. We'll have to look into those, but that's another issue.

other than that I don't see anything odd

I suggest you reopen next time you see that it stops working for you, but don't reboot it so we can have a look at it. Paste ncp-report output if this happens

closed 2018-01-13 by nachoparker

RalfJahns, 2018-01-13:
Hello,

I doubt that it is a SD card problem. SandDisk is a top brand for these products, probably you will find no other manufacturer with better reputation. And the card is new.
Additionally such messages are typical after a power cut. If there is a seriously corrupted file system, there are much more kernel log entries about the repair process than I had.

O.K. I will reopen this issue if it happens again. As I wrote remote login was not possible in that state, but I will try to connect Keyboard and monitor.

Best regards and thanks again for your reply.
Ralf