nextcloud/nextcloudpi #355: Accessing Nextcloud from outside problem

cored4, 2018-01-12:
I was thinking whether to ask about it or not but after spending a lot of time trying to figure it out I decided to ask for help. I'm trying to setup NextcloudPi to be able to access it from outside. I registered on https://freedns.afraid.org/, added my own subdomain, have dynamic dns update set up. The problem is, that subdomain just won't work for me. I tried using both Setup Wizard or manual config to set it up and neither works for me. That's how I set it up manually:
1. Run ns-forward-ports from panel (in the output I get external xx.xxx.xxx.xxx:443 TCP is redirected to internal xxx.xxx.x.xxx:443 (duration=0) ) - where external is different from IP Address that I see as my subdomain destination, should it be like that?
2. Go to freeDNS, edit subdomain, update hash.
3. Go to dnsmasq, edit subdomain, I didn't touch dnsserver - it's still on 8.8.8.8
4. Go to letsencrypt, use domain from before and my own e-mail. 
At this point I get letsencrypt timeout on http-01 challenge. "The server could not connect to the client to verify the domain".

My freedns subdomain still doesn't work. Am I doing something wrong?

ovpc, 2018-01-13:
Hi,
Thanks for reporting

Letsencrypt needs both ports :80 and :443 to verify ownership of the sub.domain

In terminal run
`wget -qO- http://ipecho.net/plain ; echo`
It will/should return your current **external** IP address
And try accessing your NCP through it. In browser https://yourIPnumber
If ports are correctly forwarded, NCP should give you a warning about connecting to Not Trusted Domain. Or your browser will warn you about a non valid certificate.

Report back here please

You are reporting three issues, which should be tackled in this order:
1 port forwarding
2 dynamic DNS
3 getting a valid certificate for your domain

cored4, 2018-01-13:
Thank you for your reply. I ran that command and the output was the same ip as on my subdomain in freedns.afraid.org. When I try to go to that IP from browser I get "Secure Connection Failed" message in Firefox.

ovpc, 2018-01-13:
Firefox should also provide option "Advanced" click on it and add exception for your IP number. To see if port forward is working, you might need to reload/refres.

cored4, 2018-01-13:
https://i.imgur.com/4p9U2hj.png that's the whole page

ovpc, 2018-01-13:
In Firefox menu go to settings>Security>CertificateManager>ServerTab>click add exception>past your IP and save/confirm exception.
Then re-load page.

cored4, 2018-01-13:
Got error "Unable to obtain idenfitication status for this site" while trying to add exception.

ovpc, 2018-01-13:
Ofcourse, but you can add/confirm it anyway.
You can also add exception for your freedns.afraid.org domain while in there ;-)

cored4, 2018-01-13:
I can't click on the button to confirm security exception though.

ovpc, 2018-01-13:
tried with your free dns domain too?

cored4, 2018-01-13:
Yes, same thing.

ovpc, 2018-01-13:
Are you on Telegram? Join me/us at https://t.me/NextCloudPi We can go private, if you want I can try access it for you.
If not output of nc-info here please

ovpc, 2018-01-13:
You can still put output of nc-info here if you want to re-open.

closed 2018-01-13 by ovpc