nextcloud/nextcloudpi #356: Can't enable SSH

aleprovencio, 2018-01-12:
When trying the method described [here](https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-a-Raspberry-Pi) even though trying a new non-existent the system always returns with the error message "The user pi is using the default password. Refusing to activate SSH".

Sorry for not pasting the output of `sudo ncp-report` because I guess that should be done via ssh.

ovpc, 2018-01-13:
You can run nc-report from ncp-web at https://nextcloudpi.local:4443

nachoparker, 2018-01-13:
you first have to do

user: _pi_
password: _something different_

in order to change the default password, that is insecure.

Then, you can log in through user _pi_ with that password, or also you can create new users and passwords without problems

I am planning on simplifying this by disabling the _pi_ user when you use this tool

closed 2018-01-13 by nachoparker

ovpc, 2018-01-15:
Extra info
In order to enable SSH, the password for user pi can not remain set to the default              
             │ raspberry.                                                                             │              
             │ You HAVE to create a NEW password for pi if you want this program to enable SSH, it    │              
             │ will fail if you dont!                                                                 │              
             │ Note: Use normal AlphaNumeric, the only special characters allowed are .,@-_/  
Just using thread to save screenshots for doc/wiki
[image]
[image]
[image]
[PULL Modified SSH HERE](https://github.com/ovpc/nextcloudpi/blob/master/etc/nextcloudpi-config.d/SSH.sh)

aleprovencio, 2018-01-15:
Just to let you know, i tried changing the user `pi` password but then I got the error message `The user ncp is using the default password`. So instead of `pi` I changed passwords for the `ncp` user and everything now works as expected.

UkeFan, 2018-12-01:
For ODroid-Users: The only users name that works is "root"

ovpc, 2018-12-01:
[FAQ preset passwords](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#what-are-pre-set-userspasswords-on-nextcloudpi)