nextcloud/nextcloudpi #357: ncp-update hangs at 'Performing updates'

burgerkingGH, 2018-01-14:
Hello,

Thank you for this incredible time saving work. At the moment, ncp-update hangs on 'Performing updates'. It goes nowhere until I ctrl-c kill it, and then it proceeds, see below:

pi@raspberrypi:~ $ sudo ncp-update
Downloading updates
Performing updates
^CInstalling nc-backup
Activating nc-backup-auto
automatic backups enabled
NextCloudPi updated to version v0.44.15

Otherwise, the server seems to run fine. Any thoughts on what might be happening?

Thanks!

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.44.15
NextCloudPi image    untagged
distribution         Raspbian GNU/Linux 8 \n \l
automount            no
USB devices          sda 
datadir              /media/externaldrive/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      436G/2.7T
rootfs usage         2.5G/15G
swapfile             /media/externaldrive/swapfile/swap
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": [
            "***REMOVED SENSITIVE VALUE***",
            "***REMOVED SENSITIVE VALUE***"
        ],
        "datadirectory": "\/media\/externaldrive\/nextcloud\/data",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "12.0.4.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "logtimezone": "UTC",
        "installed": true,
        "instanceid": "oc44otuffnke",
        "loglevel": "2",
        "log_type": "file",
        "logfile": "\/var\/www\/nextcloud\/data\/nextcloud.log",
        "maintenance": false,
        "mail_smtpmode": "php",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "admin",
        "mail_domain": "ownyourbits.com",
        "theme": "",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "\/var\/run\/redis\/redis.sock",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        }
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

nachoparker, 2018-01-15:
hi, thanks for your words.

well, we cannot support you because

- you are still on raspbian 8 jessie
- you haven't updated in a looong time. You are supposed to update every month or less, ideally use automatic updates

who knows what's happening there. I suggest that you export all your accounts, calendars, contacts and data and start fresh with the latest image. You can do the web wizard in order to make sure that automatic NCP updates get activated for the future.

closed 2018-01-15 by nachoparker

valentingc, 2018-02-08:
Actually, I have exact the same problem but I'm pretty sure I updated NCP at least every month.
Seems like there was something bad in one of these updates.

**Update from v0.44.15 to v0.46.3:**

```
pi@*sensitive*~$ sudo ncp-update
Downloading updates
Performing updates
^CInstalling nc-backup
Installing nc-backup-auto
fatal: Unable to read current working directory: No such file or directory
NextCloudPi updated to version
rm: cannot remove '/tmp/ncp-update-tmp': Directory not empty
```

When I ran ncp-update again, the errors disappeared:
```Downloading updates
Performing updates
^CInstalling nc-backup
Installing nc-backup-auto
NextCloudPi updated to version v0.46.3
```
**Diagnostics**
<details>
<summary>Diagnostics</summary>

```
Cannot load Zend OPcache - it was already loaded
Gathering information...
NextCloudPi version  v0.46.3
NextCloudPi image    untagged
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /mnt/ncdata (using a cifs mount for data)
data in SD           no
data filesystem      cifs
data disk usage      ***REMOVED SENSITIVE VALUE***
rootfs usage         5.6G/30G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
certs due            ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2min
```

</details>

<br />
I had that error aswell when I tried to update to v0.44.15.
It's sad to hear that we have to reinstall from scratch just to fix this issue. Should not be a big deal with a database dump but still needs some hours.

nachoparker, 2018-02-08:
Well, nc-backup and nc-backup-auto are pretty old. Many months old. 

Sorry to say, but I suggest you spend those hours and click nc-autoupdate-ncp on your new instance