nextcloud/nextcloudpi #358: Docker ownyourbits/nextcloudpi-x86 with setting custom directory

Robert-turbo, 2018-01-15:
I was trying to run ownyourbits/nextcloudpi-x86 docker image with setting custom directory as directory for /data 

I was trying to run it with `"docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/nextcloudpi/data:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 $DOMAIN"` where DOMAIN was IP adress but it didn't work. In logs I found:

```
Starting PHP-fpm
Starting Apache
AH00112: Warning: DocumentRoot [/var/www/nextcloud] does not exist
AH00112: Warning: DocumentRoot [/var/www/nextcloud] does not exist
(2)No such file or directory: AH02291: Cannot access directory '/var/www/nextcloud/data/' for error log of vhost defined at /etc/apache2/sites-enabled/nextcloud.conf:2
AH00014: Configuration check failed
Action 'start' failed.
The Apache error log may have more information.
Starting Redis
Starting Cron
Starting Postfix
Uninitialized instance, running nc-init...
Activating nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
2018-01-15 21:53:00 140236956791360 [Note] mysqld (mysqld 10.1.26-MariaDB-0+deb9u1) starting as process 138 ...
2018-01-15 21:53:00 140236956791360 [Warning] Can't create test file /data/database/2364e442ff9d.lower-test
mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
2018-01-15 21:53:00 140236956791360 [ERROR] Aborting
```

Where can be problem?

with another nextcloud docker images I was able to mount custom directory.

I was trying to do it because i need to move from Raspberry pi  2 to x86 computer.

nachoparker, 2018-01-16:
Hi,

If you want to do that, you have to mount to `/data/app/data`

nachoparker, 2018-01-16:
so to be clear

```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data -v /media/nextcloudpi/data:/data/app/data  --name nextcloudpi ownyourbits/nextcloudpi-x86 $DOMAIN
```

closed 2018-01-16 by nachoparker

Robert-turbo, 2018-01-16:
Thank you!

It helped to start services... But now I have another error.

Making /usr/local/etc/nextcloudpi-config.d persistent ...
Making /etc/apache2 persistent ...
Starting PHP-fpm
Starting Apache
Starting mariaDB
2018-01-16 10:17:30 140505686475328 [Note] mysqld (mysqld 10.1.26-MariaDB-0+deb9u1) starting as process 33 ...
Starting Redis
Starting Cron
Starting Postfix
Uninitialized instance, running nc-init...
Activating nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud is not installed - only a limited number of commands are available
Can't create or write into the data directory /data/app/data
Init done

Is it still problem with write permission? Do you know where  should I look?

nachoparker, 2018-01-16:
I am not sure of what you are trying to achieve, maybe run it without mounting the datafolder, so it inits all the way the first time, then after that mount your own datafolder, but you will have to copy the contents of a valid data folder, because it will look for the `.ocdata` folder in it.

Right now you are in an inconsistent state, so delete the container **and** the ncdata volume to start from zero.

Robert-turbo, 2018-01-16:
I would like to have easy access to data files like in https://github.com/ichiTechs/Dockerized-SSL-NextCloud-with-MariaDB configuration.

In that  case i can use restic to backup all files and database.

nachoparker, 2018-01-16:
yes, what you are trying to do should work, just that you need to have a valid datadir to mount (with the required files, including `.ocdata`.

testbird, 2018-02-05:
The docker recomended way seems to be to  run the docker image as normal (with a named volume, possibly own IP if you like: #381) then copy your nc data, and ncp /etc stuff, backup to /var/lib/docker/volumes/ncdata/_data 

A trick if you need to have the directory created at another location do the following, to have the volume ready before the first run:
`docker volume create --opt type=none --opt device=/host/path --opt o=bind ncdata`
https://github.com/moby/moby/issues/17470#issuecomment-350744378

testbird, 2018-02-06:
FYI: Only only when a **named volume** is attached for the first time, does docker copy (install) the default files that are shipped within in the image into the data volume.

nachoparker, 2018-02-06:
> FYI: Only only when a named volume is attached for the first time, does docker copy (install) the default files that are shipped within in the image into the data volume.

yes, that can be tricky for newcomers :P