nextcloud/nextcloudpi #359: ncp-backup faul

pmjklemm, 2018-01-15:
Hai,

first of all, thanks for this amazing nextcloudpi (: 

so I have a 128 GB SSD as the main storage in the rapsi. Works like a charm ! 
Now I try to set up a backup system with a 1 TB usb storage. (format vFat or FAT32)

when I try to run ncp-backup it says "free space check failed. Need 89G" 
That does not make any sense since it have 1TB free space. 

I modified the ncp-backup  file so that i bypass the free-space checking.
still the backup terminates after 4 GB. Dont know whats happening.

Any suggestions?

Thanks

------------
Additional Informations: 

lsblk:
└─sda1        8:1    0 931.5G  0 part /media/USBdrive

fdisk -l
Disk /dev/sda: 931.5 GiB, 1000204886016 bytes, 1953525168 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 8DB14219-D250-443E-91B1-8BDF129C5EE6
/dev/sda1   2048 1953523711 1953521664 931.5G Microsoft basic data

nachoparker, 2018-01-16:
mmm you are right, have to do the check over the destination dir for the data. Will fix that. 

do you have both drives plugged in? lsblk only shows one

pmjklemm, 2018-01-16:
the full lsblk:

sda           8:0    0 931.5G  0 disk             (<-- the external hdd 1TB, connected via usb)
└─sda1        8:1    0 931.5G  0 part /media/USBdrive
mmcblk0     179:0    0 119.1G  0 disk       (<-- internal 128GB SSD)
├─mmcblk0p1 179:1    0  41.5M  0 part /boot
└─mmcblk0p2 179:2    0   119G  0 part /

nachoparker, 2018-01-16:
SSD.. do you mean SD card?

pmjklemm, 2018-01-16:
Right, that is true. SD Card it is

pmjklemm, 2018-01-20:
Hey i am back :D

so backup still does not work for big datasets.

First here is a small fix for the ncp-backup (line 43):

old:
FREE=$( df    "$DATADIR" | tail -1 | awk '{ print $4 }' )

new:
FREE=$( df    "$DESTDIR" | tail -1 | awk '{ print $4 }' )

You want to check the destination directory for free space, not the data directory.
---
Still this is the output for a dataset of 90 GB:

Maintenance mode enabled
Nextcloud is in maintenance mode - no app have been loaded

check free space...
backup database...
backup base files...
backup data files...
tar: /media/DATA/nextcloud-bkp_20180120.tar: Wrote only 4095 of 10240 bytes
tar: Error is not recoverable: exiting now
error generating backup
Nextcloud is in maintenance mode - no app have been loaded

Maintenance mode disabled

---
I googled the tar wrote only X of Y bytes, does not really understand the error ???

nachoparker, 2018-02-04:
Hi, I was away for a few weeks...

Thanks, I'll fix that line ;)

About the error... seems like a write error to me. Does it also happe with a small data size?

pmjklemm, 2018-02-05:
Welcome back (:

I tested with a different file format and this fixed it. 
With FAT32, probably it terminates because it tries to copy a file larger than 2 Gb.
With ext4 there is no problem. 

- Open Problem:

There is another problem: the ncp-backup-auto function does not include data (if the option is checked in the ncp panel)

- Feature Request:

I would also really like a time system like Timecapsule: storing the data in exponential time intervals, meaning there is a backup for the last 3 days each and one for the last 3 weeks, one for the last 3 month, etc ...
Would be way more convenient (: 

Thanks for everything 👍

nachoparker, 2018-02-05:
aaah you were using FAT32... that makes sense now

about ncp-backup-auto you are right... I changed some things recently and I introduced a mistake. Please check again after `v0.45.1` and open an issue if it keeps happening to you. I'll fix it right away

well you should check the feature 'BTRFS snapshots'... it is very similar to that time capsule

```
$ ls -l /media/myCloudDrive/ncp-snapshots/
total 0
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-01-29_211702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-01-30_211703
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-01-31_221703
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-02-01_231702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-02-02_231703
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-02-03_231702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 daily_2018-02-04_231702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_161702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_181701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_191701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_201701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_211701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_221701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-04_231701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_001701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_011701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_021701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_031702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_041702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_061701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_071701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_081701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_091701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_101701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_111701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_121701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_131701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_141702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_161701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_171701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 hourly_2018-02-05_181701
drwxr-x--- 1 www-data pi 340 Dec 28 09:48 manual_2017-12-28_113633
drwxr-x--- 1 www-data pi 340 Dec 28 09:48 monthly_2017-12-28_101702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 monthly_2018-01-27_101703
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 weekly_2018-01-11_111701
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 weekly_2018-01-18_111702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 weekly_2018-01-25_111702
drwxr-x--- 1 www-data pi 328 Jan  5 16:24 weekly_2018-02-01_111702
```

thanks!

closed 2018-02-05 by nachoparker