nextcloud/nextcloudpi #360: An unfinished problem with the installation on Debian

TheBocios, 2018-01-17:
I run NextCloudPi on Debian 9.3 nad run script:
_curl -sSL [https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh](https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh) | bash_

Everything goes well until the moment shown below.
And after that the computer has full CPU and memory.
Now I do not know if the installation is finished or if something else should be done.

Terminal:
Installing nc-backup
NextCloudPi updated to version v0.44.15
Activating nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud is not installed - only a limited number of commands are available
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 1 set to string 192.168.1.151
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value mail_smtpmode set to string php
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
--2018-01-17 09:00:01--  https://github.com/nextcloud/admin_notifications/releases/download/v1.0.1/admin_notifications-1.0.1.tar.gz
Translacja github.com (github.com)... 192.30.253.113, 192.30.253.112
Łączenie się z github.com (github.com)|192.30.253.113|:443... połączono.
Żądanie HTTP wysłano, oczekiwanie na odpowiedź... 302 Found
Lokalizacja: https://github-production-release-asset-2e65be.s3.amazonaws.com/81433081/94dafc90-cdfb-11e7-95a5-4c239c33ece0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180117T080002Z&X-Amz-Expires=300&X-Amz-Signature=5acf33b3c10266a742135ea9ac714c3459dee50aa8fd542688aa77563d001ff2&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dadmin_notifications-1.0.1.tar.gz&response-content-type=application%2Foctet-stream [podążanie]
--2018-01-17 09:00:02--  https://github-production-release-asset-2e65be.s3.amazonaws.com/81433081/94dafc90-cdfb-11e7-95a5-4c239c33ece0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180117T080002Z&X-Amz-Expires=300&X-Amz-Signature=5acf33b3c10266a742135ea9ac714c3459dee50aa8fd542688aa77563d001ff2&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dadmin_notifications-1.0.1.tar.gz&response-content-type=application%2Foctet-stream
Translacja github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.18.96
Łączenie się z github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.18.96|:443... połączono.

Iolaum, 2018-01-18:
Please format your comment and it's terminal output by including code annotation tags:

\`\`\`
put terminal output here
\`\`\`

To check if nextcloudpi installed properly log in on your debian machine and open a ternimal.

```
# Type the following command
$ which nextcloudpi-config
/usr/local/bin/nextcloudpi-config
```

If the output is as above then the installed should have completed. type `sudo nextcloudpi-config` and start setting up your nextcloudpi instance.

TheBocios, 2018-01-19:
Nextcloudpi installed properly, but stopping the script at the place:
```
# NCP notifications
  local URL=$( wget -q -O - https://api.github.com/repos/nextcloud/admin_notifications/releases | \
                grep browser_download_url | head -1 | cut -d '"' -f 4 )
  cd /var/www/nextcloud/apps
  wget "$URL" -O admin_notifications.tar.gz
  tar -xf admin_notifications.tar.gz
  rm *.tar.gz
  chown -R www-data:www-data *
  sudo -u www-data php /var/www/nextcloud/occ app:enable admin_notifications
```
and can not continue because it hangs.
In addition, everything looks correct except for the administrator's password.
Despite entering
login: admin
password: ownyourbits
and you can not connect.

Iolaum, 2018-01-19:
The [url in the script](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-init.sh#L140-L141) may be wrong, @nachoparker will have to check that when he can.

Good idea to try and get the file manually. I 'm surprised that you couldn't do it. When I run:

```
$ wget https://github.com/nextcloud/admin_notifications/releases/download/v1.0.1/admin_notifications-1.0.1.tar.gz
--2018-01-19 21:03:02--  https://github.com/nextcloud/admin_notifications/releases/download/v1.0.1/admin_notifications-1.0.1.tar.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/81433081/94dafc90-cdfb-11e7-95a5-4c239c33ece0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180119%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180119T210302Z&X-Amz-Expires=300&X-Amz-Signature=eeaf1566c61a5f06572915838ee957de6c94a2f82bf42dd1ee549834cd818853&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dadmin_notifications-1.0.1.tar.gz&response-content-type=application%2Foctet-stream [following]
--2018-01-19 21:03:02--  https://github-production-release-asset-2e65be.s3.amazonaws.com/81433081/94dafc90-cdfb-11e7-95a5-4c239c33ece0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180119%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180119T210302Z&X-Amz-Expires=300&X-Amz-Signature=eeaf1566c61a5f06572915838ee957de6c94a2f82bf42dd1ee549834cd818853&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dadmin_notifications-1.0.1.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.128.75
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.128.75|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 18353 (18K) [application/octet-stream]
Saving to: ‘admin_notifications-1.0.1.tar.gz’

admin_notifications-1.0.1.tar.gz                   100%[================================================================================================================>]  17.92K  86.8KB/s    in 0.2s    

2018-01-19 21:03:03 (86.8 KB/s) - ‘admin_notifications-1.0.1.tar.gz’ saved [18353/18353]
```

I get the release file properly. Can you try another website as a test? You may have connectivity issues.

TheBocios, 2018-01-22:
I downloaded the package "manually" and finished the installation. Most likely a problem with my network's access.

closed 2018-01-22 by TheBocios