nextcloud/nextcloudpi #361: Dead link in Wiki (FAQ)

KurtPfeifle, 2018-01-18:
Second question in [Wiki-FAQ](https://github.com/nextcloud/nextcloudpi/wiki/FAQ) says "Don't forget the [Required Tasks](https://github.com/nextcloud/nextcloudpi/wiki/Required-Tasks-for-NextCloudPi)", but the link is dead...

ovpc, 2018-01-25:
Link is not dead, it works, but wiki page has no content yet. 
Feel free to contribute ;)

Pant, 2018-01-31:
@KurtPfeifle, thanks for reporting this.

This page was made at the very beginning of the wiki and was deleted while improving the wiki. As now it is not needed and the steps that where in that page are being described in other pages, I removed the line from the FAQ page with the 'dead' link.

closed 2018-01-31 by Pant

KurtPfeifle, 2018-01-31:
@ovpc wrote:

> *"Feel free to contribute ;)"*

That's all nice and well... however, I visited the Wiki because (amongst other things) I wanted to *learn* what the "Required Tasks" are. So, as a newbie to NextcloudPi, how should I know? \*shrug\*

In any case, you guys seem to rate it as *"improvement the Wiki"* when you spread the topic of "Required Tasks" across some other (unidentified) Wiki pages.

Oh, well...   :-(

Pant, 2018-01-31:
For the record, this page was like a tutorial for setting up the nextcloudpi such as: `nc-datadir`, `letsencrypt`, `no-ip`/`freDNS`, etc... Which later, all these got replaced by our Wizard, making this page have no reason to exist anymore.

The Wiki needs a lot of love and a lot of improvements. Of course there are leftovers from previous edits, such as the one you identified.

Although I didn't manage to find any other mention for "Required Tasks" page in the wiki. If you do find any, please come back here and note it.
> In any case, you guys seem to rate it as "improvement the Wiki" when you spread the topic of "Required Tasks" across some other (unidentified) Wiki pages.

KurtPfeifle, 2018-01-31:
@Pant:

Thx for your pointer to the Wizard, which resolves the mystery for me now.

nachoparker, 2018-02-05:
you can also check ownyourbits.com where I wrote articles about all the steps required for NCP, if you want to learn more ;)