nextcloud/nextcloudpi #362: Update to 12.0.4 error

redone101, 2018-01-19:
Paste the output of `sudo ncp-report`

Hello,

i just wanted to update my nextcloudpi from 12.0.3. to 12.0.4 from the GUI/Webbrowser, but get this error:

Check for expected files
The following extra files have been found:

    data-11-28-17

I don't know if it depends on fact, that i installed a new image and made a restore, which worked without any problems, by the way.

"NextCloudPi v0.44.15 is up to date"   and i can run nc-update without any problems.

Here is the ncp-report output:

sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_11-23-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      14G/30G
rootfs usage         14G/30G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.3.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.3.250
gateway              192.168.3.1
interface            eth0
certificates         ******************(took this out ;)
certs due            37 days 
NAT loopback         no
uptime               12min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
[... 95 lines cut ...]
```

{"reqId":"ztzbqNb3oSx2whk98P2w","level":3,"time":"2018-01-18T02:22:49+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"--","url":"--","message":"Exception: {\"Exception\":\"GuzzleHttp\\\\Exception\\\\ConnectException\",\"Message\":\"cURL error 28: Resolving timed out after 10519 milliseconds\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(103): GuzzleHttp\\\\Exception\\\\RequestException::wrapException(Object(GuzzleHttp\\\\Message\\\\Request), Object(GuzzleHttp\\\\Ring\\\\Exception\\\\ConnectException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(165): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(125): GuzzleHttp\\\\Client->send(Object(GuzzleHttp\\\\Message\\\\Request))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Http\\\/Client\\\/Client.php(138): GuzzleHttp\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(99): OC\\\\Http\\\\Client\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/AppFetcher.php(66): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->fetch('\\\"2018-01-16 21:...', '[{\\\"id\\\":\\\"spreedm...')\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(162): OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher->fetch('\\\"2018-01-16 21:...', '[{\\\"id\\\":\\\"spreedm...')\\n#12 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Installer.php(394): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->get()\\n#13 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(258): OC\\\\Installer::isUpdateAvailable('twofactor_backu...', Object(OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher))\\n#14 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(155): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->isUpdateAvailable('twofactor_backu...')\\n#15 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/updatenotification\\\/lib\\\/Notification\\\/BackgroundJob.php(79): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->checkAppUpdates()\\n#16 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/BackgroundJob\\\/Job.php(59): OCA\\\\UpdateNotification\\\\Notification\\\\BackgroundJob->run(NULL)\\n#17 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/BackgroundJob\\\/TimedJob.php(54): OC\\\\BackgroundJob\\\\Job->execute(Object(OC\\\\BackgroundJob\\\\JobList), Object(OC\\\\Log))\\n#18 \\\/var\\\/www\\\/nextcloud\\\/cron.php(124): OC\\\\BackgroundJob\\\\TimedJob->execute(Object(OC\\\\BackgroundJob\\\\JobList), Object(OC\\\\Log))\\n#19 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Exception\\\/RequestException.php\",\"Line\":49}","userAgent":"--","version":"12.0.3.3"}
{"reqId":"ztzbqNb3oSx2whk98P2w","level":3,"time":"2018-01-18T02:22:59+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"--","url":"--","message":"Exception: {\"Exception\":\"GuzzleHttp\\\\Exception\\\\ConnectException\",\"Message\":\"cURL error 28: Resolving timed out after 10520 milliseconds\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(103): GuzzleHttp\\\\Exception\\\\RequestException::wrapException(Object(GuzzleHttp\\\\Message\\\\Request), Object(GuzzleHttp\\\\Ring\\\\Exception\\\\ConnectException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(165): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(125): GuzzleHttp\\\\Client->send(Object(GuzzleHttp\\\\Message\\\\Request))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Http\\\/Client\\\/Client.php(138): GuzzleHttp\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(99): OC\\\\Http\\\\Client\\\\Client->get('https:\\\/\\\/apps.ne...', Array)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/AppFetcher.php(66): OC\\\\App\\\\AppStore\\\\Fetcher\\\\Fetcher->fetch('\\\"2018-01-16 21:...', '[{\\\"id\\\":\\\"spreedm...')\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/AppStore\\\/Fetcher\\\/Fetcher.php(162): OC\\\\App\\\\AppStore\\\\Fetcher\\\\AppFetcher->fetch('\\\"2018-01-16 21:...', '[{\\\"i [... cut ...] s host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.3.3"}
{"reqId":"WmDVzn8AAQEAAAZh5nwAAAAJ","level":2,"time":"2018-01-18T17:13:50+00:00","remoteAddr":"73.23.197.18","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"73.23.197.18\" tried to access using \"85.124.11.75\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.3.3"}

```
</details>

MichaelSchmidts, 2018-02-01:
I have the same problem form 12.0.4 to 12.0.5 form the GUI/Webbrowser. With other files:  

Check for expected files
The following extra files have been found: mnt

nachoparker, 2018-02-04:
Seems like those files were put there at some point, maybe from ncp-web?

In any case, just delete them or move them, they shouldn't be there. You can try again after that.

closed 2018-02-04 by nachoparker