nextcloud/nextcloudpi #363: as a follow-up to Issue #334

Beres-Berlin, 2018-01-21:
Hi!

using/testing nextcloudpi on my first raspberry (model 3 b), that I bought (together with a WD Pidrive) specifically for this purpose. First Linux experience as well =). First had nextcloud installed on its own, then discovered your nextcloudpi-image and chose it because of security. So far, I get along quite well. Thanks for the great work!

I have two questions though:
- in issue #334 you write (if I understand correctly...) that one should change the datadir to /media/myCloudDrive instead of using the symlink /media/USBDrive, which can be lost after reboot. In fact, in my installation (0.44.15 via ncp-update) this happens after reboot. I can re-create the symlink via sudo /...../nc-automount-links, and everything works but I'd like to follow your advice and change all references to the device /media/myCloudDrive which is mounted on reboot. By the way how do you do that without an entry in /etc/fstab?

Question A) Why does "move data to usb", which you wrote (if I'm correct) create the symlink in the first place?

Question B) Which references have to be changed and where do I find them - config.php I know about, but where to find the mysqld configuration file, for example?

Thanks again!

Pant, 2018-01-31:
Hello @Beres-Berlin, thanks for reporting this.

Answer A:
From the Web Panel (:4443) the nc-datadir does not links anything. It just sets the data directory to the Nextcloud app.

Answer B:
What @BETEP-ok did in the #334 is the following:

Optionally, change `myCloudDrive` with a directory name of your choice. 
```
sudo /usr/local/etc/nc-automount-links sudo systemctl disable nc-automount-links sudo ln -s /media/myCloudDrive /media/USBdrive
sudo systemctl status nc-automount-links
```
from the last command you should see something like:
> Jan 04 19:40:38 nextcloudpi nc-automount-links-mon[352]: myCloudDrive CREATE,ISDIR

Check for the link:
```
ls /media/
```
Reboot and recheck for the link. If it works you should run `nc-datadir` from the Web Panel where you should put a path like `/media/myCloudDrive/ncdata` (remember to replace `myCloudDrive` if you changed it to a different name). Then you check if Nextcloud works. If not try a final reboot.

I am closing the issue for now. Reopen if after reboot the link don't persists .

closed 2018-01-31 by Pant

nachoparker, 2018-02-05:
thanks @Pant 

yes, a way of updating everything to use myCloudDrive is to re-run that stuff from ncp-web. just check what options are using USBdrive and change them

What I am interested in is why the link dissappears in your case. If you see that again, could you paste the output of

```
systemctl status nc-automount-links
```