nextcloud/nextcloudpi #364: Creating SSL certificate failed

StefvanEgmond, 2018-01-21:
Hey, I got an error while making a new cert. for SLL because it didn't do it automaticly.

I got this as log:

----------------------------------------------------------
Launching letsencrypt
Bootstrapping dependencies for Debian-based OSes... (you can skip this with --no-bootstrap)
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
augeas-lenses is already the newest version (1.8.0-1+deb9u1).
ca-certificates is already the newest version (20161130+nmu1).
gcc is already the newest version (4:6.3.0-4).
libaugeas0 is already the newest version (1.8.0-1+deb9u1).
libffi-dev is already the newest version (3.2.1-6).
libssl-dev is already the newest version (1.1.0f-3+deb9u1).
openssl is already the newest version (1.1.0f-3+deb9u1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
virtualenv is already the newest version (15.1.0+ds-1).
0 upgraded, 0 newly installed, 0 to remove and 31 not upgraded.
Creating virtual environment...
Installing Python packages...
Had a problem while installing Python packages.

pip prints the following errors:
=====================================================
Collecting argparse==1.4.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 11))
  Downloading argparse-1.4.0-py2.py3-none-any.whl
Collecting pycparser==2.14 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 17))
  Downloading pycparser-2.14.tar.gz (223kB)
Collecting asn1crypto==0.22.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 21))
  Downloading asn1crypto-0.22.0-py2.py3-none-any.whl (97kB)
Collecting cffi==1.10.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 24))
  Downloading cffi-1.10.0.tar.gz (418kB)
Collecting ConfigArgParse==0.12.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 61))
  Downloading ConfigArgParse-0.12.0.tar.gz (41kB)
Collecting configobj==5.0.6 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 63))
  Downloading configobj-5.0.6.tar.gz
Collecting cryptography==2.0.2 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 65))
  Downloading cryptography-2.0.2.tar.gz (427kB)
Collecting enum34==1.1.2 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 96))
  Downloading enum34-1.1.2.tar.gz (46kB)
Collecting funcsigs==1.0.2 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 99))
  Downloading funcsigs-1.0.2-py2.py3-none-any.whl
Collecting idna==2.5 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 102))
  Downloading idna-2.5-py2.py3-none-any.whl (55kB)
Collecting ipaddress==1.0.16 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 105))
  Downloading ipaddress-1.0.16-py27-none-any.whl
Collecting linecache2==1.0.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 108))
  Downloading linecache2-1.0.0-py2.py3-none-any.whl
Collecting ordereddict==1.1 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 111))
  Downloading ordereddict-1.1.tar.gz
Collecting packaging==16.8 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 113))
  Downloading packaging-16.8-py2.py3-none-any.whl
Collecting parsedatetime==2.1 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 116))
  Downloading parsedatetime-2.1-py2-none-any.whl
Collecting pbr==1.8.1 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 119))
  Downloading pbr-1.8.1-py2.py3-none-any.whl (89kB)
Collecting pyOpenSSL==16.2.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 122))
  Downloading pyOpenSSL-16.2.0-py2.py3-none-any.whl (43kB)
Collecting pyparsing==2.1.8 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 125))
  Downloading pyparsing-2.1.8-py2.py3-none-any.whl (54kB)
Collecting pyRFC3339==1.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 134))
  Downloading pyRFC3339-1.0-py2.py3-none-any.whl
Collecting python-augeas==0.5.0 (from -r /tmp/tmp.E03WXzUuFe/letsencrypt-auto-requirements.txt (line 137))
[... 52 lines cut ...]

Certbot has problem setting up the virtual environment.

We were not be able to guess the right solution from your pip
output.

Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment
for possible solutions.
You may also find some support resources at https://certbot.eff.org/support/ .
Done. Press any key...

----------------------------------------------------------

Output of sudo ncp-report:

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

----------------------------------------------------------
Yes, I got nothing else...
Im currently running NextCloudPi v0.44.15

nachoparker, 2018-02-05:
mmm it works for me, now on certbot-auto 0.21.1

the report fails because letsencrypt fails. try `sudo ncp-diag` and paste the info here

StefvanEgmond, 2018-02-07:
Already working again 👍

StefvanEgmond, 2018-02-07:
After updating all things

nachoparker, 2018-02-07:
great

closed 2018-02-07 by nachoparker