nextcloud/nextcloudpi #365: No internet connectivity

SukerYan, 2018-01-21:
Paste the output of `sudo ncp-report`
I very like Nextcloudpi. I don't report a issue here, just a suggestion. I know the scripts will check the internet status by ping google.com. Unfortunately I am in China. We can't visit any google service. Is it possible to change another domain to check the internet? Such as amazon.com, this is ok for Chinese people. Thanks.

tflidd, 2018-01-21:
You can replace google.com by any other domain:
https://github.com/nextcloud/nextcloudpi/search?q=google.com&type=Code&utf8=%E2%9C%93

We could ping different services and the setup is only considered offline when all fail. Or at least take some service that is not blocked anywhere.

SukerYan, 2018-01-21:
@tflidd Thanks.

Pant, 2018-01-31:
I made a pull request #389 changing 'google.com' to 'nextcloud.com'.

nachoparker, 2018-02-05:
wow didn't think of that... originally I was pinging github, but then it seems like it filters ping in some countries, like Germany, so I changed to google...

another option is to check just 8.8.8.8...

@SukerYan, can you ping 8.8.8.8 from china?

@Pant thank you for your PR!

SukerYan, 2018-02-05:
no, i can not ping 8.8.8.8 from China. :(

nachoparker, 2018-02-05:
can you ping 4.2.2.1 or 4.2.2.2?

SukerYan, 2018-02-05:
they are fine. thanks.

nachoparker, 2018-02-07:
applied in `v0.46.1`

nachoparker, 2018-02-08:
I'll reopen and implement it a bit better, but just relying on the return value of the clone operation

nachoparker, 2018-02-08:
Implemented in `v0.46.4`

closed 2018-02-08 by nachoparker