nextcloud/nextcloudpi #366: Lost database configuration after nc-backup?

Ricardosgeral, 2018-01-22:
Hi to all. Congrats on all the work on nextcloupi. 

I'm facing a problem.
Everything was running ok but then i decided to make a nc-backup

I have the external disk mounted as MyCloudDrive

however i've done a backup and didn't realize that i was doing the backup to the default USBdrive folder that appears in the default window in nc-backup

the backup didn't ended and I have got an error (which i didn't save but it was something about space limitation, which is unlikely because i have 3T and only using 300Gb).

After that i can't access my nextcloud instance. It seems that i've lost the configuration 

[image]

What should i do?
I have an previous nc-export. Can i use nc-import? or shoud i do something else.
need help

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir               (doesn't exist)
rootfs usage         2.1G/15G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.4.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.4
gateway              192.168.1.1
interface            wlan0
certificates         ricarcloud.duckdns.org
certs due            81 days
NAT loopback         no
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

  [Symfony\Component\Console\Exception\CommandNotFoundException]
  There are no commands defined in the "config" namespace.

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Jan 22 06:25:02.426828 2018] [ssl:warn] [pid 696:tid 1992867840] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Jan 22 06:25:03.000777 2018] [mpm_event:notice] [pid 696:tid 1992867840] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Mon Jan 22 06:25:03.000842 2018] [core:notice] [pid 696:tid 1992867840] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>

Windyo, 2018-01-23:
If you're going to do nc-restore, backup your data - I literally just opened a ticket because nc-restore borked my instance. Got everything back, but still.

Regarding nc-backup it looks like you overwrote your config files with the backup one - no idea how that could have happened. the nc-backup https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-backup.sh doesn't really do anything you wouldn't expect it to.

Maybe try to see if nextcloud is in maintenance mode ?

ovpc, 2018-01-25:
Hi,
Thanks for reporting.
Looks you forgot to run nc-init after the restore?
A [how-to backup and restore is available here](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-nextcloudpi-config)

Ricardosgeral, 2018-01-25:
hi, thanks for relpy
run nc-init after restore?
Are u sure. the link with instructions don't mention any nc-init after
nc-restore.

From what i understand,That would put everything from initial instalation
only with a user (administrator) and password ownyourbits

Pant, 2018-01-31:
@Ricardosgeral you shouldn't run `nc-init` for sure because it will replace previous settings in the database (such as users). 
Also did you run `nc-export-ncp`? Based on the guide on the wiki you should have done that, in order to import the configuration as well.

nachoparker, 2018-02-05:
Hi,

it is strange that backing up removed your `config.php`, which is what I think happened. Did the backup finish? did it fail? something else must have happened

do you have backups?

Ricardosgeral, 2018-02-05:
Hi,
I think that the problem may have been related with the confusion I
constantly do between "USBdrive" and "myCloudDrive"

I've done the restore with /media/USBdrive... but i've placed in nc-datadir
/media/myCloudDrive... Could this cause that problem.

What is diference between USBdrive and myCloudDrive. Somethimes USBdrive
appears in /media sometimes it doen't. I'm confused with the both of then

nachoparker, 2018-02-05:
if you run 

```
ls -l /media
```
you will see that your real data is in `/media/myCloudDrive`, and `/media/USBdrive` is a link to it, so they are equivalent.

The reason of this link is to be able to have a default that 'just works' for the configuration options, in case the user doesn't know what the label of the drive is (in your case myCloudDrive)

So did NCP break doing backup? or was it doing restore?

Ricardosgeral, 2018-02-05:
in my case it was doing restore. That was last week, This weekend i have
installed a fresh instance and then did a restore (430Gb) and it worked.
Meanwhile  i've done a backup and no problem.

Maybe it was a problem with power during the restore...

nachoparker, 2018-02-05:
Ok, if the *restore* failed, then that makes more sense. The instance had been partially restored only.

If you capture a new failure to restore, please report it to us.

Glad all is good now

Cheers,

closed 2018-02-05 by nachoparker