nextcloud/nextcloudpi #367: You can actually use external hdd with NTFS, if sharing as samba - contrary to the official FAQ

cloo, 2018-01-22:
I've done it just now. 
So I suggest at least changing the misleading text in FAQ.
Or even better - making it possible via GUI.
For people, who want to access their old archives via nextcloud UI, formatting HDD's is a dealbreaker.

theCalcaholic, 2018-02-08:
Please note that NTFS doesn't support the linux permission system, therefore there might be security risks involved with having your cloud data on an NTFS partition.

cloo, 2018-02-08:
Agree, it is a security risk. 
But if you admit that it is possible, the answer "No, you cannot use it" probably means, you simply forbid people doing it? How impolite and out of FOSS spirit :)

theCalcaholic, 2018-02-08:
Hey, I didn't write the FAQ! I just think that it should be mentioned. ;)

ovpc, 2018-02-09:
Thanks @cloo for reporting, i've added a line to the [FAQ](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#can-i-use-an-external-usb-drive-with-ntfsfat-filesystem) to clarify you can read/write to NTFS/FAT fs.
It is just not supported by devs of NC as a location to store ncdata. Meaning you wont get much help if you do it anyway, and run into issues.

nachoparker, 2018-02-09:
I too added a couple lines. Thanks both

closed 2018-02-09 by nachoparker