nextcloud/nextcloudpi #368: NC beta testing?

testbird, 2018-01-23:
Is it possible use NCP to install a NC beta (13) release?
Would that be hard to install, or easy with ncp-web?

testbird, 2018-01-23:
@ovpc Just found your pull request, could you explain how to use it?

ovpc, 2018-01-23:
Hi
It does not work yet, some bug in the modified code  of my pull request.
I'm sure Nacho will fix it when he's back from his trip abroad.
As a temporary work around, I installed 13.0.0RC1 by making a copy of /usr/local/etc/nextcloud-config-d/nc-nextcloud.sh, calling it nc-nextcloud-beta.sh, and adding "pre" to the "releases" part of the download location (line 130). Also need to make new script executable with: sudo chmod +x nc-nextcloud-beta.sh.

testbird, 2018-01-23:
OK, thanks! What about ${VER_}? How have you set that? From the command line?

ovpc, 2018-01-24:
You can leave that, it takes the version you fill in/want to install, from the version field of the form.

testbird, 2018-01-30:
Bummer, there does not seem to be a NC update option in the docker image, therefore no beta testing possible... :-(

ovpc, 2018-01-30:
Dont know about Docker image.
For NCP a temporary patch ([in nc-nextcloud-beta.sh](https://github.com/ovpc/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-nextcloud-beta.sh)) to install any RC thru ncp-web or ncp-config.
You need to download the file to your ncp's  /usr/local/etc/nextcloudpi-config.d/
Make it read+write to and owned by root:www-data and run the script thru ncp-web or ncp-config. 

You will be prompted to run nc-init, which will take care of caching and database settings.

ovpc, 2018-02-06:
nc-nextcloud.sh tested in v0.45.3 thru ncp-config to install NC13.0.0 (still Beta and in prereleases) and installs fine.

nachoparker, 2018-02-06:
great, thanks!