nextcloud/nextcloudpi #369: NC-Restore borking NCP

Windyo, 2018-01-23:
# After running ncp-backup, doing ncp-restore borks install, even on fresh instance

How it happened:
- did a backup of a working instance
- did some apt-get updates
- everything blew up, to the point nothing was recoverable
- reflashed SD, performed basic setup (ssh enabled, certs copied, etc)
- launched ncp-restore
- finishes sucessfully
- Nextcloud is offline and no error logs are generated.

I _could_ setup from scratch again but downloading apps and setting permissions is annoying enough that I really don't want to. Any ideas ?

Cheers

Windyo, 2018-01-23:
OK, have it narrowed down.

the borking happens when you get to the database restore option.
Setting a DB path that nextcloudpi can't access will make the restoration error out on DB retrieval... Which ironically enough, in my case, made it work, as I still had a valid backup of my db.

Windyo, 2018-01-23:
Further notes:
Using a backup when you have TOTP enabled is a BadIdea unless you have your backup codes ready. Restore will restore TOTP, but the private key will have changed, meaning codes are not valid anymore.

Windyo, 2018-01-23:
Final notes (sorry - I'm doing restores on three pi's just to compare how they differ and to save time):
It seems that using nextcloudpi-setup to set the ncdata and ncdatabase paths prior to using nc-restore makes the restoration go a tad more smoothly (though it still fails on DB restore). Maybe linked to above issue.

Windyo, 2018-01-23:
OK looking at https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-restore.sh I think the problem is at line123 onwards - I hadn't included a data backup.

From what I understand the script overrides whatever was restored, sets the data dir as var/www/nextcloud/data, then performs a variety of cleanup operations.... assuming the data dir is var/www/nextcloud/data.
If you modified your datadir before, this results in a nonfunctional NextCloud instance.

I would replace that with the value currently set in nc-datadir.sh, or the value restored from config.php.

The best way would probably to offer a prompt here.

ovpc, 2018-01-25:
Thanks for reporting.
After nc-restore, did you run nc-init?
[How to backup and restore](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-nextcloudpi-config)

Also please include output of nc-info.

Windyo, 2018-01-25:
Again with the issue closing ?
Should I stop reporting issues then ?

The page you linked to does not indicate to run nc-init.

nc-info is not a valid command.

ncp-report returns this
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

There are no additional details after running for 8 hours, because I ran the command last night.

ovpc, 2018-01-25:
nc-info can be found in ncp-web or ncp-config
also at
/usr/local/etc/nextcloudpi-config.d/nc-info.sh

[image]

Windyo, 2018-01-25:
Launching nc-info
Gathering information...
NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda sdb
datadir              /mnt/raid0/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      690G/2.7T
rootfs usage         2.9G/15G
swapfile             /mnt/raid0/swap
Nextcloud check      ok
Nextcloud version    12.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   10.0.4.128
gateway              10.0.4.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               14:32

You should enable dnsmasq to use your domain inside home

You should run Lets Encrypt for trusted encrypted access

You should enable automount to uyyse your plugged in USB drive

Windyo, 2018-01-25:
Finally got a return from ncp diagnostics, dumping it here

<--! Paste this in GitHub report -->                                                                                                                                                                     
                                                                                                                                                                                                         
<details>                                                                                                                                                                                                
<summary>NextCloudPi diagnostics</summary>                                                                                                                                                               
                                                                                                                                                                                                         
```                                                                                                                                                                                                      
                                                                                                                                                                                                         
NextCloudPi version  v0.44.15                                                                                                                                                                            
NextCloudPi image    NextCloudPi_12-04-17                                                                                                                                                                
distribution         Raspbian GNU/Linux 9 \n \l                                                                                                                                                          
automount            no                                                                                                                                                                                  
USB devices          sda sdb                                                                                                                                                                             
datadir              /mnt/raid0/ncdata                                                                                                                                                                   
data in SD           no                                                                                                                                                                                  
data filesystem      ext2/ext3                                                                                                                                                                           
data disk usage      721G/2.7T                                                                                                                                                                           
rootfs usage         2.9G/15G                                                                                                                                                                            
swapfile             /mnt/raid0/swap                                                                                                                                                                     
Nextcloud check      ok                                                                                                                                                                                  
Nextcloud version    12.0.5.3                                                                                                                                                                            
HTTPD service        up                                                                                                                                                                                  
PHP service          up                                                                                                                                                                                  
MariaDB service      up                                                                                                                                                                                  
Redis service        up                                                                                                                                                                                  
Postfix service      up                                                                                                                                                                                  
internet check       ok                                                                                                                                                                                  
[... cut ...]
d\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php\",\"Line\":254}","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.5.3"}                           
                                                                                                                                                                                                         
```                                                                                                                                                                                                      
</details>

Windyo, 2018-01-30:
Just as a quick update, Tried again on another Pi, same issue. Works fine if I force the script to stop before the DB restore.

nachoparker, 2018-02-05:
hi @Windyo 

I have been away for a few weeks. Thanks for all your testing and info.

Could you make a brief summary of where you are at, or what the issue is exactly at this point? We can take it from there ;)

Windyo, 2018-02-05:
yo @nachoparker ! hope your trip was great. Thanks for looking into this.

Currently I have a working restore, but I still have the problems where Nextcloud "hangs" and some database inconsistencies. At this point i'm considering a brand new setup, as I have tried and failed to get an older image of NP to flash on the SD (problems started arising after the raspbian stretch update).
^ This should probably be its own thread, see #371 

Regarding NC-restore itself, I still have the same behaviour:
- run it normally, everything borks
- force it to stop before the database restore option, instead manually restoring a DB backup I did: everything works fine.

Tried it on two pies, during three restores > same result.

I tried this:
- on a "normal" instance I had blanked out using nc-init
- on a "normal" instance without blanking it first
- on a fresh instance flashed from the latest image

results stay consistent.

As far as setup goes, DB, SWAP and DATA files are all on /mnt/raid0/

let me know if I can get you more info

nachoparker, 2018-02-07:
hi,

if I understand correctly, there's some issue with restore and you think it's with the database step.

maybe you can try running the script line by line and see where it fails?

Windyo, 2018-02-14:
@nachoparker #107 was me ;) I actually still use it in Raid1 for security reasons. Nextcloud is currently our main data storage for home data and I can't afford some of the pictures and stuff being lost.
I managed to stop saturation a bit by switching to Wifi for NCP, now having write times around 1.5times longer than without Raid, with read times faster by around 25%.
As we're using it for massive (6TB) data long-term storage with few writes this actually fits my needs quite well. I still wouldn't advise most people to use Raid1 though :)

Regarding the DB restore step > I can pretty much confirm it's at the DB restore step. edited the script to PAUSE just before that, and the restore went off without a hitch :), with nextcloud happily being restored. Of course the DB wasn't restored, because I stopped the script before that, so I used a manual copy of it to finalize things, and all went well. My instance is now fully restored, but restoring the DB must be a manual step otherwise the restore still fails.

No process hogging anything, actually if I don't PAUSE the script, the script itself runs fine. Nextcloud however won't launch once the script has finished.

Tbh I'm not techy enough to go deeper into this. If you tell me what you want me to do i'll try my best to give you further info. Not sure what I can test more

nachoparker, 2018-02-15:
I am almost positive that this issue has to do with you using RAID, because it is the only report that we have about this.

I guess it would be interesting to see why this happens. If you leave script unchanged

- restore seems to go fine, but
- NC doesn't work

Could you access the database and check that the nextcloud tables are there? otherwise it's fine... we'll wait for someone that runs into this and reads this issue and is able to look into these technical details.

> I can't afford some of the pictures and stuff being lost.

have you considered [this](https://github.com/nextcloud/nextcloudpi/wiki/How-to-periodically-backup-to-a-second-USB-drive-with-NextCloudPi)?

Windyo, 2018-02-15:
.... I'll buy another SD, set it up with a standard USB storage, and see if I can reproduce the error. Wondering if it has to do with the mountpoint more than the fact that it's RAID.

I'll update when I've done that - so not before a week or so.

Regarding "access DB and check if nextcloud tables are there" I'll need some pointers on how to do that, sorry :/

-----------------------------------------
To finalize the Raid1 discussion yes I have considered periodic backups. I ended up going with Raid1 because:
- Raid1 ensures data consistency
- I don't have to test my Raid1 backup every month to heck the backup succeeded
- I was worried that the periodic backup would try to copy over the entire datadir rather than doing a diff, which would saturate the Pi with 6TB of data every so often.

^ none of these are completely valid arguments, but as the setup works generally fine and gives me peace of mind, I see no reason to change it, esp. as I ended up having OK-ish write times :)

nachoparker, 2018-02-15:
> I was worried that the periodic backup would try to copy over the entire datadir rather than doing a diff, which would saturate the Pi with 6TB of data every so often.

You are right. that is the case right now.

the perfect solution is to use btrfs-snapshots, but I have still not implemented the incremental backup based on them. Hopefully I'll have time soon.

I'll close but we can still talk, even reopen later. It helps me organize

closed 2018-02-15 by nachoparker

nachoparker, 2018-02-27:
mmm just to rule it out...

can you try without moving the swap and database, and without raid? let's see if we can find what is the variable combination that makes it fail.

if it does not fail, maybe move sawp, if still ok database... one step at a time

Windyo, 2018-02-27:
So just the datadir on the USB then ? Because my last test was without raid.

I won't have time to get to this for at least a month, some perosnal stuff going on. I'll try to pinpoint where it fails when I can...

Windyo, 2018-10-03:
FYI this does not happen anymore.
Did a restore recently, went off without a hitch.
What the bug was is unknown, but can confirm it is solved, at least insofar as prior testing protocols go.

rawesomeawesome, 2022-03-23:
I'm having this exact issue on my pi4. my backup doesnt include the database and when i run nc-restore it hangs on "restore database...". running nc-init returns baffling errors "nextcloud is not installed" which is just nonsense given that im using the prebuilt pi image of ncp. I only have a single backup as well, due to nc-backup now hanging on "check free space" on my primary nextcloudpi instance.