nextcloud/nextcloudpi #370: No access to nextcloud (Redis Service down)

Beres-Berlin, 2018-01-24:
Hi!

I cannot connect to my nextcloudpi (via http/android app) since a recent reboot (sudo reboot now). Before the reboot, I edited the /etc/rc.local file to include two commands (sudo mount -a and sudo /usr/local/etc/nc-automount-links). Also possible is that the previewgenerator script was running in the background.

ncp-report gives this:

NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      4.1G/350G
rootfs usage         2.1G/30G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.178.29
gateway              192.168.178.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               10:12

...further down, it gives more detail on redis failure:

ExecStop=/bin/kill -s TERM $MAINPID (code=exited, status=1/FAILURE)An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(52): Redis->get('dbb53bbecf397fc...')
#1 /var/www/nextcloud/lib/private/App/InfoParser.php(54): OC\Memcache\Redis->get('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/private/legacy/app.php(615): OC\App\InfoParser->parse('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/private/AppFramework/App.php(61): OC_App::getAppInfo('files')
#4 /var/www/nextcloud/lib/private/legacy/app.php(190): OC\AppFramework\App::buildAppNamespace('files')
#5 /var/www/nextcloud/lib/private/legacy/app.php(116): OC_App::registerAutoloading('files', '/var/www/nextcl...')
#6 /var/www/nextcloud/lib/base.php(673): OC_App::loadApps(Array)
#7 /var/www/nextcloud/lib/base.php(1086): OC::init()
#8 /var/www/nextcloud/console.php(56): require_once('/var/www/nextcl...')
#9 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#10 {main}

I tried to reboot several times (with a reverted rc.local), but the redis problem persists.

Beres-Berlin, 2018-01-24:
...systemctl status redis-server.service gives:

 redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
   Active: activating (start) since Wed 2018-01-24 09:09:48 UTC; 643ms ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)
  Process: 8342 ExecStopPost=/bin/run-parts --verbose /etc/redis/redis-server.post-down.d (code=exited, status=0/SUCCESS)
  Process: 8339 ExecStop=/bin/kill -s TERM $MAINPID (code=exited, status=1/FAILURE)
  Process: 8335 ExecStop=/bin/run-parts --verbose /etc/redis/redis-server.pre-down.d (code=exited, status=0/SUCCESS)
  Process: 8329 ExecStartPost=/bin/run-parts --verbose /etc/redis/redis-server.post-up.d (code=exited, status=0/SUCCESS)
  Process: 8348 ExecStartPre=/bin/run-parts --verbose /etc/redis/redis-server.pre-up.d (code=exited, status=0/SUCCESS)
Cntrl PID: 8352 ((s-server))
   CGroup: /system.slice/redis-server.service
           └─8352 (s-server)

Jan 24 09:09:48 nextcloudpi systemd[1]: Starting Advanced key-value store...
Jan 24 09:09:48 nextcloudpi run-parts[8348]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example

Beres-Berlin, 2018-01-24:
...used nc-restore to restore a backup. Redis Problem still persists:
.
.
.
Launching nc-restore
INFO: overwriting old instance
extracting backup file /media/myCloudDrive/ncp-backups/nextcloud-bkp_20180122.tar...
restore files...
Job for redis-server.service failed because of unavailable resources or another system error.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
.
.
.

journalctl -xe gives this:
--
-- The result is failed.
Jan 24 10:30:41 nextcloudpi systemd[1]: redis-server.service: Unit entered failed state.
Jan 24 10:30:41 nextcloudpi systemd[1]: redis-server.service: Failed with result 'resources'.
Jan 24 10:30:42 nextcloudpi systemd[1]: redis-server.service: Service hold-off time over, schedul
Jan 24 10:30:42 nextcloudpi systemd[1]: Stopped Advanced key-value store.
-- Subject: Unit redis-server.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit redis-server.service has finished shutting down.
Jan 24 10:30:42 nextcloudpi systemd[1]: Starting Advanced key-value store...
-- Subject: Unit redis-server.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit redis-server.service has begun starting up.
Jan 24 10:30:42 nextcloudpi run-parts[9895]: run-parts: executing /etc/redis/redis-server.pre-up.
Jan 24 10:30:43 nextcloudpi run-parts[9901]: run-parts: executing /etc/redis/redis-server.post-up
Jan 24 10:30:43 nextcloudpi systemd[1]: redis-server.service: PID 9900 read from file /var/run/re
Jan 24 10:30:43 nextcloudpi run-parts[9906]: run-parts: executing /etc/redis/redis-server.pre-dow
Jan 24 10:30:43 nextcloudpi sudo[9912]:       pi : TTY=pts/0 ; PWD=/etc ; USER=root ; COMMAND=/bi
Jan 24 10:30:43 nextcloudpi sudo[9912]: pam_unix(sudo:session): session opened for user root by p
lines 1710-1732/1732 (END)
--
-- Unit redis-server.service has finished shutting down.
Jan 24 10:30:39 nextcloudpi systemd[1]: Starting Advanced key-value store...
-- Subject: Unit redis-server.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit redis-server.service has begun starting up.
Jan 24 10:30:40 nextcloudpi run-parts[9870]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Jan 24 10:30:40 nextcloudpi run-parts[9876]: run-parts: executing /etc/redis/redis-server.post-up.d/00_example
Jan 24 10:30:40 nextcloudpi systemd[1]: redis-server.service: PID 9875 read from file /var/run/redis/redis-server.pid does not exist or is a zombi
Jan 24 10:30:41 nextcloudpi run-parts[9882]: run-parts: executing /etc/redis/redis-server.pre-down.d/00_example
Jan 24 10:30:41 nextcloudpi kill[9885]: Usage:
Jan 24 10:30:41 nextcloudpi kill[9885]:  kill [options] <pid> [...]
Jan 24 10:30:41 nextcloudpi kill[9885]: Options:
Jan 24 10:30:41 nextcloudpi kill[9885]:  <pid> [...]            send signal to every <pid> listed
Jan 24 10:30:41 nextcloudpi kill[9885]:  -<signal>, -s, --signal <signal>
Jan 24 10:30:41 nextcloudpi kill[9885]:                         specify the <signal> to be sent
Jan 24 10:30:41 nextcloudpi kill[9885]:  -l, --list=[<signal>]  list all signal names, or convert one to a name
Jan 24 10:30:41 nextcloudpi kill[9885]:  -L, --table            list all signal names in a nice table
Jan 24 10:30:41 nextcloudpi kill[9885]:  -h, --help     display this help and exit
Jan 24 10:30:41 nextcloudpi kill[9885]:  -V, --version  output version information and exit
Jan 24 10:30:41 nextcloudpi kill[9885]: For more details see kill(1).
Jan 24 10:30:41 nextcloudpi systemd[1]: redis-server.service: Control process exited, code=exited status=1
Jan 24 10:30:41 nextcloudpi run-parts[9889]: run-parts: executing /etc/redis/redis-server.post-down.d/00_example
Jan 24 10:30:41 nextcloudpi systemd[1]: Failed to start Advanced key-value store.
-- Subject: Unit redis-server.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit redis-server.service has failed.
--
-- The result is failed.
Jan 24 10:30:41 nextcloudpi systemd[1]: redis-server.service: Unit entered failed state.
Jan 24 10:30:41 nextcloudpi systemd[1]: redis-server.service: Failed with result 'resources'.
Jan 24 10:30:42 nextcloudpi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Jan 24 10:30:42 nextcloudpi systemd[1]: Stopped Advanced key-value store.
-- Subject: Unit redis-server.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit redis-server.service has finished shutting down.
Jan 24 10:30:42 nextcloudpi systemd[1]: Starting Advanced key-value store...
-- Subject: Unit redis-server.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit redis-server.service has begun starting up.
Jan 24 10:30:42 nextcloudpi run-parts[9895]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Jan 24 10:30:43 nextcloudpi run-parts[9901]: run-parts: executing /etc/redis/redis-server.post-up.d/00_example
Jan 24 10:30:43 nextcloudpi systemd[1]: redis-server.service: PID 9900 read from file /var/run/redis/redis-server.pid does not exist or is a zombi
Jan 24 10:30:43 nextcloudpi run-parts[9906]: run-parts: executing /etc/redis/redis-server.pre-down.d/00_example
Jan 24 10:30:43 nextcloudpi sudo[9912]:       pi : TTY=pts/0 ; PWD=/etc ; USER=root ; COMMAND=/bin/journalctl -xe
Jan 24 10:30:43 nextcloudpi sudo[9912]: pam_unix(sudo:session): session opened for user root by pi(uid=0)

Pant, 2018-01-31:
In your last comment there seems to be something wrong with your copy-pasting from your terminal. Please run `journalctl -xe > journalctl-output.txt`, then open the file `journalctl-output.txt` with a text editor, and post the output here again.

nachoparker, 2018-02-05:
I wonder why redis is down... can you check the logs, then try to bring it up again and if it fails, check the logs again?

something like

```
sudo cat /var/log/redis/redis-server.log
sudo systemctl restart redis-server
sudo cat /var/log/redis/redis-server.log
```

nachoparker, 2018-02-07:
I'll close after 14 days of inactivity, please reopen if there's new info

closed 2018-02-07 by nachoparker

Windyo, 2018-10-02:
Hey @nachoparker @Pant have the same exact issue.
Nextcloud down after a normal reboot with redis down.
rebooting doesn't fix it, restoring doesn't fix it.

have th exact same journal -xe output.

any ideas ?

Windyo, 2018-10-02:
hmmm, redis.log gives

```7947:M 02 Oct 12:06:46.483 # Server started, Redis version 3.2.6
7947:M 02 Oct 12:06:46.561 # Short read or OOM loading DB. Unrecoverable error, aborting now.
7947:M 02 Oct 12:06:46.561 # Internal error in RDB reading function at rdb.c:1428 -> Unexpected EOF reading RDB file```

Windyo, 2018-10-02:
LOL that filesize
```-rw-rw----  1 redis redis 4175694 Oct  2 10:24 dump.rdb```

Windyo, 2018-10-02:
OK, I fixed it.

I don't really know what the problem was, only that shit was corrupted after shutdown, so I'm going to dump all components. Maybe someone more intelligent than me can figure this out. Note that due to lots of smallish changes my install is not fully standard (uninstalled mysql and reinstalled the full mysql-server packages for example, based on ncp 14.0.1)

# Symptom
- accessing nextcloud shows "internal server error". Accessing Nextcloud log shows Redis fucked up.

# System State
`/var/www/nextcloud/config/config.php` lists the redis config and the socket
pinging the socket `/var/run/redis/redis.sock` with `sudo -u www-data redis-cli -s /var/run/redis/redis.sock ping` gives you connection refused
`vim /etc/redis/redis.conf` give syou the redis conf, including the running directory, the `daemonize` parameter is `TRUE`,, the socket file is correctly specified there
`sudo vim /etc/systemd/system/redis.service` lists the same run directories and socket files.

# Checks
ensure `redis.service` is `Type=Forking`
ensure the `RunTimeDirectory` (default `/var/run/redis/`) is owned by `redis:redis`
If not, make these changes and run `sudo systemctl daemon-reload`

#Cleanup
delete the old dump (or at least move it to a backup file) `sudo rm /var/lib/redis/dump.rdb`
rerun the redis server `sudo service redis-server start`

#Prevention

Create a script "safe-shutdown.sh"
```#!/bin/sh
sudo service apache2 stop
sudo service mysql stop
sudo service redis-server stop
sudo umount /mnt/raid0```

make it executable
symlink it to `/etc/rc0.d` and `/etc/rc6.d`

@nachoparker @Pant

nachoparker, 2018-10-02:
Never seen that before. Thanks for letting us know, we'll keep an eye on this

Windyo, 2018-10-02:
Seems to be brand new with 14.0.1. I can reproduce it 100% of the time with a backup of my old nextcloud instance.

Happens on system shutdown only - reboots seem to be fine.

That said, with my instance, it could be anything from mysql server to the Raid config that handles data storage so I don't think investing anymore energy into this specific instance of the bug is interesting. We'll see if it rears its head ever again.

pasxalisk, 2018-10-05:
I had exactly the same problem and after searching a while i solved it like this:

sudo apt-get install --reinstall redis-server