nextcloud/nextcloudpi #371: Older NCP images

Windyo, 2018-01-24:
Hey is there anyway to get an older NCP image ?

I keep running into issues with the latest one on raspbian stretch, but everything was working fine before. I would like to test with another image.

Cheers

ovpc, 2018-01-25:
Hi,
Thanks for reporting.
Which image are you using? Did you check m5sum?
What are the returning issues you run into?

Windyo, 2018-01-25:
Hi,

I'm using the one currently online ( NextCloudPi_12-04-17  ). Ideally I'm lookin gfor the image that was online on  21 Nov 2017 as I know that one was still stable.

Issues I have ATM are the Pi completely freezing sometimes (unreachable via ssh, unresponsive via keyboard input), and apache throwing error 503 intermittently. Funny thing is I've checked the apache logs, nothing there. I've checked all the logs I could, there's nothing.

and regarding the freeze I thought it was because of the overclock, so I removed that, but it still happens.

Literally started happening after I updated to the version that supports raspbian stretch. I also thuoght it could be a problem during upgrade, so I reflashed the SD with thenew image, upadted everything, then restored just nextcloud. Still happens.

So now I want to test with an older image see if that fixes it :)

Thanks

ovpc, 2018-01-25:
Sounds like a bad power source or faulty sd-card .....
Probably nothing to do with image, no one else reported any problems and lots been using it....
When it happens again paste output of:
`
sudo dmesg`

closed 2018-01-25 by ovpc

Windyo, 2018-01-25:
Why has this been closed ?

The NCP installation has been running fine for months, and only started bugging out after update.
I am not even requesting your time to help debugging, just a link to prior images so I can do my debugging.

A modicum of interest or politeness would be appreciated, instead of closing the issue arbitrarily on someone who is experiencing a problem.

ovpc, 2018-01-25:
Your question (about older NCP images) has been answered, there seems to be no reason to keep it open. You can still discuss it here or feel free to open an new issue. You can also join discussions and ask help on our [Telegram channel](https://t.me/NextCloudPi) and [Nextcloud community](https://help.nextcloud.com/)

Windyo, 2018-01-25:
The hell it has, you said "I don't think so".

You don't have access, sure, that doesn't mean another contributer or nacho doesn't have them.

ovpc, 2018-01-25:
Please read answer again. 
> You can install any version of NC using nc-nextcloud from ncp-web or ncp-config.

Windyo, 2018-01-25:
Yeah but that installs NC itself - I am looking for the image with raspbian jessie and the underlying packages, to see if it's something with that. which is why I want the image for NCP, not just NC itself.

Not sure if that makes sense.

Pant, 2018-01-31:
As far as I can tell, you can clone the repository, checkout at the tag you want to use, and produce the image. There are info in the front page on the github project (https://github.com/nextcloud/nextcloudpi#how-to-build). 

But, using a such an old image, with Raspbian Jessie wouldn't be a good idea, as there are many updates in ncp since then and many things will not work as expected.

It is explained in Nacho's blog: https://ownyourbits.com/2017/08/20/nextcloudpi-updated-to-raspbian-9-stretch/

>  [...] It is important that people start from the newest version as all software updates will be run over Raspbian Stretch from now on. This will also guarantee a life cycle of two years for security updates, and hopefully things will be quiet until the release of Nextcloud 13.

nachoparker, 2018-02-05:
hi,

I have older images if you are interested, but I would rather we investigate what's going on ;)

Anyway, which one do you want?

Windyo, 2018-02-06:
@nachoparker if you have the last image before the Raspbian Jessie update that would be great. To clarify, the last image you made available, on which doing ncp-update updated raspbian to jessie.

My objective is to make a backup of the current SD which (kinda) works, then reflash my SD to the last image I know works.
If using that old image removes any issues with NC, then we can investigate what the problem is.
If it doesn't, it means it's either my NC config, or my hardware, that's messed up.

I'de rather finish testing and eliminating possible sources for the bug before dragging anyone else into this :)

Thanks a lot.

nachoparker, 2018-02-06:
alright, I'll look for it, but bear in mind that ncp-update won't work on such an old image

nachoparker, 2018-02-07:
Hi, you can get the image from https://ownyourbits.com/downloads/testing

Keep in mind that updates won't work well on that image, and that a *ton* of fixes and improvements have taken place.

It is impossible that it was working better before, in my opinion, but we can try to find out what's happening to you

Windyo, 2018-02-14:
Hey @nachoparker just thought I would update here.

So I dl'd it and reflashed, and I didn't have any issues anymore.
So of course I updated to the latest version again to check if I still had issues - had none.

Found this weird so I did a counter check. Seems the culprit was actually the nextcloud sync client. The newer version has no issues at all, while the old one still won't sync fully.... AND puts NCP offline every once in a while.
I thought about sync auth errors putting it offlin, as Nextcloud is notorious for not liking being spammed by auth errors. But logs don't show anything.

So to sum it up:
- thanks for the image
- you were right, image wasn't the culprit
- probably Nextcloud-client related
- issue not reproductible with new client. Haven't changed hardware or anything on the server side.

Lacking any way to reproduce these specific errors, I'll stop investigating this.

nachoparker, 2018-02-15:
ok, I'm relieved... thanks for letting us know!