nextcloud/nextcloudpi #373: Odroid HC-1 Image - nc-static-IP not working

rspierenburg, 2018-01-25:
I'm having issues with setting Static IP in the NCP-Web using nc-static-ip.  I tried to set it to 192.168.1.9 and it tells me it couldn't restart DHCPCD as it couldn't be found, but it says that it changed the IP to 192.168.1.9 but nc-info still shows it as 192.168.1.102 which is the DHCP address.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.44.15
NextCloudPi image    NextCloudPi_11-22-17
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      98G/3.6T
rootfs usage         1.9G/7.1G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    12.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.102
gateway              192.168.1.1
interface            enx001e0630ca1a
certificates         none
certs due            none
NAT loopback         no
uptime               2:09

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.102",
            "2": "www.spierenburg.ca"
        },
        "datadirectory": "\/media\/myCloudDrive\/ncdata",
        "overwrite.cli.url": "https:\/\/spierenburg.ca",
        "dbtype": "mysql",
        "version": "12.0.5.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "ocuh990ujuea",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
[... 83 lines cut ...]
```

{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:00+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:00+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:00+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:01+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:01+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:01+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:01+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:01+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:04+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:09+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:10+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:10+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:10+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) A [... cut ...]
{"reqId":"WmnQrX8AAQEAAAUQ7yIAAFEK","level":3,"time":"2018-01-25T12:49:12+00:00","remoteAddr":"24.138.71.16","user":"robert","app":"PHP","method":"POST","url":"\/index.php\/apps\/audioplayer\/scanforaudiofiles","message":"Undefined index: picture at \/var\/www\/nextcloud\/apps\/audioplayer\/lib\/Controller\/ScannerController.php#671","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/63.0.3239.132 Safari\/537.36","version":"12.0.5.3"}

```
</details>

nachoparker, 2018-02-05:
thanks for reporting. I'll look at it

nachoparker, 2018-04-12:
after a quick test, it seems that armbian uses NetworkManager/dhclient, instead of dhcpcd which is used by Raspbian.

nachoparker, 2018-05-10:
fixed in `v0.54.5`

closed 2018-05-10 by nachoparker

rspierenburg, 2018-05-10:
Thank you so much.

nachoparker, 2018-05-10:
you are welcome! thank you for your feedback