nextcloud/nextcloudpi #375: "Your data directory is invalid" after first backup

piexad, 2018-01-26:
Hello, 
I've made my first backup today. But something wrent wrong at the end. 

What I've done : 
`sudo -u www-data php occ maintenance:mode --on`
Result : OK, server in maintenance mode.

`rsync -Aax nextcloud /my/directory/to/save/nextcloud-dirbkp_`date +"%Y%m%d"`/`
Result : OK, backup copied in my directory.

`mysqldump nextcloud --single-transaction -h localhost -u root -p[myprettypassword] > nextcloud-sqlbkp_`date +"%Y%m%d"`.bak`
Result : Seems OK. The sqlbkp*.bak is in my /nextcloud directory. 

`cp sqlbkp*.bak /my/directory/to/save/`
Result : sqlbkp*.bak copied. 

In my /nextcloud/directory : 
`sudo -u www-data php occ maintenance:mode --off`

And this is dramatic right now : 
`Nexctcloud is in maintenance mode - no app have been loaded` 

`Your data directory is invalid `
`Ensure there is a file called ".ocdata" in the root of the data directory`

`Cannot create "data" directory`
`This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions`

`An unhandled exception has been thrownn:`
`Exception : Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:145`
`Stack trace:`
`#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Consolle\Output\ConsoleOutput))`
`#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')`

When I inspect /var/www/nextcloud/ there is no .ocdata. 
My data are in a USBDrive.

What can I do ?

Ricardosgeral, 2018-01-26:
Same problem here

piexad, 2018-01-26:
Not really the same because when I am on the nextcloud webpage, I canno't see log boxes and it says that it can't find ".ocdata".

In french : 

    Votre répertoire n'est pas valide

    Assurez-vous que le répertoire de données contient un fichier ".ocdata" à sa racine.
    Impossible de créer le dossier "data"

    Ce problème est généralement résolu en donnant au serveur web un accès en écriture au répertoire racine. Voir https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions

ovpc, 2018-01-26:
> @piexad When I inspect /var/www/nextcloud/ there is no .ocdata.

 .ocdata is located at /var/www/nextcloud/data/ :

` ls -lha /var/www/nextcloud/data/`

Your /var/www/nextcloud directory is, it seems, not owned and writable by user www-data . You can use the nc-fix-permissions script, in ncp-web or ncp-config in terminal, to try and correct this, or use chown command from the terminal.

piexad, 2018-01-26:
Ok I can see the ".ocdata" but it's not in /var/www/nextcloud/data because my datas are in an USB HDD. 
In media I have two directory (weird ??) USBdrive1 and myCloudDrive, both containing an .ocdata. 

When using ncp-config : 
`Launching nc-fix permissions`
`Nextcloud is in maintenance mode - no app have been loaded`
`data dir not found`
`Done. Press any key...`

But I can't turn off maintenance mode (same error than in my first post). 
Tried to change datadir (nc-datadir) for myCloudDrive or USBDrive1 : 
`Nextcloud is in maintenance mode - no app have been loaded`
`Error reading data directory. Is the NextCloud running and configured?`
`Done. Press any key...`

Quite confusing

piexad, 2018-01-26:
ok so I've made chown for /var/www/nextcloudpi ; USBdrive1 and myCloudDrive. 
Then I realized that in config.php the directory was USBdrive. So I changed it in USBdrive1. 
Still don't understand what happened but it works. How can I figure it out ?

nachoparker, 2018-02-05:
it seems like you are doing a manual backup with rsync, instead of using `nc-backup`

if that is the case, please use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for general help. because it is not a NCP bug I will close it.

please, reopen if I am wrong

closed 2018-02-05 by nachoparker

boraygowda, 2018-02-19:
Hi All, I have a similar problem, but with the same end result.
In my case, I'm running NC in an Ubuntu guest/VM inside a Windows host. I'm mounting a shared folder on the VM so that my files can be accessible from Windows.  The shared folder and all files under it automatically have the user:group as _root:vboxsf_.  I'm unable to change the ownership to _www-data:www-data_.  I've tried adding _vboxsf_ and root users to _www-data_ groups.  But any forced ownership changes to the shared directory from within the VM is not working.

Any ideas as to how I can surmount this issue?
Would appreciate any help.

Thank you.