nextcloud/nextcloudpi #379: I want to open maintenance mode, but I received this .....

chiyi4488, 2018-01-27:
I want to open maintenance mode, but I received this .....

`Could not open input file: occ`

Why?

(I type `sudo -u www-data php occ maintenance:mode --on` in SSH ) 

If you need IP, you can contact me
My e-mail: chiyi4869@gmail.com

Paste the output of `sudo ncp-report`
<--! Paste this in GitHub report -->
<details>
<summary>NextCloudPi diagnostics</summary>

```

theCalcaholic, 2018-01-29:
What's the output of `sudo ls -l /var/www/nextcloud/occ`?

chiyi4488, 2018-01-30:
No ,can't
pi@nextcloudpi:~ $` sudo ls -l /var/www/nextcloud/occ`
-rwxr-x--x 1 www-data www-data 283 Dec  4 07:18 /var/www/nextcloud/occ

theCalcaholic, 2018-01-30:
I think you have to do `sudo -u www-data -s /bin/sh /var/www/nextcloud/occ`.

OCC has to be executed as the www-data user.

theCalcaholic, 2018-01-30:
That looks correct... What happens when you invoke occ directly (e. g. `sudo -u www-data -s /bin/sh /var/www/nextcloud/occ list`)?

chiyi4488, 2018-01-31:
pi@nextcloudpi:~ $ sudo -u www-data -s /bin/sh /var/www/nextcloud/occ list
/var/www/nextcloud/occ: 2: /var/www/nextcloud/occ: cannot open ?php: No such file
/var/www/nextcloud/occ: 3: /var/www/nextcloud/occ: /bin: Permission denied
/var/www/nextcloud/occ: 4: /var/www/nextcloud/occ: Syntax error: "(" unexpected

chiyi4488, 2018-01-31:
I have many questions now, can you ssh to me?
Join my facebook or email, I tell you my ip
This is my facebook:
This is my email:
If you can help me, I will be very happy

chiyi4488, 2018-01-31:
I am a noobie at programming. Please help me.

theCalcaholic, 2018-01-31:
Still looks like a permission problem - what's the output of `sudo ls -ld /bin` and `sudo ls -l /bin/php`?

What are you running nextcloudpi on (raspberrypi 2/3, VM, ...)?
I'll send you an email regarding ssh later on. But you should be careful with such offers. Anyone who reads this issue could send you an email and ask for access... Don't trust any emails coming from different adresses than mine (thecalcaholic@web.de) or those associated to the repository maintainer nacho.

chiyi4488, 2018-01-31:
OK, I hope I can learn something from your major
I will delete my email in the above message.
i run nextcloudpi from raspberry pi 3 .

theCalcaholic, 2018-01-31:
I have received your email :)

theCalcaholic, 2018-02-04:
Okay, I've had a look at your NCP server. For some reason all the commands didn't work when specifying the shell to use for ww-data with `sudo -u www-data -s <shell>`. Without it, however, everything worked as expected.

I could succesfully execute:
`sudo -u www-data /var/www/nextcloud/occ <command>`
`sudo -u www-data php /var/www/nextcloud/occ <command>`
`sudo -u www-data /usr/bin/php /var/www/nextcloud/occ <command>`

So there are no permission or path problems with neither php nor occ.

theCalcaholic, 2018-02-04:
If the issue with setting the shell explicitely doesn't occur for anyone else and your problem is solved, you might want to close the issue. ;)

chiyi4488, 2018-02-05:
Wow, it cool.
thank you!!

chiyi4488, 2018-02-05:
I still have a new problem, I can not use a 1TB hard drive as the data location.

closed 2018-06-12 by nachoparker