nextcloud/nextcloudpi #380: NCP security issues

theCalcaholic, 2018-01-27:
While working on PR  #372, I noticed some bad practices in the

theCalcaholic, 2018-01-29:
I'd be willing to create a pull request which addresses these issues, but some fixes should be decided on by the maintainers (e.g. how to store config data for the apps).

If you're interested I can propose a solution.

Pant, 2018-01-31:
@theCalcaholic, thanks for identifying these security issues. The main developer is not available at the moment but please when you have time, write here what you propose and if you want make pull request with based on your opinions on how things should be and Ignacio will take a look at them as soon as he is available.

theCalcaholic, 2018-02-09:
Okay, I have looked into these things a bit. For demonstration purposes I have created a branch on my fork of NCP which I will link to. Here's what I have come up with:

1. ncp-web should be run as a dedicated user, so that the risks of someone getting access to ncp-launcher.sh are minimized.
2. I have [added a check](https://github.com/theCalcaholic/nextcloudpi/commit/f14e9de20856459e40beff3b4ac5835f840d8241) to prevent path traversal attacks when requesting a ncp app to be run.
3. All apps (under /usr/local/etc/nextcloudpi-config.d/) should be owned by root and have permissions 500 or at least 700 to prevent reading, writing and executing by anyone else.
4. I have developed methods to easily store and access app configurations in deciated `.conf` files (e.g. configuration for letsencrypt.sh would be stored in letsencrypt.conf) in ini file format and [added helper functions for loading and saving from within batch files](https://github.com/theCalcaholic/nextcloudpi/commit/2e8da22a7376708a267a4d185c7539b9de60c1dc). 
This makes it very easy and flexible to define a formular in a ncp app. An example can be found [here](https://github.com/theCalcaholic/nextcloudpi/blob/2e8da22a7376708a267a4d185c7539b9de60c1dc/etc/nextcloudpi-config.d/letsencrypt.sh#L16-L22). When the variables that can be seen in the example are defined, a json representation of the whole form can be retrieved with a call to `form2json`. This function also filters out the value of password fields before returning them. Maybe it would be even better to put the form in another document and just source it from within ncp-launcher.
As mentioned before, the config values themselves [are stored in ini file format in dedicated files]. This makes it very easy (and less hacky than so far ;) ) to read and update them. For these purposes the functions `load_config <config_file>` and `save_config <config_file> <semikolon_separated_key_value_pairs>` are defined in library.sh.
Any of these functions should be called via ncp-launcher.sh from the ncp web panel backend. I'd propose to add a command parameter to the script to determine the correct action.

I'm aware that the 4. point I'm proposing means a considerable amount of work (which I'll gladly be willing to contribute to); but it has a good amount of advantages over the current architecture:
- Field types and representations can be easily expanded (without having to encode everything in a single string
- It would provide a clean separation between code and configuration data - no more rewriting `.sh` scripts.
- Setting and getting configuration data can only be done by ncp-launcher.sh and therefore doesn't require any access permissions for other users
- Updates of ncp apps can be installed a lot easier without messing with the configuration data

nachoparker, 2018-02-09:
Hi,

Give me a couple days to look into this more carefully, but I'll address some of the concerns from the first message

> Passwords are stored in plain text in files readable for the user www-data. However some of these passwords would allow access to a tool (the web panel) acting with root permissions (privilege escalation).

>    the app scripts are writable by www-data by default - they should only be write- and readable by root to prevent malicious script injection/execution by an attacker which managed to get control of www-data.

the scripts sadly need to be writable by www as of today because the configuration is saved in the script file itself. Originally there was no ncp-web so this was more acceptable. In order to prevent injection/execution I sanitize the input in the PHP backend as you already know.

I think that we are safe from now, but I totally aggree that we should change this. It would be no small change though.

> I didn't test it, but as far as I can see there are no measures in place to prevent ncp-launcher.sh from executing scripts outside /usr/local/etc/nextcloudpi-config.d/ (e.g. by supplying "../../../../bin/evil.sh" as ref parameter in a launch request). The simplest fix would be to use refuse file parameters containing / or just using the part behind the last / in ncp-launcher.sh .

We should include that case in the sanitizer, good one. Right now, you need a CSRF single use token to use the backend functions, and you can only get it by opening ncp-web, which is password protected, so I think we are still safe. Can you include a check to make sure that the script is in the intended folder?

> From a quick look at the backup and restore scripts it seems, that the mysql root user doesn't need a password for accessing the database. This allows an attacker who gets control over any user to read all databases.

yes it does, but it is included in `/root/.my.cnf`. This is randomly generated. No problems here

> Current passwords should not be readable inside the web panel - at least password fields should be used.

Well, we would have to detect a password field, not very hard. I am not very worried by this though, there are many apps that allow you to see the password fields. I would be happy to improve this as one of the many ncp-web improvements. When I am a bit more free I would like to discuss other improvements that I have in my list, but haven't had time yet to address. With your help we can make this progress faster!

---

> Additionally to the fixes described above I'd propose to change the way NCP apps store there configuration to a more flexible and more secure solution. This would also add flexibility to the way the forms are displayed in the web panel (e.g. easily adding more field types).

I have been thinking that too, the only problem is that it would be quite a bit of work, but eventually it has to be changed if we are to continue improving the web. Keep in mind that all those things happen because I am reusing code from ncp-config, and changes we do with ncp-web in mind need to not break ncp-config, so most likely these changes would imply a full rewrite.

It will be done if it needs to be done, but have to understand the lack of manpower and time.

nachoparker, 2018-02-09:
> ncp-web should be run as a dedicated user, so that the risks of someone getting access to ncp-launcher.sh are minimized.

I am not sure of what you mean... ncp-web needs to be run by the www-data user, as any other web app. ncp-launcher.sh lives in a directory only accessible by this user.

I'll fully respond to your second message later

theCalcaholic, 2018-02-09:
@nachoparker With nginx it is very easy to run different web pages with different users. I've never done this with apache, but apparently this is the way to do it: http://httpd.apache.org/docs/2.0/suexec.html

Using this adds considerably to security, because an attacker who gets control over one web service will never be able to read other services' data.

theCalcaholic, 2018-02-09:
@nachoparker In my last message I presented a concept for data storage for the ncp apps. Basically, it changes the way data is accessed to always use the (for this purpose adapted) ncp-launcher.sh script.

That way, none of the ncp app files would need to be accessible by www-data; but yes, it's not a small change. Apart from that, bash is not the most fun language to handle data in. ;)

theCalcaholic, 2018-02-09:
I added some illustrations to [my previous comment](https://github.com/nextcloud/nextcloudpi/issues/380#issuecomment-364315737).

theCalcaholic, 2018-02-22:
I added a script to test the form2json method: [json_test.sh](https://github.com/theCalcaholic/nextcloudpi/tree/feature/ncp-apps-config-separation/etc/json_test.sh)

It prints the form definition in `etc/nextcloudpi-config.d/letsencrypt.form` in json format.

nachoparker, 2018-02-27:
I like it! this deep changes were long overdue, so I think it's the right moment to implement them.

This was too much code reuse from ncp-config to ncp-web, and if we separate code and data better, we can achieve much more on ncp-web.

I suggest you send a PR with this, and we can work out the details over there.

---

> @nachoparker With nginx it is very easy to run different web pages with different users. I've never done this with apache, but apparently this is the way to do it: http://httpd.apache.org/docs/2.0/suexec.html

Using this adds considerably to security, because an attacker who gets control over one web service will never be able to read other services' data.

We could also create a separate 'enhancement request' covering this.

nachoparker, 2018-02-27:
I'll add my own suggestions on how the structure of the ncp-apps could be in the future

---

- create a directory per ncp-app with code + data + translations
```
├── nc-backup
│   ├── l10n
│   │   ├── de.json
│   │   └── es.json
│   ├── nc-backup.cfg
│   └── nc-backup.sh
└── nc-restore
    ├── l10n
    │   ├── de.json
    │   └── es.json
    ├── nc-restore.cfg
    └── nc-restore.sh
```

- update ncp-config to be able to read and set config files (we'll use your code). Same for ncp-web.
- new metadata to include in nc-restore.cfg
  - displayed name ( instead of 'file name without .sh' )
  - type of the field ( password, text, dropdown, path )

- classify ncp-apps into categories, and group them in directories ( let's call them _category directories_ ).
  - create a metadata file inside each category folder with icon, description, name
  - create a directory with translations inside each category folder

```
.
├── data_management
│   ├── category.cfg
│   ├── icon.png
│   ├── l10n
│   │   ├── de.json
│   │   └── es.json
│   ├── nc-backup
│   │   ├── l10n
│   │   │   ├── de.json
│   │   │   └── es.json
│   │   ├── nc-backup.cfg
│   │   └── nc-backup.sh
│   └── nc-restore
│       ├── l10n
│       │   ├── de.json
│       │   └── es.json
│       ├── nc-restore.cfg
│       └── nc-restore.sh
└── security
    ├── category.cfg
    │   ├── icon.png
    ├── l10n
    │   ├── de.json
    │   └── es.json
    ├── fail2ban
    │   ├── fail2ban.cfg
    │   ├── fail2ban.sh
    │   └── l10n
    │       ├── de.json
    │       └── es.json
    └── modsecurity
        ├── l10n
        │   ├── de.json
        │   └── es.json
        ├── modsecurity.cfg
        └── modsecurity.sh
```

The idea is that we add/remove folders and all of the ncp-app is contained on those folders.

Also, we'll have to think what we do with `nc-import` and `nc-export`.

nachoparker, 2018-02-27:
actually I am thinking that we should create a new branch for this

nachoparker, 2018-02-27:
Still reading this in depth... so you suggest to have a basic `.form` file and bash functions to convert it to and from json.

How about saving everything directly in json, which is easy to do from php, and read it in bash? that way not only we are safe from sourcing malicious variables, but also seems more simple.

We can use [jq](https://github.com/stedolan/jq) for this

nachoparker, 2018-02-27:
this seems like a pure bash option

https://github.com/dominictarr/JSON.sh

nachoparker, 2018-12-28:
Implemented in _rework_ branch

closed 2018-12-28 by nachoparker